openscad/openscad #1744: while loop

guowcas, 2016-07-21:
Hi guys, 
It turns out that current OpenSCAD only has a "for(){}" loop grammar. Some time the while loop is much powerful, for example if you are dealing with hex geometry, or many other recursive issues. 
Is there any plan to include the while loop to OpenSCAD grammar? 
Many thanks.

doug-moen, 2016-07-21:
A while loop needs one or more mutable variables to represent the iteration
state. We don't have mutable variables, so a standard while loop can't
work. Many of the usual loop idioms found in imperative languages need to
be recoded as recursive functions or recursive modules in OpenSCAD.

The closest thing to what you are requesting is an experimental feature,
the C-style for loop, which has to be enabled explicitly in a developer
build using Preferences->Features in the GUI. It's not in the standard
release. Using this feature, you explicitly declare each iteration
variable, with an initial value, and with an expression that updates the
iteration variable to a new value at the end of each iteration. You also
specify a "while" condition. So this feature is a pretty good approximation
of a `while` loop within a functional language. Eg,
```
[ for (i = 0; i < 10; i = i + 1) i ]
```

However, this feature is currently restricted to use within list
constructors: it is used to generate a sequence of values for inclusion
within a list that you are constructing. The feature could, in theory, be
made available at the top level of a script, for generating a series of
shapes, just like the existing `for` loop, but that hasn't been done yet.
So, maybe your feature request would be satisfied by making the C-style for
loop available at the top level? It might depend on the code you want to
write. Consider posting an example of what you want to do.

guowcas, 2016-07-22:
Thanks a lot for your information. I will keep it up when the C-style for loop releases. Currently the recursive task is not deep, so I can handle it with stone-age copy-paste.

t-paul, 2018-04-06:
Closing as c-style for loop (https://github.com/openscad/openscad/pull/1537) is available in the dev version.

closed 2018-04-06 by t-paul