openscad/openscad #1746: surface() with PNG is broken

MichaelAtOz, 2016-07-22:
@t-paul Looks like surface() with PNG is broken.

As per [forum](http://forum.openscad.org/Surface-renders-but-adding-any-other-object-doesn-t-td17948.html#a17951) even 
```
surface("smiley.png",invert=false,center=true); // smiley.png from the wiki of surface()
cube(1); 
```
Crashes. Uses all available memory.

2015.03-2 & a recent snapshot.

t-paul, 2016-07-22:
Hmm, on Linux it renders in just under 1 minute with both 2015.03-1 and nightly build.

MichaelAtOz, 2016-07-22:
Without the cube() it is instant;
```
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 2
Geometry cache size in bytes: 69052720
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 0 seconds
   Top level object is a 3D object:
   Facets:     478601
Rendering finished.
```
Using this 

[image]

The3rdIcon, 2016-11-11:
On Windows I was having this issue as well with PNG files. I get a warning about Polyset has degenerate polygons but nothing shows up in the preview window. Please let me know if you need any other information from me to help resolve this situation.

The3rdIcon, 2016-11-11:
This is still an issue with version 2015.03-2. It crashes on my windows machine.

adegard, 2020-08-30:
I have the same issue. 
Surface is not working when introduce over other drawings...
Someone found the trick?

kintel, 2024-03-27:
@adegard Do you still have this problem? This works for us on all platforms..