openscad/openscad #1747: i want to build a scad based customizer, please help

joinalahmed, 2016-07-22:


amarjeetkapoor1, 2016-07-22:
@joinalahmed there is project going on as part of GSOC  2016 related to customizer you could view and test that. Plus you should also suggest improvements, contribute in it and give your version of customizer. 
 

https://github.com/openscad/openscad/tree/gsoc2016 
http://www.openscad.org/news.html

joinalahmed, 2016-07-22:
i've seen your app and i want to work on it, basially build a customiser
that typically work like A THINGIVERSE CUSTOMISER

amarjeetkapoor1, 2016-07-22:
I think there are a some links that you should go through or you might have already visited them to know exact status of project and discussions related to project.

https://github.com/openscad/openscad/wiki/Google-Summer-of-Code-2016 

Rest we can talk about this project on mailing list of openscad ( http://forum.openscad.org/ )or IRC (#openscad) channel and you will also get guidance from teepee and kintel.

closed 2016-07-22 by amarjeetkapoor1

joinalahmed, 2016-07-22:
Thankyou i'll look forward