openscad/openscad #1753: scripts/uni-build-dependencies.sh ignores MAKEFLAGS

krichter722, 2016-07-25:
Running `scripts/uni-build-dependencies.sh` and watching CPU usage indicates that it ignores or overwrites MAKEFLAGS which make builds unnecessarily slow.

experienced with openscad-2015.03-1-582-g9519cc9

donbright, 2016-08-12:
are you want to set the number of cpus with MAKEFLAGS="-j 5" or something like that?

krichter722, 2016-08-13:
> are you wanting to set the number of cpus with MAKEFLAGS="-j 5" or something like that?

For example, but that's not a support request for a workaround, but to keep `MAKEFLAGS` in order to KISS.

donbright, 2016-09-04:
ok so here's how i think we can fix this is 

1. Get rid of all invocations of 'make -j$NUMCPU'. replace them with just 'make'

2. Now, at the beginning of build scripts, check if the user has set NUMCPU, then just add '-j $NUMCPU' to MAKEFLAGS environment variable. 

Result:

People using NUMCPU=3 will see no changes. Everything will work as normal.

People using MAKEFLAGS=-j3 will see it 'just work', except I am not sure about 'boost' which uses 'b2' to build instead of make, but boost doesn't take much time to build in the first place as it is mostly a header-based library and we only build a few parts of it -- it usually takes more time to uncompress and copy the files than to compile the boost files. 

This will require a lot of testing but It seems like a good thing to try. 

grep NUMCPU scripts/* gives this  (lines to change)

```
common-build-dependencies.sh:  make -j"$NUMCPU"
common-build-dependencies.sh:  make -j$NUMCPU
common-build-dependencies.sh:  DESTDIR="$DEPLOYDIR" make -j$NUMCPU
common-build-dependencies.sh:  make -j$NUMCPU
common-build-dependencies.sh:  make -j$NUMCPU
common-build-dependencies.sh:  make -j$NUMCPU
common-build-dependencies.sh:  make -j$NUMCPU
common-build-dependencies.sh:  make -j$NUMCPU
macosx-build-dependencies.sh:  make -j"$NUMCPU" install
macosx-build-dependencies.sh:  make -j"$NUMCPU" 
macosx-build-dependencies.sh:  make -j"$NUMCPU" install
macosx-build-dependencies.sh:  make -j"$NUMCPU" install
macosx-build-dependencies.sh:  make -j"$NUMCPU" install
macosx-build-dependencies.sh:  ./b2 -j"$NUMCPU" -d+2 $BOOST_TOOLSET cflags="-mmacosx-version-min=$MAC_OSX_VERSION_MIN -arch x86_64 $CXXSTDFLAGS" linkflags="-mmacosx-version-min=$MAC_OSX_VERSION_MIN -arch x86_64  $LDSTDFLAGS -headerpad_max_install_names" install
macosx-build-dependencies.sh:  make -j"$NUMCPU" install
macosx-build-dependencies.sh:  make -j"$NUMCPU" install
macosx-build-dependencies.sh:  make -j"$NUMCPU"
macosx-build-dependencies.sh:  make -j$NUMCPU
macosx-build-dependencies.sh:  make -j$NUMCPU
macosx-build-dependencies.sh:  make -j$NUMCPU
macosx-build-dependencies.sh:  make -j$NUMCPU
macosx-build-dependencies.sh:  make -j$NUMCPU
macosx-build-dependencies.sh:  make -j$NUMCPU
macosx-build-dependencies.sh:  make -j$NUMCPU
macosx-build-dependencies.sh:if [ ! $NUMCPU ]; then
macosx-build-dependencies.sh:  NUMCPU=$(sysctl -n hw.ncpu)
macosx-build-dependencies.sh:  echo "Setting number of CPUs to $NUMCPU"
mingw-x-build-dependencies.sh:if [ ! $NUMCPU ]; then
mingw-x-build-dependencies.sh:	echo "note: you can 'export NUMCPU=x' for multi-core compiles (x=number)";
mingw-x-build-dependencies.sh:	NUMCPU=1
mingw-x-build-dependencies.sh:	if [ $NUMCPU -gt 2 ]; then
mingw-x-build-dependencies.sh:		NUMJOBS=$((NUMCPU/2))
mingw-x-build-dependencies.sh:echo make $PACKAGES MXE_TARGETS=$MXE_TARGETS -j $NUMCPU JOBS=$NUMJOBS
mingw-x-build-dependencies.sh:make $PACKAGES MXE_TARGETS=$MXE_TARGETS -j $NUMCPU JOBS=$NUMJOBS
publish-macosx.sh:export NUMCPU=$(sysctl -n hw.ncpu)
release-common.sh:if [ ! $NUMCPU ]; then
release-common.sh:  echo "note: you can 'export NUMCPU=x' for multi-core compiles (x=number)";
release-common.sh:  NUMCPU=1
release-common.sh:echo "NUMCPU: " $NUMCPU
release-common.sh:            make $TARGET -j$NUMCPU
release-common.sh:            make $TARGET -j$NUMCPU
release-common.sh:        make -j$NUMCPU $TARGET
release-common.sh:            make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  make -j"$NUMCPU" install
uni-build-dependencies.sh:  INSTALL_ROOT=$tmpinstalldir make -j"$NUMCPU" install
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:      $BJAMBIN -j$NUMCPU toolset=clang
uni-build-dependencies.sh:    $BJAMBIN -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  GLEW_DEST=$DEPLOYDIR $MAKER -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:#  make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:  make -j$NUMCPU
uni-build-dependencies.sh:#  make -j$NUMCPU
uni-build-dependencies.sh:if [ ! $NUMCPU ]; then
uni-build-dependencies.sh:  echo "Note: The NUMCPU environment variable can be set for parallel builds"
uni-build-dependencies.sh:  NUMCPU=1
uni-build-dependencies.sh:echo "Number of CPUs for parallel builds:" $NUMCPU
```

kintel, 2016-09-05:
I agree. Whoever want to play, please do this in a branch so we can spin it up on multiple platforms as well as letting @krichter722 test it as well before merging to master.