openscad/openscad #1755: openscad crashes with "0x000055555560fd11 in std::_Vector_base<Eigen::Matrix<int, 3, 1, 0, 3, 1>, Eigen::aligned_allocator_indirection<Eigen::Matrix<int, 3, 1, 0, 3, 1> > >::_Vector_impl::_Vector_impl"

krichter722, 2016-07-25:
`openscad` crashes at start when the file below is opened with backtrace

    Thread 1 "openscad" received signal SIGSEGV, Segmentation fault.
    0x000055555560fd11 in std::_Vector_base<Eigen::Matrix<int, 3, 1, 0, 3, 1>, Eigen::aligned_allocator_indirection<Eigen::Matrix<int, 3, 1, 0, 3, 1> > >::_Vector_impl::_Vector_impl
        (__a=..., this=<optimized out>) at /usr/include/c++/5/bits/stl_vector.h:91
    warning: Source file is more recent than executable.
    91		: _Tp_alloc_type(__a), _M_start(), _M_finish(), _M_end_of_storage()
    (gdb) bt
    #0  0x000055555560fd11 in PolysetUtils::tessellate_faces(PolySet const&, PolySet&) (__a=..., this=<optimized out>) at /usr/include/c++/5/bits/stl_vector.h:91
    #1  0x000055555560fd11 in PolysetUtils::tessellate_faces(PolySet const&, PolySet&) (__a=..., this=<optimized out>) at /usr/include/c++/5/bits/stl_vector.h:128
    #2  0x000055555560fd11 in PolysetUtils::tessellate_faces(PolySet const&, PolySet&) (__a=..., this=<optimized out>) at /usr/include/c++/5/bits/stl_vector.h:265
    #3  0x000055555560fd11 in PolysetUtils::tessellate_faces(PolySet const&, PolySet&) (a=..., this=<optimized out>) at /usr/include/eigen3/Eigen/src/StlSupport/StdVector.h:75
    #4  0x000055555560fd11 in PolysetUtils::tessellate_faces(PolySet const&, PolySet&) (a=..., this=<optimized out>) at src/linalg.h:14
    #5  0x000055555560fd11 in PolysetUtils::tessellate_faces(PolySet const&, PolySet&) (inps=..., outps=...) at src/polyset-utils.cc:87
    #6  0x00005555556f3210 in evaluate_csg_term_from_geometry(State const&, std::vector<boost::shared_ptr<CSGTerm>, std::allocator<boost::shared_ptr<CSGTerm> > >&, std::vector<boost::shared_ptr<CSGTerm>, std::allocator<boost::shared_ptr<CSGTerm> > >&, boost::shared_ptr<Geometry const> const&, ModuleInstantiation const*, AbstractNode const&) (state=..., highlights=std::vector of length 0, capacity 0, background=std::vector of length 0, capacity 0, geom=..., modinst=0x555556a385f0, node=...) at src/CSGTermEvaluator.cc:117
    #7  0x00005555556f4ea2 in CSGTermEvaluator::visit(State&, AbstractPolyNode const&) (this=0x7fffffffada0, state=..., node=...) at src/CSGTermEvaluator.cc:142
    #8  0x0000555555667b66 in Traverser::traverse(AbstractNode const&, State const&) (this=this@entry=0x7fffffffab80, node=..., state=...) at src/traverser.cc:41
    #9  0x0000555555667a9f in Traverser::traverse(AbstractNode const&, State const&) (this=this@entry=0x7fffffffab80, node=..., state=...) at src/traverser.cc:30
    #10 0x0000555555667a9f in Traverser::traverse(AbstractNode const&, State const&) (this=this@entry=0x7fffffffab80, node=..., state=...) at src/traverser.cc:30
    #11 0x0000555555667a9f in Traverser::traverse(AbstractNode const&, State const&) (this=this@entry=0x7fffffffab80, node=..., state=...) at src/traverser.cc:30
    #12 0x0000555555667cd9 in Traverser::execute() (this=this@entry=0x7fffffffab80) at src/traverser.cc:11
    #13 0x00005555556f37a7 in CSGTermEvaluator::evaluateCSGTerm(AbstractNode const&, std::vector<boost::shared_ptr<CSGTerm>, std::allocator<boost::shared_ptr<CSGTerm> > >&, std::vector<boost::shared_ptr<CSGTerm>, std::allocator<boost::shared_ptr<CSGTerm> > >&) (this=this@entry=0x7fffffffada0, node=..., highlights=std::vector of length 0, capacity 0, background=std::vector of length 0, capacity 0) at src/CSGTermEvaluator.cc:37
    #14 0x000055555572cca3 in MainWindow::compileCSG(bool) (this=this@entry=0x555555f7d0e0, procevents=procevents@entry=true) at src/mainwin.cc:1112
    #15 0x000055555572dec8 in MainWindow::csgReloadRender() (this=0x555555f7d0e0) at src/mainwin.cc:1712
    #16 0x00005555558da27b in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=_o@entry=0x555555f7d0e0, _c=_c@entry=QMetaObject::InvokeMetaMethod, _id=_id@entry=48, _a=_a@entry=0x7fffffffb5a0) at objects/moc_MainWindow.cpp:513
    #17 0x00007ffff466eaea in QMetaMethod::invoke(QObject*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) const (this=this@entry=0x7fffffffb788, object=object@entry=0x555555f7d0e0, connectionType=Qt::DirectConnection, 
        connectionType@entry=4294948880, returnValue=..., val0=..., val1=..., val2=..., val3=..., val4=..., val5=..., val6=..., val7=..., val8=..., val9=...)
        at kernel/qmetaobject.cpp:2212
    #18 0x00007ffff46741ad in QMetaObject::invokeMethod(QObject*, char const*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) (obj=obj@entry=0x555555f7d0e0, member=0x55555592d63b "csgReloadRender", type=4294948880, type@entry=Qt::AutoConnection, ret=..., val0=..., val1=..., val2=..., val3=..., val4=..., val5=..., val6=..., val7=..., val8=..., val9=...)
        at kernel/qmetaobject.cpp:1479
    #19 0x000055555572ca85 in MainWindow::compileDone(bool) (val9=..., val8=..., val7=..., val6=..., val5=..., val4=..., val3=..., val2=..., val1=..., val0=..., member=<optimized out>, obj=0x555555f7d0e0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs.h:407
    #20 0x000055555572ca85 in MainWindow::compileDone(bool) (this=this@entry=0x555555f7d0e0, didchange=<optimized out>) at src/mainwin.cc:1011
    #21 0x0000555555736925 in MainWindow::compile(bool, bool) (this=0x555555f7d0e0, reload=reload@entry=true, forcedone=forcedone@entry=false) at src/mainwin.cc:940
    #22 0x0000555555736c53 in MainWindow::actionReloadRenderPreview() (this=<optimized out>) at src/mainwin.cc:1707
    #23 0x0000555555736e45 in MainWindow::checkAutoReload() (this=<optimized out>) at src/mainwin.cc:1665
    #24 0x00005555558da4dd in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=_o@entry=0x555555f7d0e0, _c=_c@entry=QMetaObject::InvokeMetaMethod, _id=_id@entry=103, _a=_a@entry=0x7fffffffbd70) at objects/moc_MainWindow.cpp:568
    #25 0x00007ffff4691fca in QMetaObject::activate(QObject*, int, int, void**) (sender=sender@entry=0x555556c22720, signalOffset=<optimized out>, local_signal_index=local_signal_index@entry=0, argv=argv@entry=0x0) at kernel/qobject.cpp:3713
    #26 0x00007ffff46927d7 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) (sender=sender@entry=0x555556c22720, m=m@entry=0x7ffff48ac840 <QTimer::staticMetaObject>, local_signal_index=local_signal_index@entry=0, argv=argv@entry=0x0) at kernel/qobject.cpp:3578
    #27 0x00007ffff47116d0 in QTimer::timeout(QTimer::QPrivateSignal) (this=this@entry=0x555556c22720) at .moc/moc_qtimer.cpp:197
    #28 0x00007ffff469e878 in QTimer::timerEvent(QTimerEvent*) (this=0x555556c22720, e=<optimized out>) at kernel/qtimer.cpp:247
    #29 0x00007ffff4692e53 in QObject::event(QEvent*) (this=0x555556c22720, e=<optimized out>) at kernel/qobject.cpp:1261
    #30 0x00007ffff4f5605c in QApplicationPrivate::notify_helper(QObject*, QEvent*) (this=this@entry=
        0x555555cd4cf0, receiver=receiver@entry=0x555556c22720, e=e@entry=0x7fffffffc0a0) at kernel/qapplication.cpp:3716
    #31 0x00007ffff4f5b516 in QApplication::notify(QObject*, QEvent*) (this=0x7fffffffc4c0, receiver=0x555556c22720, e=0x7fffffffc0a0) at kernel/qapplication.cpp:3499
    #32 0x00007ffff466362b in QCoreApplication::notifyInternal(QObject*, QEvent*) (this=0x7fffffffc4c0, receiver=0x555556c22720, event=event@entry=0x7fffffffc0a0)
        at kernel/qcoreapplication.cpp:965
    #33 0x00007ffff46b889d in QTimerInfoList::activateTimers() (event=0x7fffffffc0a0, receiver=<optimized out>)
        at ../../include/QtCore/../../src/corelib/kernel/qcoreapplication.h:224
    #34 0x00007ffff46b889d in QTimerInfoList::activateTimers() (this=0x555555e14c60) at kernel/qtimerinfo_unix.cpp:637
    #35 0x00007ffff46b8da1 in timerSourceDispatch(GSource*, GSourceFunc, gpointer) (source=<optimized out>) at kernel/qeventdispatcher_glib.cpp:177
    #36 0x00007ffff61971a7 in g_main_context_dispatch (context=0x7fffe00016f0) at /build/glib2.0-7IO_Yw/glib2.0-2.48.1/./glib/gmain.c:3154
    #37 0x00007ffff61971a7 in g_main_context_dispatch (context=context@entry=0x7fffe00016f0) at /build/glib2.0-7IO_Yw/glib2.0-2.48.1/./glib/gmain.c:3769
    ---Type <return> to continue, or q <return> to quit---
[... cut ...]
            polyhedron(
                points=[[0,0,0],
                    [3*cooling_space,0,0],
                    [3*cooling_space,wall_thickness,0],
                    [0,wall_thickness,0],
                    [3*cooling_space,0,0],
                    [3*cooling_space,wall_thickness,0]
                    ],
                faces=[[0,1,2,3],
            [5,4,1,2],[0,3,5,4],[0,1,4],[2,3,5]
            ]);
        }
    };

    transformer_stand(5, 2, 45);

experienced with 2015.03-1 on Ubuntu 16.04

t-paul, 2016-07-25:
Does not crash with nightly build, please check with 2015.03-2 once that's available.

Ping @hyperair ... is it possible to update the PPA to 2015.03-2?

MichaelAtOz, 2016-07-25:
Note that it has mis-oriented polygons.

hyperair, 2016-08-01:
@t-paul yeah will do. Looks like I uploaded 2015.03-2 sometime back and forgot to deal with a build failure. https://launchpad.net/~openscad/+archive/ubuntu/releases/+build/9890857

kintel, 2016-08-17:
I'm pretty sure this is a duplicate of #1325 - fixed in 2015.03-2

kintel, 2016-08-29:
closing as duplicate of #1325

closed 2016-08-29 by kintel