openscad/openscad #1756: "qmake openscad.pro && make" fails due to "/usr/include/boost/fusion/container/vector/detail/preprocessed/vector10.hpp:23:9: error: ‘BOOST_CXX14_CONSTEXPR’ does not name a type"

krichter722, 2016-07-25:
Running `qmake openscad.pro && make CXXFLAGS="-I/usr/include/libxml2/ -I/usr/include/glib-2.0/ -I/usr/lib/x86_64-linux-gnu/glib-2.0/include/ -I/usr/include/harfbuzz/ -I/usr/include/freetype2/ -I/usr/include/boost/ -std=gnu++11"` fails due to

    /usr/include/boost/fusion/container/vector/detail/preprocessed/vector10.hpp:23:9: error: ‘BOOST_CXX14_CONSTEXPR’ does not name a type

Adding `CXXFLAGS` above is necessary to avoid errors because of headers which aren't found.

`./scripts/check-dependencies.sh`  reveals trouble, but no suggestion for a solution

    $ ./scripts/check-dependencies.sh
    depname     minimum     found       OKness      
    qt          4.4         5.5.1       OK          
    qscintilla2 2.7         2.9.1       OK          
    cgal        3.6         4.7         OK          
    gmp         5.0         5.0.5       OK          
    mpfr        3.0         3.1.1       OK          
    boost       1.35        1.56        OK          
    opencsg     1.3.2       1.4.        OK          
    glew        1.5.4       1.7.0       OK          
    eigen       3.0         3.2.2       OK          
    glib2       2.0         2.48.1      OK          
    fontconfig  2.10        2.11.       OK          
    freetype2   2.4         2.6.1       OK          
    harfbuzz    0.9.19      1.0.1       OK          
    bison       2.4         3.0.4       OK          
    flex        2.5.35      2.6.0       OK          
    make        3           4.1         OK          
    Warning: you have pkgconfig under /usr/local/lib
    Please verify these local copies don't conflict with the system

experienced with openscad-2015.03-1-584-g1c4c931

t-paul, 2016-07-25:
This looks more like a system issue as those path seem pretty normal. What system is that and is there any qmake warnings?

Also please use some sensible issue titles.

t-paul, 2016-07-25:
Actually, looking at the warning from `check-dependencies.sh`, do you maybe point `pkgconfig` to `/usr/local` so it does not find the system installed libraries anymore?

donbright, 2016-09-04:
is this some kind of BSD system?

closed 2019-01-25 by t-paul