openscad/openscad #1757: For loop ranges with > 10000 elements

reverendbedford, 2016-07-26:
I note in control.cc that for loop ranges are hardcoded to be limited to less than 10000 elements.  I have a use case that requires a longer range than that.  I'd be happy to contribute a commit upping or removing the upper limit, but I'm wondering if this will affect something else.  What's the reason for the hard coded upper limit?

t-paul, 2016-07-26:
This is mainly a safe-guard against generating huge object trees that may never finish rendering anyway. Having that fixed hard-coded value is not a good solution obviously, so idea how to improve there are welcome.
What's the use case that needs that amount of elements?

doug-moen, 2016-07-26:
@t-paul scooped me, but I thought the issue is that script evaluation occurs in the user interface thread. So if you iterate for, say, a billion times, or whatever, then you can freeze the GUI for minutes or hours. There's no way to interrupt script evaluation, you have to forcibly kill OpenSCAD. (You can avoid freezing the GUI by executing an `echo` statement frequently enough, but that also really slows down evaluation.)

So I think that this restriction on for loops is an attempt to help save people from shooting themselves in the foot. It's not a very effective restriction, because you can still freeze the GUI using nested loops, or by using `minkowski` on concave shapes.

@t-paul: One use case for high iteration loops is searching a large parameter space, and generating a shape each time the search succeeds. So you wouldn't generate a shape on each iteration. A more likely use case is creating a complex object with lots of detail. Some OpenSCAD users have powerful workstations with vast quantities of memory, so they can build more complex objects.

@reverendbedford You can work around this restriction using nested loops.

My suggestion is to remove the loop restriction, and open a separate issue to fix script evaluation so that it occurs in a separate thread, and is interruptible.

reverendbedford, 2016-07-26:
@t-paul 

> What's the use case that needs that amount of elements?

Filling some solid structure with lightweighting support material.  Think the Eifel tower.  For simple solids and a nice periodic lightweighting frame it's easy to fill in starting with a single unit cell and tiling, but for more complicated structures I've had to do the generation with an outside program and import all the elements of the frame structure.  The process actually works fairly well for a 9999 element example problem, though rendering does take awhile.  I'm hoping to try a test case with 40000 elements.

@doug-moen 
I'll work around it then by splitting up the loop into 2 nested fors.  Just wondering if there's some fundamental reason for the 10000 limit, as it would be easy to remove.