openscad/openscad #1760: Parameters: Range steps as floats

stylesuxx, 2016-07-26:
thingiverse allows steps in a range to be floats, like so [0:0.1:100]. I think it would be great to have that too.

I am willing to contribute, just let me know if this is a welcomed change.

amarjeetkapoor1, 2016-07-26:
Before getting your feedback on Blog. I also thought about this today  and also tried to find how to do this but was unable to find any in-build method in QT that would support it. so, I  was not able to find any direct method but we can implement this may be by using some hack.

Thanks for feedback.

stylesuxx, 2016-07-26:
I see, well that is unfortunate. So one way around would be to multiply min and max by 10^n, n being the places after comma, and dividing back again for display purposes.

Not very beautiful, still I think it is worth being supported.

amarjeetkapoor1, 2016-07-28:
@stylesuxx @t-paul 
I did it and it is working with a=34.003 //[12:0.2:40]  with following output [http://imgur.com/a/AHgA6](http://imgur.com/a/AHgA6)

but with a=34.003 //[12:40]   with following output [http://imgur.com/a/SbSB8](http://imgur.com/a/SbSB8)

stylesuxx, 2016-07-28:
Looking good - thank you.

From the screenshots I would assume you are rounding the values to step size? Are those changes pushed already? Would really like to try it out :-)

amarjeetkapoor1, 2016-07-28:
I am using the same method the one you described. and I have pushed the code you can check https://github.com/openscad/openscad/tree/gsoc2016-refactored

stylesuxx, 2016-07-28:
Hmm, When I set 0.2 as step size, I can interestingly also set values that are not multiples of 0.2 [image] This only happens when dragging the indicator, not while scrolling with the wheel, does QT not respect the step size when dragging the indicator?

amarjeetkapoor1, 2016-07-28:
I also encounter this problem earlier. I think QT itself doesn't respect the step size when dragging the indicator . but we can do it forceful but that would not be good hack.

stylesuxx, 2016-07-29:
Hmm, I could also imagine simply adjusting the scale like so:

    0-1: 2 steps with stepping width of 1, min: 0, max: 1, step 1:
    0: 1 * 0
    1: 1 * 1
    
    0-1: 3 steps with stepping width of 0.5 min: 0 max: 2, step 1:
    0.0: 0.5 * 0
    0.5: 0.5 * 1
    1.0: 0.5 * 2

Not sure if this explanation is clear, basically I would adjust the min and max values so the step size is always 1, but min and max values are adjusted to the user provided scale.

amarjeetkapoor1, 2016-07-29:
Basically, I read this solution but don't  know whether to do apply use it or not. I will ask the mentor and I its okay I will do it. And thanks for finding the solution.

amarjeetkapoor1, 2016-07-31:
@stylesuxx Just want to know from user point of view. What should be result of following conditions.
a=0.25; // [0:0.5:2]
What should be steps of slider
0, 0.5, 1,1.5,2 
or 
0.25 , 0.75 ,1.25,1.75

amarjeetkapoor1, 2016-07-31:
I checked it with Thingiverse customizer and its using first way.

closed 2016-08-04 by amarjeetkapoor1