openscad/openscad #1761: Parameters: Slider updates although still being dragged

stylesuxx, 2016-07-26:
I encounter this problem on my notebook when working with the touchpad. It is very hard to pin a number with the slider when in auto preview mode. I have the feeling that the slider is updating when it registered no movement for some time. I think it would be better to only update when the slider is not being dragged any more.

I am willing to contribute, just let me know if this is a welcomed change.

amarjeetkapoor1, 2016-07-26:
Thanks @stylesuxx for feedback and I have solved this issue.

amarjeetkapoor1, 2016-07-26:
@t-paul , Should same thing also be applied to spin-box and vector input widget?

t-paul, 2016-07-26:
Yes, if that makes the UI more responsive, I think that would be a good idea.

closed 2016-08-04 by amarjeetkapoor1