openscad/openscad #1763: Preview not rendering correctly

bond4u, 2016-08-03:
It seems to be hardware issue.
It was working fine on my old Dell Latitude, but it isn't rendering now on my new Dell Precision.
[_dell_latitude_lib_info.txt](https://github.com/openscad/openscad/files/399921/_dell_latitude_lib_info.txt)
[_dell_precision_lib_info3.txt](https://github.com/openscad/openscad/files/399922/_dell_precision_lib_info3.txt)
[image]
[image]

MichaelAtOz, 2016-08-04:
For others, note the above .txt files is the Help/Library_info.

As it is 'GL Renderer: Intel(R) HD Graphics 4600' on Windows, try Edit/Preferences/Advanced/Enable_Goldfeather

bond4u, 2016-08-04:
Got it working.
I had latest AMD driver, but not Intel driver.
support.dell.com driver detection offered new Intel Graphics driver.
After installing that OpenSCAD started crashing, but at the same time I got tooltip in systray saying that I need to assign a processor to application.
Once I did that, OpenSCAD started working and preview is rendering correctly now.

t-paul, 2016-08-04:
Thanks for the follow-up, I'll close the issue as that seems to be resolved with the updated driver.

closed 2016-08-04 by t-paul