openscad/openscad #1764: Floating point comparison problem

ChristianECooper, 2016-08-04:
In simple terms:
```
in = [several floating point values];
mid = (min(in) + max(in)) / 2;
out_less = [for (v=in) if (v < mid) v];
out_equal = [for (v=in) if (v == mid) v];
out_more = [for (v=in) if (v > mid) v];
// len(out_more) == len(in) !
```

In more detailed terms...

I've had a problem with recursion not exiting when sorting 3D coordinate vectors using the typical functional recursion approach to sorting (I,.e. pivot/filter hi-equal-lo/concatenate).

The points are being generated from trig functions, so I'm not sure of their exact internal values, all I'm seeing is the truncated output after I converted the functions to modules and used echo to dump their internal states.  I'll see if I can generate some code to build small set of coordinates to show the problem.  

What I can say with certainty though is that the when the values are scaled by 1/3 the code works fine, but when scaled by 2/3 it fails.

Here's the relevant debugging output:

```
ECHO: "Depth: 5, pivot:[-1.33333, -0.866025, 2.75]"
ECHO: "Depth: 5, intermediate -  len:12, coords: [[-1.33333, -1.73205, 0], [-1.33333, 0, 0.5], [-1.33333, -1.73205, 1], [-1.33333, 0, 1.5], [-1.33333, -1.73205, 2], [-1.33333, 0, 2.5], [-1.33333, -1.73205, 3], [-1.33333, 0, 3.5], [-1.33333, -1.73205, 4], [-1.33333, 0, 4.5], [-1.33333, -1.73205, 5], [-1.33333, 0, 5.5]], path: [0, 0, 1, 1, 1]"
ECHO: "Depth: 5, broken x/y: true/false, split 1: 0/0/12, split 2: 6/0/6, split 3: 6/0/6"
```

The first line indicates the pivot coordinate being used to separate the vector of coordinates.

The second line shows you the 12 coordinates being sorted at this level of the recursion.  As you can see they all _appear_ to have the same x coordinate, but varying y and z coordinates.

The third line shows the aberrant behaviour, after separation using x , y & z all values are greater than the pivot, but skipping x comparison and sorting on just y & z splits into six less than and six greater than, with none equal.

As you'll see from the code below, the x value of the pivot is calculated as the mid point of the min/max of the x values in the prior recursion, so there are only limited possible outcomes when comparing each of the coordinates against the pivot:

1. There are multiple x values so the pivot is a mid point and one of the following can happen: 
  * 1 or more are less than,  1 or more are equal to, none greater than, (unlikely edge case due to fp errors)
  * 1 or more are less than,  none or more are equal to, 1 or more greater than, (typical)
  * none are less than,  1 or more are equal to, 1 or more greater than, (another unlikely edge case)
2. There is a single x value so the mid point is that value and: 
  * 12 equal to.

But it would appear that I am getting all 12 greater than!

Here's the code I've written:

```
// Compare axis values (Returns: a < b => -1, a == b => 0, a > b => 1)
function cmp_axis(a, b) = 
    (a < b) 
        ? -1 
        : (a > b) 
            ? 1 
            : 0;

// Compare 3D coordinates (Returns: a < b => -1, a == b => 0, a > b => 1)
function cmp_coords(a, b, skip_x=false, skip_y=false) = 
    let(
        x = skip_x ? 0 : cmp_axis(a[0], b[0]),
        y = skip_y ? 0 : x != 0 ? x : cmp_axis(a[1], b[1]),
        z = y != 0 ? y : cmp_axis(a[2], b[2])
    ) z;

// Sort coordinates by X then Y then Z axes
module sort_coords(coords) {
    _sort_coords(coords, _pivot_coords(coords), 0, []);
}

// Indicates an invalid separation result
function is_pivot_broken(lt, eq, gt) = 
    (lt == 0 && eq == 0 && gt > 1) || (lt > 1 && eq == 0 && gt == 0);

module _sort_coords(coords, pivot, depth, path) {
    if (depth < 20) {
        echo(str("Depth: ", depth, ", pivot:", pivot));
        if (len(coords) < 2) {
            echo(str("Depth: ", depth, ", end state - len:", len(coords), ", coords ", coords, ", path: ", path));
        } else {
            echo(str("Depth: ", depth, ", intermediate -  len:", len(coords), ", coords: ", coords, ", path: ", path));
            ll_xyz = [for (c=coords) if (cmp_coords(c, pivot) == -1) c];
            eq_xyz = [for (c=coords) if (cmp_coords(c, pivot) == 0) c];
            gl_xyz = [for (c=coords) if (cmp_coords(c, pivot) == 1) c];
            
            xyz_broken = is_pivot_broken(len(ll_xyz), len(eq_xyz), len(gl_xyz));
            ll_yz = !xyz_broken ? ll_xyz : [for (c=coords) if (cmp_coords(c, pivot, skip_x=true) == -1) c];
            eq_yz = !xyz_broken ? eq_xyz : [for (c=coords) if (cmp_coords(c, pivot, skip_x=true) == 0) c];
            gl_yz = !xyz_broken ? gl_xyz : [for (c=coords) if (cmp_coords(c, pivot, skip_x=true) == 1) c];
            
            yz_broken = xyz_broken && is_pivot_broken(len(ll_yz), len(eq_yz), len(gl_yz));
            ll = !yz_broken ? ll_yz : [for (c=coords) if (cmp_coords(c, pivot, skip_x=true, skip_y=true) == -1) c];
            eq = !yz_broken ? eq_yz : [for (c=coords) if (cmp_coords(c, pivot, skip_x=true, skip_y=true) == 0) c];
            gl = !yz_broken ? gl_yz : [for (c=coords) if (cmp_coords(c, pivot, skip_x=true, skip_y=true) == 1) c];
            
            lp = _pivot_coords(ll);
            gp = _pivot_coords(gl);
            echo("Depth: ", str(depth, ", broken x/y:", xyz_broken, "/", yz_broken, ", split 1: ", len(ll_xyz), "/", len(eq_xyz), "/", len(gl_xyz), ", split 2: ", len(ll_yz), "/", len(eq_yz), "/", len(gl_yz), ", split 3: ", len(ll), "/", len(eq), "/", len(gl)));
            _sort_coords(ll, lp, depth + 1, concat(path, [0]));
            _sort_coords(gl, gp, depth + 1, concat(path, [1]));
        } 
    } else {
        echo("Aborted");
    }
}

// Find the mid point of a volume containing all the supplied coordinates
function _pivot_coords(coords) = 
    let (
        x = [for (c=coords) c[0]],
        y = [for (c=coords) c[1]],
        z = [for (c=coords) c[2]]) 
    [(min(x) + max(x)) / 2, (min(y) + max(y)) / 2, (min(z) + max(z)) / 2];
```

MichaelAtOz, 2016-08-04:
>  I'll see if I can generate some code to build small set of coordinates to show the problem. 

I think you will need to do so.

> As you can see they all appear to have the same x coordinate, but varying y and z coordinates.

Note that echo() does not output accurate (ie full precision) floats, so visual comparison is not helpful.

Without understanding all your code, with floats from trig, you are very unlikely to get anything equal to your calculated mid value.

MichaelAtOz, 2016-08-04:
&
> System information: Windows 7 SP1 (v6.1.1.0) 16 CPUs 63.90 GB RAM
 :eyes:

ghpenguin, 2016-08-05:
Comparison operators against floating point numbers produce unpredictable behaviors. 
I learned from both Python and OpenSCAD. 

Below is a Python function that finds a coordinate in 3D space.  I convert all floating point numbers to integers set by precision value. 
'''
Duplicate vertices 
prec = 5: 
 [20.645694, -15.000026, 15.771953], //  7 2-1
 [20.645725, -14.999988, 15.771950], //  8 2-2

'''
def find_vertex1(target, vsrc, prec=5):
        Mx = 10 ** prec
        x = int(Mx * target.x)
        y = int(Mx * target.y)
        z = int(Mx * target.z)

        size = len(vsrc)
        for i in range(size):
                v = vsrc[i]
                x_ = int(Mx * v.x)
                y_ = int(Mx * v.y)
                z_ = int(Mx * v.z)
                if x == x_ and y == y_ and z == z_:
                        return i
        return -1

You can use above technique in OpenSCAD also.
OpenSCAD provides two integer functions:
echo(ceil(4.4),ceil(-4.4));     // produces ECHO: 5, -4
echo(floor(4.4),floor(-4.4));    // produces ECHO: 4, -5

t-paul, 2016-08-06:
@ChristianECooper please add a concise example for the issue specific to OpenSCAD. In general we can't solve problems inherent to how floating point numbers work.

ChristianECooper, 2016-08-10:
OK, here's the code as concise as I could make it.  (Sorry, lots of library collation went into this...)

Calling test(true) will cause the failure, calling test(false) will draw a simple object.

As you can see the only difference is between scaling sine/cosine values and not scaling them.  These values are later used as scale factors.
```
// Get the length of a vector or return -1 for scalars
function safe_len(vec) = len(vec) != undef ? len(vec) : -1;

// Get a value at an index or return the default scalar value if vec is scalar or the index is invalid
function either_or(vec, i, def=undef) = 
    let(l = len(vec))
    l != undef && i < l 
        ? vec[i] 
        : def;

// Return a vector with all vec1 then vec2 elements
function extend(vec1, vec2) = 
    let (
        l = len(vec1),
        ll = len(vec1) + len(vec2))
    [for (i=[0:ll-1]) i < l ? vec1[i] : vec2[i-l]];

// Sum using bisection rather than accumulation to
// reduce depth of recursion for long summations 
// from n to log base 2 of n
function sum(vec) = _sum(vec, 0, len(vec)-1);
function _sum(vec, i, j) =
        i == j
            ? vec[i]
            : let(m = floor((i + j) / 2))
              _sum(vec, i, m) + _sum(vec, m + 1, j);
    
// Convert cartesian coordinates to column vectors
function cart2_2_cv(pt) = [[pt[0]], [pt[1]], [0]];
function cart3_2_cv(pt) = [[pt[0]], [pt[1]], [pt[2]]];

// Convert column vectors to cartesian coordinates
function cv_2_cart2(cv) = [cv[0][0], cv[1][0]];
function cv_2_cart3(cv) = [cv[0][0], cv[1][0], cv[2][0]];

// Basic rotation matrices
function Rx(theta) = [
    [cos(theta), -sin(theta), 0], 
    [sin(theta), cos(theta),  0], 
    [0,          0,           1]
];
function Ry(theta) = [
    [cos(theta),  0, sin(theta)], 
    [0,           1, 0], 
    [-sin(theta), 0, cos(theta)]
];
function Rz(theta) = [
    [1, 0,          0], 
    [0, cos(theta), -sin(theta)], 
    [0, sin(theta), cos(theta)]
];
// Create the combined rotation matrix
function R(rv) = Rz(rv[0]) * Ry(rv[1]) * Rx(rv[2]);

// Rotate an individual point
function rotate_point(point_rv, rotation_rv) =
    cv_2_cart3(R(rotation_rv) * cart3_2_cv(point_rv));

// Compare a single axis value (Returns: a < b => -1, a == b => 0, a > b => 1)
function cmp_axis(a, b) = 
    (a < b) 
        ? -1 
        : (a > b) 
            ? 1 
[... 276 lines cut ...]
    // Displacement - Simple +ive z offset
    displacements = [for (i=[0:steps]) (1/2) * [0, 0, i]];
    
    // Rotations - None
    rotations = undef;

    // Build the cyclical scalings
    cycles = 12;
    ratio = fail 
        ? 2/3 
        : 1;
    scalings = [for (i=[0:steps]) 
                    [2 + cos(ang * i * cycles) * ratio, 
                     2 + sin(ang * i * cycles) * ratio]
               ];

    // Build
    extruderly(
        points, 
        displacements, 
        rotations, 
        scalings);
}
test(true);
```

ChristianECooper, 2016-08-10:
@ghpenguin Thanks for that, I had already tried quantisation of the values using pretty much the same technique (see the function `quantizeVectorPoints` (and its matrix sibling) above.)  But that didn't seem to solve the issue.

@MichaelAtOz I do a lot of number crunching of large data sets, so cores and memory make my life a bit easier! :)

ChristianECooper, 2016-08-10:
@MichaelAtOz The loss of precision with echo statements has been a real issue for me when tracking this down, there are few tools that would make life easier sorting this out:

1. A config option to display all precision
1. A config option to dump a stack trace when hitting the recursion limit (and optionally the state for each stack layer.)
1. An extension to echo or maybe a `trace` function, so that it could be used with functions or inside functions.  I've no idea how this would work, but maybe making echo either return undef, or perform a pass through of any children might be a good start.

I'm a Java/Python(/OpenSCAD too I suppose) guy, with some very rusty C++ skills, but if I can find the time I'd love to try and help.  even if it is just tidying up the standard libraries and documenting them.

doug-moen, 2016-08-10:
>    2. A config option to dump a stack trace when hitting the recursion
>    3. An extension to echo or maybe a trace function, so that it could be

Chris Cooper wrote:

   1. A config option to display all precision
   2. A config option to dump a stack trace when hitting the recursion
   limit (and optionally the state for each stack layer.)
   3. An extension to echo or maybe a trace function, so that it could be
   used with functions or inside functions. I've no idea how this would work,
   but maybe making echo either return undef, or perform a pass through of any
   children might be a good start.

I've done preliminary work on all three issues.

1. Printing a floating point number exactly is more difficult than it
sounds. The goal is to print the number correctly and exactly, so that when
you read it back in (eg, paste the literal back into OpenSCAD source code),
you get back the exact same number. There's no way to do this using the C
or C++ standard libraries, so programming language implementations (Python,
Javascript, etc) use third party library code. There are two choices.
Python (and other older languages) use David Gay's dtoa.c: this is a
nightmare to work with, and also it's slow. I got it working, but I don't
think we want this code in OpenSCAD, because of the maintenance issues it
would create. Google created a new, fast implementation for their
Javascript interpreter, and packaged it as the double-conversion library,
which is available on github, and which is a standard package on Ubuntu.
After playing with this, I created a wrapper with a more convenient
interface for use in OpenSCAD. This works great on Ubuntu. What I haven't
investigated is where to get the double-conversion library for MacOS and
Windows.

2. I wrote up a proposal of how to implement a stack trace for OpenSCAD,
but I haven't actually implemented it. See:
https://github.com/openscad/openscad/issues/381#issuecomment-187260087

3. There are open issues for an echo that works inside functions. Here's an
old one: https://github.com/openscad/openscad/issues/741 There's a more
recent discussion that I can't find where we propose to add the following
expression syntax: `echo(echoargs) expr`. The semantics are to echo the
echoargs, then evaluate the expr and return its result.

ChristianECooper, 2016-08-24:
BTW, is that enough structure for you to reproduce the problem?

nophead, 2016-08-24:
I don't think this is a comparison problem. I think the logic is flawed when you have floating point numbers that only differ in the bottom bit. 

Suppose all the X values are the same except for one that is one LSB less. The average of the min and max could be equal to the smallest one due to truncation with all the others one LSB greater. If the Y value of the one with the lower X value happened to be greater than the pivot Y that makes that one greater than the pivot as well. So all would then be greater than the pivot and none equal or less than it.

So I think the simple version can definitely fail.  I haven't thought through the more complex version.

ChristianECooper, 2016-08-24:
@nophead I'm not sure I follow, am I missing something subtle?

Concentrating only on the X coordinate values, say we have two unique values for X across all supplied coordinates.  X<sub>0</sub> and X<sub>1</sub>, where X<sub>0</sub> + LSB = X<sub>1</sub>.   I.e. the values cannot be represented as being any closer together in FP arithmetic without being the same value. 

If we then calculate a pivot X<sub>p</sub> as a mean of all X values, this will result in a value that is quantised to either X<sub>0</sub> or X<sub>1</sub>.  This being dependent upon the multiplicity of each of the two X values, any truncation, and that there is no number between them that can be represented.

So given that X<sub>p</sub> can only be one of the two values, a comparison for each coordinate against the X<sub>p</sub> must result in at least one value being equal and, all other values being either exclusively smaller/larger than X<sub>p</sub>.  

We cannot compare and find all values smaller or all values larger than X<sub>p</sub>, but that is what I am seeing. 

That's like saying: <sup>(X + X + X)</sup>/<sub>3</sub> < X ...

Which leaves me very confused.  (And missing the flexibility of LaTeX markup. :))

nophead, 2016-08-24:
Yes one X value could be equal and all the others  1 bit greater. But the coordinate with the equal X value could have a Y value greater than the pivot so when you do the coordinate comparison with the pivot all points could be greater than it.

This is why I think the simple version can fail. Your complex version should detect that and stop looking at X so I am not sure why that fails as eventually it should just be comparing Z and then your argument holds.

nophead, 2016-08-24:
The point where it goes wrong isn't in the sort it is in divided_faces calculation. It is index that causes the infinite recursion.

nophead, 2016-08-24:
In divide_face_with_mean() you calculate a mean_point and look for its index in unique_points. Why would the mean necessarily be equal to one of the points? I think index() recurses forever if it is not found.

doug-moen, 2016-08-24:
@ChristianECooper said: That's like saying: (X + X + X)/3 < X ...

Sure, that situation is possible with floating point arithmetic. Division by 3 is not exact: it returns an approximate result that can be less than the mathematical result.

For example, (.7+.7+.7)/3 == 0.6999999999999998

Thanks for narrowing the problem down to something simple I could recognize.

nophead, 2016-08-24:
I think this fixes the problem:
```
function index(vec, val) = _index(vec, val, 0, len(vec));
function _index(vec, val, lo, hi) = 
    let (
        mid = floor((lo + hi) / 2),
        val_mid = vec[mid],
        cmp = cmp_coords(val, val_mid))
    cmp < 0 && hi > lo
        ? _index(vec, val, lo, mid)
        : cmp > 0 && hi > lo
            ? _index(vec, val, mid + 1, hi)
            : mid;
```

ChristianECooper, 2016-08-25:
@doug-moen Sorry for the confusion, the remark at the end was my poorly worded and flippant over-simplification.

I'm aware of the problems of FPA, and that's what we are discussing, but in the remark at the end I was assuming a context of integers, maybe a more accurate flippancy would be:

![Chris' strange variery of formal logic](https://latex.codecogs.com/gif.latex?%5Cbegin%7Barray%7D%7Blcccl%7D%20%28d%2C%20n%2C%20x%20%5Cin%20%5Cmathbb%7BZ%7D%29%20%26%20%5Cexists%28d%20%3E%200%29%20%26%20%5Cfrac%7Bxn%20&plus;%20d%7D%7Bx%7D%20%3C%20n%20%26%20%5CRightarrow%20%26%20%5Ctext%7BUniverse%20is%20broken%7D%20%5C%5C%20%28d%2C%20n%2C%20x%20%5Cin%20%5Cmathbb%7BFP%7D%29%20%26%20%5Cexists%28d%20%3E%200%29%20%26%20%5Cfrac%7Bxn%20&plus;%20d%7D%7Bx%7D%20%3C%20n%20%26%20%5CRightarrow%20%26%20%5Ctext%7BBusiness%20as%20usual%7D%20%5Cend%7Barray%7D)

Apologies for inventing my own variant on logical notation, and a new number classification! :)

@nophead Yes, a call to `index` will recurse forever if the target value is not in the available values, but you'll note that `unique_points` is indirectly constructed from `all_points`, which itself is partially constructed from `mean_points`, so I would expect each mean coordinate to be already constructed and present in the `unique_points` list and thus indexable without a risk of infinite recursion.

Which comes back to the problem that `cmp_coords` is unable to compare two vectors containing three nominally identical FP values and find them equal, I tried your suggested change to `_index`, but that didn't resolve the problem unfortunately, the runaway recursion is fixed, but the wrong point is being found and the resultant model is broken.  If you render then look at the rear of the model, you'll see incorrect coordinates being repeatedly selected. :(

P.S. Are you the same nophead I occasionally conversed with in the world of RepRap several years ago?

nophead, 2016-08-25:
So the problem is not quite as described. Basically the mean_point calculated in divide_with_mean() is not in the list mean_points. I don't think there is anything wrong with floating point comparison. You are looking for an exact match which is notoriously problematical with FP. Is it possible you calculate mean_points with a different vertex order? I don't think FP addition is commutative to the last bit so that would explain why it is not always found.

Changing index to find the nearest point seems to draw the right shape but isn't very efficient.

Yes I am the nophead that was part of the core team of RepRap.

ChristianECooper, 2016-08-25:
> I don't think FP addition is commutative to the last bit so that would explain why it is not always found.

Now that I hadn't thought about, I shall investigate further...

t-paul, 2019-02-19:
Closing as there's no progress for quite some time, discussion could continue on the mailing list if needed...

closed 2019-02-19 by t-paul