openscad/openscad #1765: echo() escaped character & HTML conflict

MichaelAtOz, 2016-08-04:
1000 words;
[image]
Some `escaped characters` (newline in this case, not tab) prevent `HTML` from being displayed.
Conversely, lines 11 &12, `HTML` prevents `escaped tab` from being displayed.

I haven't tried other escape sequences.

In OpenSCAD 2015.3-2 and 2016.07.20 on Win7.

t-paul, 2016-09-11:
Someone on IRC asked about this issue, so I guess it would make sense to also define what the expectation is. I never really expected the console to render HTML. It might be useful in some cases but is both risky in the long run (do we maybe want to support loading scripts directly from the web?) and sort-of conflicts with command line usage where HTML is not directly usable.

As mid-term solution, maybe split console (= for messages generated by the app) and output (= messages generated by the scripts)?

doug-moen, 2016-09-11:
My expectation is that I can use echo for debugging, and for printing the values of variables, including string variables. This means that echo should print strings literally, and not interpret them as HTML. The current behaviour should be considered a bug.

If we want to support HTML output to the console, then we should create a new function for that. The small number of people who want this will also want the ability to print to the console without having "ECHO:" preprended to the output.

MichaelAtOz, 2016-09-11:
The current behavior (baring bug in this report) is not a bug, and is useful. It is built into Qt, so eliminating it would presumable be more work.
During debug it is very useful to format output.
I seem to recall someone having quite extensive use of html, I think it was runsun  [e.g. BOM - forum](http://forum.openscad.org/OpenSCAD-and-Woodworking-td13966.html)

MichaelAtOz, 2016-09-11:
> but is both risky in the long run (do we maybe want to support loading scripts directly from the web?) and sort-of conflicts with command line usage where HTML is not directly usable.

It is a [subset](http://doc.qt.io/qt-5/richtext-html-subset.html) of html

bulgakovk, 2016-09-13:
I noticed that If I use only one tab symbol - HTML still renderes correct but tab replaces by empty symbol.
[image]

HTML with two tabs still works nice but all tab symbols replace by only one space.
Is it problem with QT widget or something wrong with printing functions? Probably you can give me advice where I can see this code?

>Someone on IRC asked about this issue
It was me btw.

MichaelAtOz, 2016-09-14:
> Probably you can give me advice where I can see this code?

echo() does:
https://github.com/openscad/openscad/blob/02439847dc0a98cf7e6ba7095b64d521ab3f6553/src/control.cc#L261
	PRINTB("%s", msg.str());

https://github.com/openscad/openscad/blob/master/src/printutils.h
void PRINT(const std::string &msg);
#define PRINTB(_fmt, _arg) do { PRINT(str(boost::format(_fmt) % _arg)); } while (0)

PRINT basically does:
https://github.com/openscad/openscad/blob/master/src/printutils.cc
	fprintf(stderr, "%s\n", msg.c_str());

boost::format() doesn't seem to do much as it is %s (string), so I presume Qt does the escape chars as well as html.

bulgakovk, 2016-09-14:
I tried to use fprintf from printutils.cc with "\n<b>Hello</b>" argument and bug still happens (escaped character prevents HTML).
Does it mean that It's something wrong with Qt Widget?

MichaelAtOz, 2016-09-20:
> Does it mean that It's something wrong with Qt Widget?

Presumably, or GCC/Runtime.

bulgakovk, 2016-09-29:
I've solved the problem with **\n** and **\t** symbols. Can someone tell me if there are any other tags which I should add to my code? Thanks.

MichaelAtOz, 2016-09-29:
Wiki says

> To include a " character in a string literal, use \". To include a \ character in a string literal, use \\. The following escape sequences beginning with \ can be used within string literals:

    \" → "
    \\ → \
    \t → tab
    \n → newline
    \r → carriage return
    \u03a9 → Ω - see text() for further information on unicode characters

MichaelAtOz, 2016-10-01:
Partial work-around. Newline tab and bold (& presumably other html) work, but 'pre' has a blank line before/after - inconsistently (see after bold).

echo("<pre>String, \nnewline, \t\ttab, <b>bold</b></pre>");
echo("First<pre>Second</pre>Third");

ChrisVesters, 2017-06-03:
What are the opinions about this? I understad that trying to have full HTML support is difficult. It might have side-effects for people who do not try to write html, but do so by accident?

Should we have a separate echo function for formatted text? By just saying 'formatted'  text we can decide ourselves what we do and do not support.

echoFormatted()

kintel, 2017-06-03:
When designing this, perhaps think about ANSI escape codes (and whatever the DOS prompt equivalent is) for people running this in a shell.

t-paul, 2020-02-16:
HTML render in console is disabled now.

closed 2020-02-16 by t-paul