openscad/openscad #1766: OpenSCAD's Customizer related issues

amarjeetkapoor1, 2016-08-05:
In customizer we are getting problem of setting number of decimal places up to which Spinbox and slider should be able to represent the numbers.  To represent the number in Spinbox exactly like what user input in scad we need to find the precision of the number then use that to setDecimal() in Qt Spinbox. 

At present the code that I am using to find the number of digits after decimal is:

```
decimalPrecision =0;
    int beforeDecimal=0;
    long double diff, rn; //rn stands for real number
    unsigned long long intNumber, multi = 1;
    number=abs(number);
    while(1){
        rn=number/multi;
        intNumber=rn;
        if(intNumber==0){
            break;
        }
        beforeDecimal++;
        multi=multi*10;
    }
    multi=1;
    while(1)
    {
        rn = (number * multi);
        intNumber = rn;  //the fractional part will be truncated here
        diff = rn - intNumber;
        if(diff == 0 || decimalPrecision>15-beforeDecimal)
            break;
        multi = multi * 10;
        decimalPrecision++;
    }

```

or  this 

```decimalPrecision=0;
    int beforeDecimal=0;
    long double diff, rn; //rn stands for real number
    unsigned long long intNumber, multi = 1;
    number=abs(number);
    while(1){
        rn=number/multi;
        intNumber=rn;
        if(intNumber==0){
            break;
        }
        beforeDecimal++;
        multi=multi*10;
    }
    multi=1;
    while(1)
    {
        rn = (number * multi);
        intNumber = rn;  //the fractional part will be truncated here
        diff = rn - intNumber;
        if(diff == 0 || decimalPrecision>15-beforeDecimal)
            break;
       if(diff > 0.9 || decimalPrecision>15-beforeDecimal){
            break;
        }
        multi = multi * 10;
        decimalPrecision++;
    }

```

But none is perfect.

Gregwar, 2016-08-18:
Actually, what you want is not possible because of the way your computer represents the floating point numbers

For example:
```c
float number = 0.0001;
printf("%.15f\n", number);
// Outputs: 0.000099999997474
```

This is because the number is represented is with 0 and 1, thus there is some data loss when converting the number from base 10 to base 2, this is also why you can't compare floats to 0

I think trying to count the decimal places in a float/double number is not a good idea, you should try to fetch this information from another source, like the original string representation of the number if possible in this case

amarjeetkapoor1, 2016-08-18:
Yes, we can't do that. Just wanted to get a reasonable solution till we use our final solution that we figured out i.e. reading string and storing  precision in AST with the node storing number itself. (Basically same as you stated).

For time being, we used above stated solution in issue with max number of digits after decimal to six.

MichaelAtOz, 2016-11-10:
(based on 2016.10.05)

Using `/* [Global] */` without other tabs show no variables.
Wiki says `Also the parameters who are under no tab will be diplayed under TAB named “parameters”.`
Presumably that should happen with Globals too??

MichaelPFrey, 2017-11-12:
Your solution might not be perfect, but after trying to find my own approach at that problem: The solution is good enough.

I really want to encourage people to specify [min:stepping:max] for each and every variable.
Often for example, negative values break the design or (as discussed here) the default value might be 1.0mm and a reasonable stepping is 0.1mm. In any case, supplying the range and stepping, conveys a lot of context information about the design intention. A smart phone holder for e.g. might be to fragile to hold a tablet without further modifications.

"If in doubt" the user can off-course always simply overwrite the values in the scad files, well knowing that he is leaving the beaten path.

Before my UI redesign, there was the issue that it was either a slider or a spinbox.
Now that the slider always also has a spinbox, there is no drawback anymore to specify range and stepping.

Off-course, it would be still nice to have access to the native string so that we can guess the users intention, but getting the user to specify range and stepping is better in the long run.

kintel, 2018-02-28:
@MichaelPFrey Is this issue covered in #2316 (or can it be) ?

MichaelPFrey, 2018-02-28:
@kintel I have to ask: Is this still an issue? "back in the days" supplying min and max values (range) automatically created just a slider. Providing no range, created a doubleSpinBox.

The decision to supply or not supply therefor dictated the widget and thus user experience. You do not want to set your smart phone size for a smart phone holder via slider, so those types of parameters, tended to have no range supplied (even-tough a range would be useful to indicate design intention) to get a double spin box (and not a slider).

This is not the case anymore.

"Now a days", it always creates a double spin box. If you supply a range,  the spin box is limited to that range. If you supply a stepping, the stepping is applied to the spin box. Also, if you supply a range, the slider appears (with range and stepping matching the spin box). Depending on how some "see" the UI, some might even over look the slider and just see it as range indicator (which is also a cool side effect) and just enter their values directly into the double spin box.

Knowing now much more about the source base then before:
Retrieving the "native text" (e.g x = 1.00) like comment.cpp is retrieving the comments is tricky to debug, as the file would then be parsed multiple times (for the interpreter, to retrieve comments and to retrieve the "native" value).
Giving the task of gathering "the native text" to the main parser would only make sense if we get other benefits from that (e.g. code reformating and error messages) and when some is willing and able to do that and nobody objects to the increased memory usage.
Even then, it could be argued, that interpreting x=1, x=1,0 and x=1,00 as different precession is a crutch, while providing [min:step:max] is even useful for people that disable the customizer (off course: it could be argued that using comments is just an other crutch, but at least one, that is forward and backward compatible [while a native syntax to limit the range of variables would break compatibility]).

Maybe I am totally overlooking something obvious.
Raising that issue "back in the days" was absolutely valid, as there was no way to set the stepping of the spinbox directly, but now there is.

kintel, 2018-03-01:
I didn't really look into this, I just noticed that this one was hanging. Your argument is sound, so I'll close this.

closed 2018-03-01 by kintel