openscad/openscad #1768: Provide simple measure tool

dronus, 2016-08-11:
Please provide a simple tool where two vertex locations can be selected in the 3d view and their distance is shown. 
This is especially useful as a quick check if the result of complex calculations matches the expectations.

mon, 2017-01-21:
I've noticed `resize` works, and therefore must know object dimensions for x/y/z. Whilst not quite "pick two vertices" could it be worked into a `size` command that returns a 3 element dimension array for an object?

kintel, 2017-01-21:
`resize()` has to perform an implicit `render()`, which requires a round-trip into CGAL and may surprise the user with a huge performance penalty.
Due to this implicit rendering, we've been hesitant to add more functionality on top of  `resize` or `render`, at least until we have a better way of making it clear that this is an expensive operation, or until we can replace the current CSG library with something significantly more performant.

However, there might exist opportunities for calculating world-space picked locations by performing CSG operations in one dimension along a ray cast through the design.

MichaelAtOz, 2017-01-22:
Would a 2D sizer be more efficient? 
a. that would help with text(), which really is problematic not knowing the size.
b. could use projection to get the 2D bit you want (cut=true) or the max size on a plane (cut=false).

projection() is slow, is it CSG based?

kintel, 2017-01-22:
'projection' does few different things (CSG union, 2D union of all projected polygons), so it may be slow.

donbright, 2017-01-25:
projection was also slow because CGAL's projection was buggy and crashy and we had to do a bunch of bizarre workarounds. like projection into a really really thin box instead of projection into a plane. (not sure if that is still needed or if it even executed anymore)

kintel, 2017-01-25:
In terms of projection stability, there is this issue: https://github.com/openscad/openscad/issues/1130

TLC123, 2020-06-04:
Here's an idea how a crude measuring tool could work without actually render the geometry in cgal.
Simplified:

Imagine  one isometric off-screen camera looking straight down the the x axis.
A crude minimal bound along the axis can be determined by the difference of the closest pixel in the z-map (resultant from a off-screen preview pass) and that cameras position.
Now imagine  one looking from the other direction obtaining the max bound.  

That axis can then be any arbitrary vector. And that distance act as a limited precision virtual caliper.
the camera pair can maybe be previewed as 1x1 pixel, for a narrow point caliper or hires as two planes sliding together along the said axis.

Do this for x y z and get a bounding box. 

The method as tons of limitation and can, come to mind, never measure subpixel geometry.
Probably can't get measurements to use as values inside open-scad code but plausibly as a Gui measure tool that writes to the console?
Some fancy iterative refinement might be necessary even for low precision, limiting the usefulness.

Kind of worth the fun to think about.

timor-raiman, 2021-05-11:
This does it for an exported STL:
https://www.shapeways.com/forum/t/meshy-a-browser-based-tool-for-doing-measurements-and-post-processing.79571/

bmuessig, 2021-05-18:
Currently at 305 USD. Let's hope someone picks it up :wink:

jordanbrown0, 2023-11-18:
Can we somehow merge the bounties for this and #3638, and close out one of them?

t-paul, 2023-11-18:
I don't think that's possible, but as we may see a first version of the measurement feature hopefully soon, there's a good chance to close both.

gsohler, 2023-11-18:
What's missing for the measurement PR to merge?

Generally I see a discrepancy from what I see consider finished and what
others do ...
If there is, please communicate ...

pca006132, 2023-11-20:
Note that bountysource is not working now: https://github.com/bountysource/core/issues/1586

kintel, 2025-02-16:
Closing as duplicate of #3638

closed 2025-02-16 by kintel