openscad/openscad #1769: Provide antialias / multisampling settings for the 3d view

dronus, 2016-08-11:
While it can be enforced by some video card settings, for some cards it can't, so it would be cool if OpenSCAD's settings provide a mean to enable polygon antialiasing or multisampling in the 3d view. 

I do not have a complete picture, but it may be true that the screen space operations needed for preview display of CGAL operations do not allow for native multisampling, in this case rendering to a larger off-screen buffer could be used. This needs to be provided by OpenSCAD as it can't be enforced by video card settings.

t-paul, 2016-08-11:
There is indeed a comment that might be related at https://github.com/openscad/openscad/blob/master/src/openscad.cc#L700
```
// turn on anti-aliasing
disabled by clifford wolf: adds rendering artefacts with OpenCSG
```
[Updating the OpenGL render code](https://github.com/openscad/openscad/wiki/Project:-Improve-OpenGL-rendering) is a big topic on the wishlist, this could be reviewed in scope of that once someone is working on that.

dronus, 2016-08-12:
I thought that the "preview" mode uses depth buffer mechanisms for CSG bool operations. Most 3rd party scenegraph frameworks won't support those tricks properly I guess, so the OpenGL code could maybe not rely on this, or the preview feature would be dropped I fear.
On the other hand, I experience full CGAL computation even for preview quite often on newer versions. I will look into this.

t-paul, 2016-08-12:
CGAL render while doing preview would be caused by:
- `render()` (explicitely forces CGAL)
- `minkowski()` (not supported by OpenCSG)
- `hull()` (not supported by OpenCSG)

(That what I currently know about, we probably need to check if there are more triggers)

MichaelAtOz, 2016-08-13:
- resize()

dronus, 2016-08-13:
Yeah, that's it. My stupid. Preview works well on designs not using these features.

bebr42, 2017-09-30:
Maybe I don't understand the solution. 
But on my (Windows 7) machine the preview is quite coarse and the edges use no antialiasing at all. 
Would be nice to fix this.

nophead, 2017-09-30:
On my last computer with an Nvidia card I could turn on antialiasing for
OpenSCAD. Now I have an Intel graphics chip the only option seems to be to
disable it or leave it to the application to decide.

bebr42, 2017-10-01:
Exactly, same situation here, Lenovo Thinkpad, which delivers otherwise nice performance, even on larger external monitors resolutions.
So we want to be able to control the 3D settings from within openScad.

stefano2734, 2026-09-20:
After 9 years it can be solved. 
Related issues by openGL and openCSG here are solved.