openscad/openscad #1770: Assignment of Variables

amarjeetkapoor1, 2016-08-12:
OpenSCAD supports that we can re-assign the same variable with some different value anytime in SCAD file but It uses only the value that is assigned at last everywhere in rest of the file. So, What is use of supporting Re-Assignment of variable as its create confusion. 

```
a=12;
echo(a);
a=45;
```
Above code should output 12 instead of 45.
It should be either that we use value of variable till it is Re-Assign or we should give error when variable is Re-Assigned.

Plus there are several other abnormalities related to assignment like 

```
a=14;
b=12;
a=b+12;
echo(a);
```
and
```
a=14;
a=a+12;
echo(a);
```
both are gives undef value.

There are also problem with storing location of assignment in ASTnode like:
```
a=12;
a=14;
```
As at present we support re-assignment. So, the value of `a` =14 for whole program. but location of ASTnode remains 1. It is not updated when `a` is re-assigned.

doug-moen, 2016-08-12:
I agree this is a bug.

OpenSCAD is a declarative language, which means that by design, it is also a "single assignment" language: each variable can be assigned exactly once, within a given scope. So a program like
```
a=12;
a=14;
```
*should* be illegal.

However, there are two problems: a lack of error reporting, and the `include` mechanism.

One problem is that OpenSCAD does not, in most cases, give error messages for illegal programs. We give error messages for parse errors, because bison doesn't give us any alternative. Ditto for CGAL errors. For illegal programs like the one above, it's less code to not check for and report an error, so we just keep running. I think that, to reduce confusion, we should print a meaningful error message and abort in cases like this.

The other problem is that the `include` mechanism allows the script that contains `include <S>` statement to override variables imported from `S`. So, if `S` defines variable `a`, then you can write
```
include <S>
a=16;
```
and override the definition of `a` within the code imported from `S`. This is a feature that people use.

The problem is that the `include` statement is currently implemented by the lexical analyzer, and we can't currently distinguish an assignment to `a` imported from `S` from an overriding assignment to `a` in the current script. Hence we don't have enough information in the parse tree to be able to detect and report an error where there are two assignments to `a` in the same scope.

@kintel has a branch where he is fixing the parse tree nodes to keep track of the source file in which they originated. Once that is integrated into the main branch, we'll have the necessary data structure so that we can fix this bug and report an error for duplicate assignments.

t-paul, 2016-08-12:
This behavior is also (mis-)used for the `-D` command line parameter evaluation, so it's not that simple to fix.

amarjeetkapoor1, 2016-08-12:
At present I am more concerned with the past point in my post that on re-assignment location of assignment is not updated which is causing little problem in customizer.
At present if 

```a=12; //[1:30]
a=14;//[100,34,2,3]
```

 It will take parameters of first assignement and default value of second which according to me is wrong. 
So, to get solve it I have done some changes in AST :

At present 
```
assignment:
          TOK_ID '=' expr ';'
            {
                bool found = false;
                for (auto& iter : scope_stack.top()->assignments) {
                    if (iter.name == $1) {
                        iter.expr = shared_ptr<Expression>($3);
                        found = true;
                        break;
                    }
                }
                if (!found) {
                    scope_stack.top()->assignments.push_back(Assignment($1, shared_ptr<Expression>($3),LOC(@$)));
                }
                free($1);
                
            }
        ;
```

After Change:
```
 assignment:
           TOK_ID '=' expr ';'
             {
                 bool found = false;
                 for (auto& iter : scope_stack.top()->assignments) {
                     if (iter.name == $1) {
                         iter.expr = shared_ptr<Expression>($3);
                         iter.LoC(LOC(@$));
                         found = true;
                         break;
                     }
                 }
                 if (!found) {
                     scope_stack.top()->assignments.push_back(Assignment($1, shared_ptr<Expression>($3),LOC(@$)));
                 }
                 free($1);
                 
             }
         ;
```
 if above logic is correct with respect to rest of OpenSCAD then parameter of last assignment will be used.

doug-moen, 2016-08-12:
Assigning a variable twice in the same scope without getting an error message is a bug, not a feature that we need to maintain. The use cases that need to be supported are:
* overriding the value of a variable that was originally defined in an `include`d script
* overriding the value of a top-level variable using the `-D` command line option

In these cases, it will be most common for the original variable definition to have customizer annotations, and the override just provides a new value without changing the customizer annotation.

I'm concerned that your proposed code change will interfere with the desired behaviour, which is that the customizer annotation for a variable doesn't change when all we are doing is overriding its value.

amarjeetkapoor1, 2016-08-12:
True but what if user overrides value with some other data type or he provide parameters when he reassign variable.

I think I should look at What Thingiverse customizer would do for this case as this problem is only related to comment syntax and  there would be no problem with annotation syntax.

doug-moen, 2016-08-12:
I don't think that Thingiverse supports either `include` (for user-supplied files) or the command line `-D` flag when you customize an OpenSCAD script on their web site. I'm thinking that the issue you raised is primarily of importance for using thingiverse comments outside the context of their web site. Eg, you download a model from thingiverse, with parameter comments, then you include that model in another script, override a variable definition, then use the customizer GUI.

amarjeetkapoor1, 2016-08-13:
I checked with Thingiverse customizer what it does for situation like this is :

```
a=1;//[1:10]
a=3;//[1,3,5,7]
```
It creates two widgets for variable `a` which is illogical according to me.
So, we can't use approach used by Thingiverse.

@t-paul  and @doug-moen 
We have to choose how we want to handle situation like above.
We can either use parameters with first assignment or last.
What is you opinion how the situation like above should be handled?

doug-moen, 2016-08-13:
For
```
a=1;//[1:10]
a=3;//[1,3,5,7]
```
we should use the second parameter spec, and for
```
a=1;//[1:10]
a=3;
```
we should use the first parameter spec.

t-paul, 2016-08-14:
I think we should always just use the parameters (and comments) attached to the assignment actually used. We can't match the thingiverse behavior but as creating two entry fields does not make much sense, that should not matter much.

MichaelPFrey, 2018-10-27:
see also #2513 and #2520

toddInPortland, 2020-07-04:
Forgive my naivete. I know a great deal about procedural languages, but much less about functional ones. Scad seems excellent to me. In spite of the steep learning curve, I can get much closer to what I want than with sketchup. But I'm finding there are are aspects of openscad that prevent abstraction, but I think could be fixed and not break scad.

I've written several scad files to construct a poseable skeleton; e.g. a module for the head, torso, pelvis, upper arm, foot, etc. They all share a few global variables such as the the diameter of a bone, the size of a joint's socket and ball. And they have a number of variables specific to each. In an effort to abstract, it makes sense (to me) to have a single parameter.scad file to declare the shared and module-specific variables and included by each of the module files. I admit that putting everything in one file would accommodate the language, but as the complexity grows, this does not seem good. 

From a usability standpoint, I'd like to point openscad at one of the modules and work on a shoulder or a foot. Call this aproach #1
--- **param.scad** ---
```
Bone_Diameter = 7;
Socket_Thickness = 2.2;
Ball_Diameter = 6;
```
--- **foot.scad** ---
```
include <param.scad>;

module foot(length = 22, ballDiameter = Ball_Diameter, boneDiameter = Bone_Diameter, socketThickness = Socket_Thickness) {
    ball(ballDiameter);
    socket(ballDiameter, socketThickness);
...
}

foot();
```

So there are a couple of problems with this. By itself, footl.scad can be opened, previewed and rendered by openscad, but the customizer does not see the variables in param.scad.  I could redeclare the variables from param.scad, but that would violate the single assignment issue discussed above.

Approach #2, yet is clunky but works, is to move the usage to the param.scad file and leave foot.scad with no includes or variable assignments, and then use a labeled array to say which item I am rendering/previewing. Like this:
--- **param.scad** ---

```
Bone_Diameter = 7;
Socket_Thickness = 2.2;
Ball_Diameter = 6;

Member_Name = 9; // [0:head, 1:torso, 2:pelvis, 3:upper arm, 4:forearm, 5:thigh, 6:shin, 7:hand, 8:foot, 9:cup]
if (Member_Name == 8) {
    foot();
}
```
--- **foot.scad** ---
```
module foot(...) {
    ...
}
```
This allows me to use the customizer, but I have to switch between openscad windows to edit vs preview. And this makes foot.scad pretty much useless except for editing.

I think #1 is the correct solution, but it seems an error that the variables *included* in foot.scad are not considered by the customizer.

mcp292, 2023-01-19:
Should this issue be left open?

toddInPortland, 2023-01-19:
Perhaps not, if there is nothing else to add.

mcp292, 2023-01-19:
I wouldn't know. Came across this today and I get a warning during re-assignment, can use other variables in assignment, and can re-assign within for loops.

jordanbrown0, 2024-11-23:
The intent is that reassignment is allowed only in very limited cases (include, Customizer, -D).  The OP's case currently gives a warning.

Working as designed, and now gives a warning when the user violates the rules.

closed 2024-11-23 by jordanbrown0