openscad/openscad #1774: CGAL Assertion Error

tgdavies, 2016-08-15:
I'm using OpenSCAD to slice up an STL file exported from OnShape, and I get the following error:

Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-3/../libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-3/../libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
Geometries in cache: 13
Geometry cache size in bytes: 60417464
CGAL Polyhedrons in cache: 13
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 10 seconds
Rendering finished.

The design previews OK, although it looks wrong from underneath -- you don't see the top surfaces.

The original OnShape project is here: https://cad.onshape.com/documents/079e9a76e80ba8037e7e51f8/w/d2a87564af757cde2f857041/e/ac1a3458879636e583ef8217

The STL was exported at medium resolution.

My code is:

module base() {
    import ("/users/tdavies/Downloads/Part Studio 1-3.stl", convexity=10);
}

module zneg() {
    translate([0,0, -10]) { cube([200,200,20], true); }
}

module bottom() {
    intersection() {
        base();
        zneg();
    }
}

module top() {
    difference() {
        base();
        zneg();
    }
}
top();
translate([90,0,0]) { rotate([180,0,0]) {bottom();}}
[zippedstl.zip](https://github.com/openscad/openscad/files/419158/zippedstl.zip)

t-paul, 2016-08-15:
Did you check the [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_imported_STL_file_only_showing_up_with_F5_but_not_F6.3F) already?

tgdavies, 2016-08-15:
Thanks for the very quick reply Paul  -- I hadn't checked the FAQ.

I tried the MS repair tool, but it just gave me a .3mf file (which I assume OpenScad can't import?)
Meshlab crashes when I try that filter on my file :-(
I haven't tried blender yet...

tgdavies, 2016-08-15:
I converted the MS repaired .3mf to STL via https://cloud.materialise.com/, and then used their repair service on it. The preview image looked better (normal from underneath), but I still get the same error when rendering:

Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-3/../libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-3/../libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
Geometries in cache: 5
Geometry cache size in bytes: 4417528
CGAL Polyhedrons in cache: 7
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 6 seconds
Rendering finished.

I've attached the 'repaired' stl file
[zippedrepairedstl.zip](https://github.com/openscad/openscad/files/419209/zippedrepairedstl.zip)

kintel, 2016-08-17:
@tgdavies The MS repair tool doesn't typically repair the issues we're sensitive to. Meshlab does a much better job, but has some robustness issues. Did you try Meshlab on the repaired file? Perhaps you manage that without hitting their crash bug?

tgdavies, 2016-08-18:
I've solved this by improving my OnShape exporting usage.

Initially I had a design in OnShape which was a number of intersecting parts, exported as a single STL file.
I've now added a boolean operation which unions all the parts into one, and now the exported STL file is rendered happily by OpenSCAD.
Of course it would be nice if OpenSCAD just took anything one threw at it, but I think we can call this "user error".

closed 2016-08-18 by tgdavies