openscad/openscad #1775: travis says i broke the build

donbright, 2016-08-18:
clicking Build History on this site

https://travis-ci.org/openscad/openscad/builds

shows that the build broke after i submitted an update patch for scripts/uni-get-dependencies.sh

why did that break the build? i cannot find any usage of get-dependencies.sh in any other scripts, travis module, etc. i tested my code on 3 different machines before submitting it and it was fine. 

it looks like it's failing to connect to X for the offscreen buffer rendering, a failure in GLXBadRenderRequest. i guess theoretically that could be caused if a certain X11 dependency was not loaded.

kintel, 2016-08-18:
I think this is an issue with the Travis VM - it happens from time to time.
However, this time the error message is different; we can open the X11 display, but it cannot find GLX.
I've restarted the build to see if that changes anything.

kintel, 2016-08-18:
That didn't help. I'll try to see if something changed with Travis, or if I can make the build pass again by going back in history a bit.

t-paul, 2016-08-18:
Something seems to have changed a couple of days ago. Since then tests fail all the time, I could not find any reason yet, but it happens for both Ubuntu versions.

t-paul, 2016-08-18:
Working: Aug 7, 2016 
```
hostname: travis-worker-gce-org-prod2-1:cf67bc07-aeba-447a-9732-85853c4f291d
version: v2.3.0-5-gef7116f https://github.com/travis-ci/worker/tree/ef7116fac34a232b5d8ecfe0eb9d834b0af9705d
instance: testing-gce-3181e557-f6b4-42f4-8b24-ce3ec8d02ad9:travis-ci-mega-trusty-1445287562
startup: 23.444502277s
```

Broken: Aug 12, 2016
```
hostname: travis-worker-gce-org-prod5-7:0b708011-8d2d-4e59-9406-8ae57634e104
version: v2.3.1-34-g4ab376b https://github.com/travis-ci/worker/tree/4ab376ba587e12cb465474238c78e88e6600aca2
instance: testing-gce-40840a2e-9ce5-44d2-acc7-b477babe70b3:travis-ci-sugilite-trusty-1471290794
startup: 21.497843841s

```

kintel, 2016-08-18:
Since this fails on both trusty and precise, and the most obvious change is the worker upgrade from 2.3.0 to 2.3.1, I kind of suspect that there might be a missing pre-installed package related to GL drivers.
It sucks that it takes so much time to run a test. It might be possible to set up an isolated test project for just testing that glx works as expected, and use that to get packages right, or as a bug report to Travis.

donbright, 2016-08-19:
14.04 build showing GLX missing

https://s3.amazonaws.com/archive.travis-ci.org/jobs/151691712/log.txt

     Xlib:  extension "GLX" missing on display ":721".

Possibly relevant link, when Xvfb on Fedora/Redhat stopped working with GLX a few years ago:

     https://bugzilla.redhat.com/show_bug.cgi?id=904851

kintel, 2016-08-19:
Github integrations tend to ask for relatively wide permissions. I think github doesn't offer too much granularity.. If it helps deciding, I've already given the travis integration repo access to openscad.
It should be possible to do it the other way; push from github using a web hook without using the Travis integration.

kintel, 2016-08-20:
I started on testing this here: https://github.com/kintel/travis-glx
Haven't quite managed to reproduce the same problem, but it's a start..

kintel, 2016-08-21:
Update:

```ldd /usr/bin/glxinfo``` shows that ```/usr/lib/fglrx/libGL.so.1``` is loaded as a GL driver.
These are ATI drivers - somehow Travis now pre-installs ATI drivers.

The easy way out would be to manually remove them. I'll take a look at it.

commit 68af43ca (2016-08-21): Fix #1775: Remove fglrx drivers which were recently introduced to travis ci
commit c5557231 (2016-08-21): Merge pull request #1780 from openscad/issue1775

Fix #1775: Remove fglrx drivers
closed 2016-08-21 by kintel