openscad/openscad #1777: Crash on OSX when rendering 2 imported STL files

salamanders, 2016-08-19:
When I import a single STL file, it works.
When I import the same STL twice in the same model, it crashes on Render.
(Tried renaming the STL file and importing it, same crash)

OSX, version 2016.07.22 (git f3b3738) 

```
14  org.openscad.OpenSCAD         	0x00000001032f5b70 gui(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >&, boost::filesystem::path const&, int, char**) + 2816
15  org.openscad.OpenSCAD         	0x00000001032f887c main + 9308
16  libdyld.dylib                 	0x00007fff8ffb75ad start + 1

MichaelAtOz, 2016-08-19:
Any STL file or a specific one? Can you share it?

kintel, 2016-08-19:
@salamanders This doesn't crash for me, so I guess it's specific to your design or your system. Can you share your design so I can try to reproduce it?
Also, what version of OS X are you running?

salamanders, 2016-08-24:
I haven't gotten the OK from the model owner yet, so can't share it. :(   No reason to keep open an issue that I can't demonstrate to you, so closing as "can't repro" until I get the green light to share the model.

salamanders, 2016-08-30:
Hey I got permission.  file attached.  
[CoolBadgeV1.stl.zip](https://github.com/openscad/openscad/files/445745/CoolBadgeV1.stl.zip)

t-paul, 2016-08-30:
Probably same root cause as #1783. Importing this twice, one translated works fine on Linux with both 2015.03-1 and OpenSCAD version 2016.08.25.nightly (git 733562b).

salamanders, 2016-08-30:
How hard is it for me to build?

It gets all not-buildy when I tried it.

```
MYUSER-macbookpro2:openscad MYUSER$ make
/usr/local/Cellar/qt5/5.6.1-1/bin/uic src/MainWindow.ui -o objects/ui_MainWindow.h -tr _
/usr/local/Cellar/qt5/5.6.1-1/bin/uic src/Preferences.ui -o objects/ui_Preferences.h -tr _
/usr/local/Cellar/qt5/5.6.1-1/bin/uic src/OpenCSGWarningDialog.ui -o objects/ui_OpenCSGWarningDialog.h -tr _
/usr/local/Cellar/qt5/5.6.1-1/bin/uic src/AboutDialog.ui -o objects/ui_AboutDialog.h -tr _
/usr/local/Cellar/qt5/5.6.1-1/bin/uic src/FontListDialog.ui -o objects/ui_FontListDialog.h -tr _
/usr/local/Cellar/qt5/5.6.1-1/bin/uic src/ProgressWidget.ui -o objects/ui_ProgressWidget.h -tr _
/usr/local/Cellar/qt5/5.6.1-1/bin/uic src/launchingscreen.ui -o objects/ui_launchingscreen.h -tr _
/usr/local/Cellar/qt5/5.6.1-1/bin/uic src/LibraryInfoDialog.ui -o objects/ui_LibraryInfoDialog.h -tr _
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -I/usr/local/Cellar/glib/2.48.1/include/glib-2.0 -I/usr/local/Cellar/glib/2.48.1/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.39/include -I/Users/MYUSER/Documents/workspace/openscad/../libraries/install/include -I/Users/MYUSER/Documents/workspace/openscad/../libraries/install/include/libxml2 -I/usr/local/Cellar/harfbuzz/1.3.0/include/harfbuzz -I/usr/local/Cellar/glib/2.48.1/include/glib-2.0 -I/usr/local/Cellar/glib/2.48.1/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.39/include -I/usr/local/Cellar/freetype/2.6.5/include/freetype2 -I/Users/MYUSER/Documents/workspace/openscad/../libraries/install/include/fontconfig -O2 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -mmacosx-version-min=10.8 -Wall -W -Wno-overloaded-virtual -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-format-security -Wno-sign-compare -Wno-c++11-extensions -fPIC -DOPENSCAD_VERSION=2016.08.30 -DOPENSCAD_SHORTVERSION=2016.08.30 -DOPENSCAD_YEAR=2016.0 -DOPENSCAD_MONTH=08.0 -DOPENSCAD_DAY=30.0 -DDEBUG -DENABLE_MDI -DENABLE_CGAL -DENABLE_OPENCSG -DUSE_QOPENGLWIDGET -D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=0 -DUSE_SCINTILLA_EDITOR -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MACEXTRAS_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_CORE_LIB -I. -Isrc -I/usr/local/Cellar/qt5/5.6.1-1/include -Isrc/libtess2/Include -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtPrintSupport.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtWidgets.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtMacExtras.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtGui.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtConcurrent.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers -Iobjects -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/AGL.framework/Headers -I../libraries/install/include/eigen3 -I../libraries/install/include -I/usr/local/Cellar/qt5/5.6.1-1/mkspecs/macx-clang -F/usr/local/Cellar/qt5/5.6.1-1/lib -o objects/scintillaeditor.o src/scintillaeditor.cpp
In file included from src/scintillaeditor.cpp:5:
src/scintillaeditor.h:8:10: fatal error: 'Qsci/qsciscintilla.h' file not found
#include <Qsci/qsciscintilla.h>
         ^
1 error generated.
make: *** [objects/scintillaeditor.o] Error 1

kintel, 2016-08-31:
@salamanders You need to build dependencies first. In this case it looks like QScintilla isn't installed.
There are two primary ways of building dependencies:
1. Run ```./scripts/macosx-build-dependencies.sh```
2. Run ```./scripts/macosx-build-homebrew.sh``` (assuming Homebrew is installed)

If you did any of that and it failed, please report that separately and we'll get it fixed.

salamanders, 2016-08-31:
Ok homebrew compile failure reported issue #1784

salamanders, 2016-08-31:
Was able to compile latest on OSX.  And no more crash!

kintel, 2016-08-31:
Thanks for testing. This still concerns me, so I might pester you later to test some of mine binaries once I get some ideas what can be wrong.

kintel, 2016-11-05:
@salamanders Could you give this a spin? http://files.openscad.org/snapshots/OpenSCAD-2016.11.05.dmg

(only thing I did was upgrade OS to 10.11 and compiler to Xcode 8)

kintel, 2016-11-07:
Fixed by #1858 
OpenSCAD-2016.11.06 has this fix.

closed 2016-11-07 by kintel

salamanders, 2016-11-07:
@kintel worked like a charm.  Same fix in the public 2016.11.06 dl?

kintel, 2016-11-07:
The public download has a different fix, but should yield the same result. Testing is always appreciated!