openscad/openscad #1781: Openscad Customizer Improvements

amarjeetkapoor1, 2016-08-24:
This will contain list of improvements that need to be done before customizer would be merged with main branch.

Present suggestions:

- [ ] Replace SET with more descriptive key in JSON.
- [ ] Replace labels "+" and "-" with three separate labels ( someone found them misleading):
    1. New - for a new Preset.
    2. Update - to update the current Preset with the current parameter values.
    3. Delete - to delete the current Preset (after a confirmation).
- [ ] Alphabetical order of Parameter and Groups.
- [ ] Default values settings could be just another Set of Parameter, an always present one.

PRouzeau, 2016-08-25:
Here  some suggestions for improvement:
1/ The widgets shall be presented in the order defined by the programmer, not in alphabetical order, to let the programmer decide the logic.

2/ The variable name display shall be in option, as with a label, the user does not need to know the variable name 

3/ There shall be an option to have the label on the left of the widget, as is traditional in input screens, I propose the following syntax.
// label on top (existing)
//- Label on left
//+ label on right
//-R label on left, with widget aligned on right 
The latter will be useful for tick box or radio buttons as it allow more text. For other widgets, the width shall be adapted to the content.

4/ The border around each widget use a lot of space and it shall be removed (or set as an option). The label shall be closer to the widget and proper vertical space management will make the discrimination with less wasted space. This is even more important if the label is set on the left of the widget.

5/ It can be useful to group some variables within a border/frame if they belong to a ‘functional group’ . Such group may have an optional title. This might be the same role as tabs, but within the same screen view, so a lower order structure. Maybe use /* [[group]] */

6/ A variable named ‘part’ is often used to select printable part. It may got a special treatment, as being the first user selection. Notably it shall not be inside tabs, but above them (in the header ?). Maybe when there are tabs, all variables defined before first tab definition shall be above tab selectors ?

7/ General ergonomy : There is no real point to have the input font size much lower than the label. They are information of equal importance for usability, so they may be in same font size, or with very low difference.

8/ General ergonomy: The arrows of the spin widget are extremely small, a tiny target difficult to reach for people not very handy or people using trackballs. Even if it looks less elegant, two side by side square buttons of the height of the widget shall be used. If you use same font for input as for label, that will make reasonable target size.

9/ Above comments may lead to more user options, so a preference window may be created. These preferences shall be attached to the .scad FILE, not to the OpenScad program, so another option is to create a syntax to define customizer preferences inline.

[image] [image]  [image]

PRouzeau, 2016-08-25:
I think we also need 'free input'  number widget as you sometimes need to enter decimal values, which are not practical with spinbox.

kintel, 2016-08-25:
@PRouzeau A small note: The purpose of this issue is to discuss the fixes necessary before merging to master. To make this more clear, we should move any non-critical, cosmetic or future wishes into separate issues for each topic.

amarjeetkapoor1, 2016-08-25:
@kintel I told him that he can add all the issues and features here in form of list. So, that everything would be together.

PRouzeau, 2016-08-25:
Hello Marius,

I followed the recommendations done here : 
https://amarjeetkapoor1.wordpress.com/2016/08/21/20th-june-2015-2/#comment-62

so you suggest that I open a new issue ? Or what shall be the channel 
for feature requests ?

Regards, Pierre

t-paul, 2016-08-25:
What about collecting all topics here for now and just categorize for needed for merge / to do later in the description on top? Once the needed for merge stuff is done, we can just move the remaining topics to a new issue and close this one.

amarjeetkapoor1, 2016-08-25:
@t-paul , @kintel  we also need to categories the points in original issue itself.

kintel, 2016-08-25:
I agree that we should split this off into issues internally, at least until we have officially shipped this.

donbright, 2016-09-04:
I am trying to test this by building the "gsoc2016-refactored" branch but my "Edit/Preferences" menu doesn't show any "Features" tab.

amarjeetkapoor1, 2016-09-04:
@donbright It should show the Features tab if you are building it with.
``` qmake CONFIG+=experimental```

donbright, 2016-09-04:
Thanks very much.
I tried loading Candle stick and i am not sure if i did it right, is it supposed to look like this?

When I 'unhide' the customizer it shows this:

[image]

When i 'x' the customizer it shows this:

[image]

amarjeetkapoor1, 2016-09-04:
No, It shouldn't be like this. It should be more like second screen shot that you shared.

but Same thing also happens with me but I used to think it is  was due to some previewing problem in OpenSCAD as when I used to render it would render correctly but didn't preview correctly on making the customizer visible .

And this problem is not only with customizer only but also with other things like console, toolbar, editor.

donbright, 2016-09-04:
There is another issue that is a bit of a problem. I ran the customizer on the candle and played around with the settings. I get this error when i try to switch branches. 

```
don@serebryanya:~/src/openscad/binmaster$ git branch
  fbsdbuild
* gsoc2016-refactored
  master
don@serebryanya:~/src/openscad/binmaster$ git checkout master
error: Your local changes to the following files would be overwritten by checkout:
	examples/Parametric/candleStand.json
Please, commit your changes or stash them before you can switch branches.
Aborting

```

It looks like the act of testing the code results in changes to a textfile that is part of the git repository.

amarjeetkapoor1, 2016-09-04:
I think its not an issue as you would have got same if you had made changes to any .scad files given in examples. As when you played with Customiser you might have updated, added or deleted some set of parameters which are saved in stated JSON file. So, content of JSON file might have got changed.

donbright, 2016-09-04:
how is this going to work on a system where the user does not have administrative access and the example files are read only?

kintel, 2016-09-05:
@donbright Good point. We should avoid writing files to the examples folder. This folder should be considered (and probably will be) read-only on most OpenSCAD installations.

t-paul, 2016-09-05:
How is that different from the *.scad files? If the file is read-only, it will not be possible to change/overwrite it. We discussed that scenario and it should be implemented already (allow json file to open read-only), but I did not yet test that specifically yet. Maybe we need some additional indication in the parameter GUI, but otherwise I don't see how this is special.

amarjeetkapoor1, 2016-09-05:
If the JSON file is in read only mode then it can't be read at present. So, it couldn't be edited when installed. And I have added a feature to the list in starting of this issue that we would allow read only JSON files to be read but not modified.

As in case of @donbright He modified the examples which were part of source code but not installed version.

t-paul, 2016-09-05:
Whops, yes, you are right, we listed it as open issue in #1781.

TLC123, 2016-10-05:
Feature requests: conditionally show/hide parameters based on other parameter selections .

t-paul, 2016-10-05:
@TLC123 added to the list, but this is not possible with the thingiverse syntax, so it's out of scope for the current implementation.

donbright, 2016-10-06:
new windows builds available of gsoc2016-refactored branch with opencsg1.4.2

http://files.openscad.org/snapshots/OpenSCAD-2016.10.05-x86-32_gsoc2016-refactored-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2016.10.05-x86-32_gsoc2016-refactored.zip
http://files.openscad.org/snapshots/OpenSCAD-2016.10.05-x86-64_gsoc2016-refactored-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2016.10.05-x86-64_gsoc2016-refactored.zip

MichaelAtOz, 2016-10-06:
This'll be the first time I looked at this. 
Is there any doco somewhere?

t-paul, 2016-10-06:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/WIP#Customizer

Original docs for Thingiverse customizer at http://customizer.makerbot.com/docs

amarjeetkapoor1, 2016-10-08:
@kintel , Any suggestion related to two issues left for initial merge. For one related to ast file, I liked the suggestion by @t-paul that we can comment the native syntax code thats dumped in ast file.

```
 //@Parameter()
a=1;
```

Or any better that you and teepee have?

kintel, 2016-10-08:
@amarjeetkapoor1 Ideally it should dump in the same format as it was read.

amarjeetkapoor1, 2016-10-08:
Yes, but it would be difficult task to dump comments correctly.

t-paul, 2016-10-08:
Trying to regenerate the Thingiverse comment format is probably quite ugly and may get into the way once we start with a native format. I see how it makes sense to ensure the generated AST dump is be compilable, but as it's mainly used for debug/tests is it worth the effort?

If the actual annotation format looks strange, we could drop that and just write comments with the data without trying to regenerate the exact format (which is especially difficult for the group markers I suspect).

kintel, 2016-10-08:
The primary purpose of dumping AST files is for black box regression testing, as we could verify that "read-dump-read" gives the exactly same result.
If it's not feasible to output comments on the same format we read it, I suggest that we simply ignore any annotations when dumping AST files, and leave this as a future enhancement.

jon-bondy, 2016-10-30:
I just checked it out, and agree wholeheartedly with the comments that the Customizer presentation should be much more compact (about 1/2 the height) and that the user should have the option to present the parameters in the order in the file.  Wonderful new feature!

MichaelPFrey, 2018-10-28:
I am going to update the todo list. For documentation:
- Default values settings could be just another Set of Parameter, an always present one. -> #2479
- Look into reducing use of multiple inheritance -> #2330
- conditionally show/hide parameters based on other parameter selections -> #2359 so... that has to wait for a native syntax