openscad/openscad #1783: Crashes on opening scad file

salamanders, 2016-08-28:
OpenSCAD version 2016.08.21 (git c555723) 
Crashes on opening either .scad file in http://www.thingiverse.com/thing:82620
OSX 10.11.6

Release version OpenSCAD version 2015.03-3 does not crash.

t-paul, 2016-08-28:
Both work fine with OpenSCAD version 2016.08.25.nightly (git 733562b) on Linux.

salamanders, 2016-08-28:
I'm also getting a crash with the following hull command.  Possibly related?  Does it work in your nightly?
```
$fn =100;

trac_radius=100;
trac_width=20;
trac_thickness=3;
// Also = number of teeth!  
// http://www.robertcailliau.eu/Alphabetical/L/Lego/Gears/Rolling%20Diameters
num_teeth = trac_radius*2;
tooth_base_width    = 2.3;
tooth_tip_width     = 0.87;
brick_width         = 7.96;
tooth_height        = 2.06;

hull() {
translate([0,tooth_height,0])
cube([tooth_tip_width, 0.1,brick_width], center=true);
cube([tooth_base_width, 0.1,brick_width], center=true);
}

t-paul, 2016-08-28:
Yes, also no issue with that `hull()` on Linux. This might be some OSX specific issue, I don't see any relevant changes between 2016-08-21 and 2016-08-25.

salamanders, 2016-08-28:
Drat. Any part of the crash log that might help? Debug mode I could run?

kintel, 2016-08-29:
@salamanders I've tested on 10.10.5 and 10.11.5 without managing to reproduce the crash.
A crash log would be helpful.
As would the output of Help->Library Info

salamanders, 2016-08-29:
Crash log and Help-Library 
https://gist.github.com/salamanders/a9c05e0a0c33b9c9f5ad1af809c52ce4

kintel, 2016-08-29:
This looks like this same issue as this thread from the mailing list: http://forum.openscad.org/Cylinder-in-Hull-crashes-2016-06-25-Mac-OS-10-9-5-td17917.html

Apparently it was solved by building from source, so it could be a build system issue on my side.
I'd still like to reproduce this so I can figure out what went wrong and potentially detect that in the future.

salamanders, 2016-08-31:
I too solved it by building from source.   (Once I got around #1784 )

kintel, 2016-11-06:
Latest reports indicate this:
* Binaries built using Xcode 7 (on Mac OS X 10.10) are broken
* Binaries built using Xcode 8 (on Mac OS 10.11- that's the minimum required OS version) works
* OpenSCAD _executable_ built on Xcode 7 using _libraries_ built on Xcode 8 works, but _not_ vice versa

-> one or more of the libraries are broken on the Xcode7 build.

kintel, 2016-11-07:
Fixed by #1858 
OpenSCAD-2016.11.06 has this fix.

closed 2016-11-07 by kintel