openscad/openscad #1784: Compile on OSX homebrew not working "fatal error: 'Qsci/qsciscintilla.h' file not found"

salamanders, 2016-08-31:
```
MYUSER-macbookpro2:openscad MYUSER$ ./scripts/macosx-build-homebrew.sh
Already up-to-date.
Unlinking /usr/local/Cellar/boost/1.61.0_1... 0 symlinks removed
eigen 3.2.9
Warning: eigen-3.2.9 already installed
boost 1.61.0_1
Warning: boost-1.61.0_1 already installed, it's just not linked
cgal 4.8.1
Warning: cgal-4.8.1 already installed
glew 2.0.0
Warning: glew-2.0.0 already installed
glib 2.48.1
Warning: glib-2.48.1 already installed
opencsg 1.4.0_1
Warning: opencsg-1.4.0_1 already installed
freetype 2.6.5
Warning: freetype-2.6.5 already installed
libxml2 2.9.4
Warning: libxml2-2.9.4 already installed
fontconfig 2.12.1
Warning: fontconfig-2.12.1 already installed
harfbuzz 1.3.0
Warning: harfbuzz-1.3.0 already installed
qt5 5.6.1-1
Warning: qt5-5.6.1-1 already installed
qscintilla2-qt5 2.8.4
Warning: Calling Formula.sha1 is deprecated!
Use Formula.sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling SoftwareSpec#sha1 is deprecated!
Use SoftwareSpec#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling Resource#sha1 is deprecated!
Use Resource#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling BottleSpecification#sha1 is deprecated!
Use BottleSpecification#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:10:in `block in <class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: glehmann/extras/qscintilla2-qt5-2.8.4 already installed
Warning: Calling Formula.sha1 is deprecated!
Use Formula.sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling SoftwareSpec#sha1 is deprecated!
Use SoftwareSpec#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling Resource#sha1 is deprecated!
Use Resource#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling BottleSpecification#sha1 is deprecated!
Use BottleSpecification#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:10:in `block in <class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

imagemagick 6.9.5-8
Warning: imagemagick-6.9.5-8 already installed
Warning: Already linked: /usr/local/Cellar/gettext/0.19.8.1
To relink: brew unlink gettext && brew link gettext
Warning: Already linked: /usr/local/Cellar/qt5/5.6.1-1
To relink: brew unlink qt5 && brew link qt5
Warning: Calling Formula.sha1 is deprecated!
Use Formula.sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling SoftwareSpec#sha1 is deprecated!
Use SoftwareSpec#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling Resource#sha1 is deprecated!
Use Resource#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:6:in `<class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Calling BottleSpecification#sha1 is deprecated!
Use BottleSpecification#sha256 instead.
/usr/local/Library/Taps/glehmann/homebrew-extras/qscintilla2-qt5.rb:10:in `block in <class:Qscintilla2Qt5>'
Please report this to the glehmann/extras tap!

Warning: Already linked: /usr/local/Cellar/qscintilla2-qt5/2.8.4
To relink: brew unlink qscintilla2-qt5 && brew link qscintilla2-qt5
MYUSER-macbookpro2:openscad MYUSER$ source setenv_mac.sh
MYUSER-macbookpro2:openscad MYUSER$ qmake openscad.pro
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
Project MESSAGE: Using QOpenGLWidget
Project MESSAGE: Using libstdc++
Project WARNING: Disabling C++11 since libstdc++ dependencies were found
MYUSER-macbookpro2:openscad MYUSER$ make
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -I/usr/local/Cellar/glib/2.48.1/include/glib-2.0 -I/usr/local/Cellar/glib/2.48.1/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.39/include -I/Users/MYUSER/Documents/workspace/openscad/../libraries/install/include -I/Users/MYUSER/Documents/workspace/openscad/../libraries/install/include/libxml2 -I/usr/local/Cellar/harfbuzz/1.3.0/include/harfbuzz -I/usr/local/Cellar/glib/2.48.1/include/glib-2.0 -I/usr/local/Cellar/glib/2.48.1/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.39/include -I/usr/local/Cellar/freetype/2.6.5/include/freetype2 -I/Users/MYUSER/Documents/workspace/openscad/../libraries/install/include/fontconfig -O2 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -mmacosx-version-min=10.8 -Wall -W -Wno-overloaded-virtual -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-format-security -Wno-sign-compare -Wno-c++11-extensions -fPIC -DOPENSCAD_VERSION=2016.08.30 -DOPENSCAD_SHORTVERSION=2016.08.30 -DOPENSCAD_YEAR=2016.0 -DOPENSCAD_MONTH=08.0 -DOPENSCAD_DAY=30.0 -DDEBUG -DENABLE_MDI -DENABLE_CGAL -DENABLE_OPENCSG -DUSE_QOPENGLWIDGET -D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=0 -DUSE_SCINTILLA_EDITOR -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MACEXTRAS_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_CORE_LIB -I. -Isrc -I/usr/local/Cellar/qt5/5.6.1-1/include -Isrc/libtess2/Include -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtPrintSupport.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtWidgets.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtMacExtras.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtGui.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtConcurrent.framework/Headers -I/usr/local/Cellar/qt5/5.6.1-1/lib/QtCore.framework/Headers -Iobjects -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/AGL.framework/Headers -I../libraries/install/include/eigen3 -I../libraries/install/include -I/usr/local/Cellar/qt5/5.6.1-1/mkspecs/macx-clang -F/usr/local/Cellar/qt5/5.6.1-1/lib -o objects/scintillaeditor.o src/scintillaeditor.cpp
In file included from src/scintillaeditor.cpp:5:
src/scintillaeditor.h:8:10: fatal error: 'Qsci/qsciscintilla.h' file not found
#include <Qsci/qsciscintilla.h>
         ^
1 error generated.
make: *** [objects/scintillaeditor.o] Error 1

kintel, 2016-08-31:
You shouldn't use ```source setenv_mac.sh``` when building against homebrew. Could you try without that?

salamanders, 2016-08-31:
oh.  OHHH... those are three different sections?
In the home page it has "1. From source" then "1. homebrew" and "2. MacPorts"
Now I can see it meant options 1/2/3.

I'm getting strange boost link errors.
```
./scripts/macosx-build-homebrew.sh
Already up-to-date.
Unlinking /usr/local/Cellar/boost/1.61.0_1... 119 symlinks removed
eigen 3.2.9
Warning: eigen-3.2.9 already installed
boost 1.61.0_1
Warning: boost-1.61.0_1 already installed, it's just not linked
Error: You must `brew link boost` before cgal can be installed
```
as you can see, I did `brew link boost` but the script unlinked boost?

kintel, 2016-08-31:
Could you try to manually ```brew link boost```, remove the ```brew unlink boost``` from the script and then rerun the script?
I don't remember why boost was unlinked. The homebrew scripts isn't too well maintained..

salamanders, 2016-08-31:
Whohoo!  That fixed it, able to compile!

commit f82bb52b (2016-09-02): Use proper qscintilla2 package. Don't unlink boost. Fixes #1784
closed 2016-09-02 by kintel