openscad/openscad #1785: Unexpected Preview on making a sub window visible

amarjeetkapoor1, 2016-09-04:
I am getting unexpected preview on making a sub windows like  console, editor, toolbar visible in master branch and also customizer in #1751 . This issue is also mentioned in #1781.

For code you can refer  
https://github.com/openscad/openscad/blob/gsoc2016-refactored/examples/Parametric/candleStand.scad

[image]

donbright, 2016-09-04:
this happens to me in gsoc branch, haven't tested master branch. 

here is a video of the bug

https://www.youtube.com/watch?v=5K38T4vEiZk

donbright, 2016-09-04:
confirmed on master branch from today

https://www.youtube.com/watch?v=oeN2epGkBvE

amarjeetkapoor1, 2016-09-04:
Earlier, I used to think this problem was local to my system.

t-paul, 2016-09-04:
Seems to happen only with Qt5, so could be related to the new OpenGL widget.

donbright, 2016-09-04:
Minimal example

this will disappear if you hit 'view/hide console' 'view/unhide console'

```
intersection() {
    cube(15);
    sphere(10);
}

```

```
OpenSCAD Version: 2016.09.04
System information: Linux 4.4.0-36-generic #55-Ubuntu SMP Thu Aug 11 18:01:55 UTC 2016 x86_64 Ubuntu 16.04.1 LTS 4 CPUs 7.71 GB RAM
Compiler: GCC "5.4.0 20160609" 64bit
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.9.1
MingW build: No
GLib version: 2.48.1
Application Path: /home/don/src/openscad/binmaster
Documents Path: /home/don/.local/share
Resource Path: /home/don/src/openscad
User Library Path: /home/don/.local/share/OpenSCAD/libraries
User Config Path: /home/don/.config/OpenSCAD
Backup Path: /home/don/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/don/.local/share/OpenSCAD/libraries
/home/don/src/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.13.0
OpenGL Version: 3.0 Mesa 11.2.0
GL Renderer: Mesa DRI Intel(R) Broadwell GT1 
GL Vendor: Intel Open Source Technology Center
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QOpenGLWidget

```

donbright, 2016-09-04:
Also happens with OpenCSG 1.3.2 on Qt5.

donbright, 2016-09-04:
Ok I noticed a big difference here.

Bug ---- RGBA(8880), depth(0), stencil(0)
No bug - RGBA(8888), depth(24), stencil(8)

This makes sense because OpenCSG depends on depth + stencil buffer.

donbright, 2016-09-05:
i can also reproduce by just changing the window size. 

https://www.youtube.com/watch?v=CYO-nqs6oiQ

interesting that the 'axes' also disappear, not just the OpenCSG objects of 'intersection'

nairaner, 2016-09-08:
Same issue as #1727

nairaner, 2016-09-08:
Possibly related issue is double-click to center not working in qt5, because of no depth buffer.

donbright, 2016-09-09:
basically what we need to do is build the OpenCSG example included in the OpenCSG source code  against qt5 and see if it messes up. 

if yes, then we will know it is Qt5 itself and not OpenSCAD.

kintel, 2016-09-10:
...but why would the axes disappear? Could this be related to #1409 as well? Any chance you could build against OpenCSG <= 1.3.3 and see if that causes the behavior to change?

nairaner, 2016-09-10:
Compiled against OpenCSG 1.3.3, but this still happens

donbright, 2016-09-11:
FYI I tried running with software renderer using LIBGL_ALWAYS_SOFTWARE=1, same bug. Maybe it is not driver related (I assume QT is using the software renderer here). 

Then... I set debugging... like so: http://www.mesa3d.org/envvars.html

`LIBGL_ALWAYS_SOFTWARE=1 LIBGL_DEBUG=1 MESA_DEBUG=1 ./openscad 

Very interesting result. Under qt4, you get no debug output. 

Under qt5, you get this, just at the moments where the problems happen:

```
Mesa: User error: GL_INVALID_OPERATION in glBindFramebuffer(buffer)
Mesa: 62 similar GL_INVALID_OPERATION errors
Mesa: User error: GL_INVALID_OPERATION in glReadBuffer(invalid buffer GL_COLOR_ATTACHMENT0)

```

Here is video

https://www.youtube.com/watch?v=8-V0kR0HYDY

nairaner, 2016-09-11:
```
User error: GL_INVALID_OPERATION in glReadBuffer
```
This is probably because depth buffer is disabled or more likely not correctly enabled. 
According to similar issue http://stackoverflow.com/a/23464276 it is possible that in Qt4 DepthBuffer was enable by default so it worked correctly.

nairaner, 2016-09-11:
Another example 
```
intersection()
{
  sphere(30);
  cube(30);
}
```

t-paul, 2016-09-11:
With the following added to openscad.cc
```
  QSurfaceFormat format;
  format.setDepthBufferSize(24);
  format.setStencilBufferSize(8);
  QSurfaceFormat::setDefaultFormat(format);
```
I get
```
OpenGL Version: 3.0 Mesa 11.2.2
GL Renderer: Gallium 0.4 on NVE7
GL Vendor: nouveau
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QOpenGLWidget
```
But the issue is still there.

nairaner, 2016-09-11:
It might be a start, because it made glGetPixels in double-click function return value instead of an error code, but it is still broken.

nairaner, 2016-09-12:
So part of the reason why it happens might be that in qt5(on the right) framebuffers get overwritten by completed frame after the process is done
[image]

kintel, 2016-09-12:
@nairaner I suspect that the issue you're chasing is slightly different. You're trying to access the depth buffer _after_ a frame has been rendered in preview mode, right? This might be very dependent on details on how OpenCSG renders to the final framebuffer. If this is the case we should probably discuss this in a separate issue.

nairaner, 2016-09-12:
Using apitrace I diffed the working frame and bugged frame and all i got is 
[image]
and all error we get is related to glBindFramebuffer.
Inside the glBindFramebuffer we get GL_INVALID_ENUM in glGetIntegerv(pname=GL_TEXTURE_BINDING_BUFFER).

@kintel I guess you are right, double-click is a different issue

donbright, 2016-09-13:
Ok uhm there seems to be evidence this is some problem between OpenCSG Framebuffers interacting with Qt's framebuffers. There is a workaround to force OpenCSG to use PBuffers instead of Framebuffers, and this bug disappears.

This is posisble by adding a single line to src/mainwin.c  at about line 1206,
`
               OpenCSG::setOption(OpenCSG::OffscreenSetting, OpenCSG::PBuffer );
`

Im doing this directly before the call to 

`                this->opencsgRenderer = new OpenCSGRenderer(this->root_products, . . . blah blah blah
`

donbright, 2016-09-13:
Here is another bizarro patch that "fixes" the bug (but only if you hide console :) it wont work for fixing the "Real issue", i am a bit confused about where the actual render happens but anyways ...)

```
@@ -2491,6 +2494,7 @@ void MainWindow::showConsole()
 
 void MainWindow::hideConsole()
 {
+       OpenCSG::freeResources();
        if (viewActionHideConsole->isChecked()) {
                consoleDock->hide();
        } else {

```

Following documentation from http://opencsg.org/#context

t-paul, 2016-09-13:
Maybe @floriankirsch has a suggestion if using PBuffers instead of Framebuffers as shown by @donbright above is a good idea with Qt5 + new QOpenGLWidget.

kintel, 2016-09-13:
Thanks @donbright for digging!

Just to clarify one this: This bug is only known to affect Qt5 under X11 right?

I cannot reproduce it on the Mac build, but I can successfully reproduce it on a Linux VM running on Mac.

Based on the info so far it sounds a bit like this is related to OpenGL resource management. Perhaps we're allocating/deallocating resources bound to the wrong context somewhere..

t-paul, 2016-09-13:
I did not verify yet, but it might also be QOpenGLWidget vs. QGLWidget, I'll try what happens with QGLWidget on Qt5 when I have the chance...

Is macOS also using the new widget?

kintel, 2016-09-13:
Yep, we prefer QOpenGLWidget except under Windows:
```
!win*: CONFIG += qopenglwidget
qopenglwidget:!qglwidget:!lessThan(QT_VERSION, 5.4): CONFIG += using_qopenglwidget
```

floriankirsch, 2016-09-13:
As the author of OpenCSG, this is something I must reproduce in a test application independent of OpenSCAD. This will take a while, sorry...

The workaround of using the PBuffer option is fine if it works, but it cannot be generally recommanded
* PBuffer are not so widely supported. In OpenCSG, there is no MacOSX implementation, for example.
* With more recent graphics hardware - from NVidia, at least - frame buffer objects are now significantly faster

kintel, 2016-09-13:
As @donbright mentioned a while back, we could port the OpenCSG example to Qt for easier testing.
That port could possibly be contributed to OpenCSG as well.

t-paul, 2016-09-13:
The issue depends on the widget used. Qt5 + QGLWidget works.

floriankirsch, 2016-09-22:
This is indeed an OpenCSG issue. I will fix it with the next release (hopefully soon).

The QOpenGLWidget apparently decides to create a new FBO when it is resized, and then the bug happens: OpenCSG on return does not restore the new FBO that was active in the context of the caller, but the previous FBO. Then an obsolete or invalid FBO is active, and rendering does not work.

kintel, 2016-09-22:
@floriankirsch nice catch!

floriankirsch, 2016-09-29:
I have just released OpenCSG 1.4.2 with the fix.

donbright, 2016-09-30:
Seems OK on ubuntu 16

```
don@serebryanya:~/src/openscad/binmaster$ apitrace trace ./openscad
# play around with csg preview, hide console, unhide console
don@serebryanya:~/src/openscad/binmaster$ apitrace dump openscad.trace | grep Framebuffers
1631 glGenFramebuffers(n = 1, framebuffers = &1)
10707 glGenFramebuffers(n = 1, framebuffers = &22)
276868 glDeleteFramebuffers(n = 1, framebuffers = &1)
276870 glGenFramebuffers(n = 1, framebuffers = &41)
282696 glDeleteFramebuffers(n = 1, framebuffers = &22)
282697 glGenFramebuffers(n = 1, framebuffers = &42)
765726 glDeleteFramebuffers(n = 1, framebuffers = &41)
765728 glGenFramebuffers(n = 1, framebuffers = &62)
1580055 glDeleteFramebuffers(n = 1, framebuffers = &62)

```

donbright, 2016-10-07:
also works on windows (latest snapshot 10/04)

nairaner, 2016-10-08:
I also cant reproduce it on the OpenCSG 1.4.2

kintel, 2016-10-08:
thx all, let's consider this  fixed.

closed 2016-10-08 by kintel