openscad/openscad #1786: ./scripts/uni-build-dependencies.sh fails due to "Policy CMP0026 is not set: Disallow use of the LOCATION target property." on Ubuntu 16.04

krichter722, 2016-09-08:
Following https://github.com/openscad/openscad the `./scripts/uni-build-dependencies.sh` command fails due to `Policy CMP0026 is not set: Disallow use of the LOCATION target property.` for CGAL. This might be due to the version 3.5.1 on Ubuntu 16.04.

    > ./scripts/uni-build-dependencies.sh
    Entering auto-apt mode: ./scripts/uni-build-dependencies.sh
    Exit the command to leave auto-apt mode.
    BASEDIR: /home/richter/openscad_deps
    DEPLOYDIR: /home/richter/openscad_deps
    PATH modified
    LD_LIBRARY_PATH modified
    LD_RUN_PATH modified
    OPENSCAD_LIBRARIES modified
    GLEWDIR modified
    Please re-run qmake-qt4 and run 'make clean' if necessary
    Note: The NUMCPU environment variable can be set for parallel builds
    Using basedir: /home/richter/openscad_deps
    Using deploydir: /home/richter/openscad_deps
    Using srcdir: /home/richter/openscad_deps/src
    Number of CPUs for parallel builds: 1
    Eigen3 already installed. not building
    gmp already installed. not building
    mpfr already installed. not building
    boost already installed. not building
    Building CGAL 4.4 ...
    curl --insecure -O https://gforge.inria.fr/frs/download.php/file/33524/CGAL-4.4.tar.bz2
      % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                     Dload  Upload   Total   Spent    Left  Speed
    100 13.1M  100 13.1M    0     0   622k      0  0:00:21  0:00:21 --:--:-- 1139k
    -- The CXX compiler identification is GNU 5.4.0
    -- The C compiler identification is GNU 5.4.0
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    == Setting paths ==
    -- Build CGAL from release in directory CGAL-4.4
    -- Packagenames: CGAL-4.4
    == Setting paths (DONE) ==

    == Generate version files ==
    -- CGAL_MAJOR_VERSION=4
    -- CGAL_MINOR_VERSION=4
    -- CGAL_BUGFIX_VERSION=0
    -- CGAL_SONAME_VERSION=10
    -- CGAL_SOVERSION     =10.0.3
    -- CGAL_REFERENCE_CACHE_DIR=
    -- Building shared libraries
    -- Targetting Unix Makefiles
    -- Using /usr/bin/c++ compiler.
    -- USING CMake version: 3.5.1
    -- System: Linux
    -- USING GCC_VERSION = '5.4.0 20160609'
    == Generate version files (DONE) ==

    == Set up flags ==
    -- Build type: Debug
    -- USING CXXFLAGS = ' -g'
    -- USING LDFLAGS = ' '
    == Set up flags (DONE) ==

    == Detect external libraries ==
    -- External libraries supported: GMP;GMPXX;MPFR;ZLIB;OpenGL;LEDA;MPFI;RS;RS3;OpenNL;TAUCS;Eigen3;BLAS;LAPACK;QGLViewer;ESBTL;Coin3D;NTL;IPE
    -- Preconfiguring library: GMP ...
    -- GMP has been preconfigured:
    --   UseGMP-file:      
    --   GMP include:      /home/richter/openscad_deps/include
    --   GMP libraries:    /home/richter/openscad_deps/lib/libgmp.so
    --   GMP definitions:  
    -- USING GMP_VERSION = '5.0.5'
    -- Preconfiguring library: MPFR ...
    -- MPFR has been preconfigured:
    --   UseMPFR-file:      
    --   MPFR include:      /home/richter/openscad_deps/include
    --   MPFR libraries:    /home/richter/openscad_deps/lib/libmpfr.so
    --   MPFR definitions:  
    -- USING MPFR_VERSION = '3.1.1'
    -- Looking for pthread.h
    -- Looking for pthread.h - found
    -- Looking for pthread_create
    -- Looking for pthread_create - not found
    -- Looking for pthread_create in pthreads
    -- Looking for pthread_create in pthreads - not found
    -- Looking for pthread_create in pthread
    -- Looking for pthread_create in pthread - found
    -- Found Threads: TRUE  
    -- Boost version: 1.56.0
    -- Found the following Boost libraries:
    --   thread
    --   system
    -- Boost include:     /home/richter/openscad_deps/include
    -- Boost libraries:   /home/richter/openscad_deps/lib/libboost_thread.so;/home/richter/openscad_deps/lib/libboost_system.so;/usr/lib/x86_64-linux-gnu/libpthread.so
    -- Boost definitions: 
    -- USING BOOST_VERSION = '1.56.0'
    == Detect external libraries (DONE) ==

    == Write compiler_config.h ==
    -- Performing Test CGAL_CFG_BOOST_VARIANT_SWAP_BUG - Success
    -- Performing Test CGAL_CFG_DENORMALS_COMPILE_BUG - Success
    -- Performing Test CGAL_CFG_FPU_ROUNDING_MODE_UNWINDING_VC_BUG - Success
    -- Performing Test CGAL_CFG_IEEE_754_BUG - Success
    -- Performing Test CGAL_CFG_ISTREAM_INT_BUG - Success
    -- Performing Test CGAL_CFG_MATCHING_BUG_5 - Success
    -- Performing Test CGAL_CFG_MATCHING_BUG_6 - Success
    -- Performing Test CGAL_CFG_MATCHING_BUG_7 - Success
    -- Performing Test CGAL_CFG_MATCHING_BUG_8 - Success
    -- Performing Test CGAL_CFG_NESTED_CLASS_FRIEND_DECLARATION_BUG - Success
    -- Performing Test CGAL_CFG_NO_LIMITS - Success
    -- Performing Test CGAL_CFG_NO_NEXTAFTER - Success
    -- Performing Test CGAL_CFG_NO_STL - Success
    -- Performing Test CGAL_CFG_NO_WARNING_CPP_DIRECTIVE_BUG - Success
    -- Performing Test CGAL_CFG_NUMERIC_LIMITS_BUG - Success
    -- Performing Test CGAL_CFG_OUTOFLINE_MEMBER_DEFINITION_BUG - Success
    -- Performing Test CGAL_CFG_TEMPLATE_IN_DEFAULT_PARAMETER_BUG - Success
    -- Performing Test CGAL_CFG_TYPENAME_BEFORE_DEFAULT_ARGUMENT_BUG - Success
    -- Performing Test CGAL_CFG_USING_BASE_MEMBER_BUG_2 - Success
    == Write compiler_config.h (DONE) ==

    == Generating build files ==
    Configuring libCGAL
    -- Requested component: MPFR
    -- Requested component: GMP
    CMake Warning (dev) at src/CMakeLists.txt:51 (get_target_property):
      Policy CMP0026 is not set: Disallow use of the LOCATION target property.
      Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
      command to set the policy and suppress this warning.

      The LOCATION property should not be read from target "CGAL".  Use the
      target name directly with add_custom_command, or use the generator
      expression $<TARGET_FILE>, as appropriate.

    Call Stack (most recent call first):
      src/CGAL/CMakeLists.txt:11 (collect_cgal_library)
    This warning is for project developers.  Use -Wno-dev to suppress it.

    libCGAL is configured
    -- Sources for CGAL component library 'CGAL_ImageIO/CMakeLists.txt' detected
    CMake Error at src/CMakeLists.txt:81 (add_subdirectory):
      add_subdirectory given source
      "/home/richter/openscad_deps/src/CGAL-4.4/src/CGAL_ImageIO/CMakeLists.txt"
      which is not an existing directory.
    Call Stack (most recent call first):
      src/CMakeLists.txt:139 (configure_component)

    -- Sources for CGAL component library 'CGAL_Qt3/CMakeLists.txt' detected
    CMake Error at src/CMakeLists.txt:81 (add_subdirectory):
      add_subdirectory given source
      "/home/richter/openscad_deps/src/CGAL-4.4/src/CGAL_Qt3/CMakeLists.txt"
      which is not an existing directory.
    Call Stack (most recent call first):
      src/CMakeLists.txt:139 (configure_component)

    -- Sources for CGAL component library 'CGAL_Core/CMakeLists.txt' detected
    CMake Error at src/CMakeLists.txt:81 (add_subdirectory):
      add_subdirectory given source
      "/home/richter/openscad_deps/src/CGAL-4.4/src/CGAL_Core/CMakeLists.txt"
      which is not an existing directory.
    Call Stack (most recent call first):
      src/CMakeLists.txt:139 (configure_component)

    -- Sources for CGAL component library 'CGAL_Qt4/CMakeLists.txt' detected
    CMake Error at src/CMakeLists.txt:81 (add_subdirectory):
      add_subdirectory given source
      "/home/richter/openscad_deps/src/CGAL-4.4/src/CGAL_Qt4/CMakeLists.txt"
      which is not an existing directory.
    Call Stack (most recent call first):
      src/CMakeLists.txt:139 (configure_component)

    -- Sources for CGAL component libraries 'CGAL_Core/CMakeLists.txt;CGAL_ImageIO/CMakeLists.txt;CGAL_Qt3/CMakeLists.txt;CGAL_Qt4/CMakeLists.txt' detected
    == Generating build files (DONE) ==

    -- Configuring incomplete, errors occurred!
    See also "/home/richter/openscad_deps/src/CGAL-4.4/bin/CMakeFiles/CMakeOutput.log".
    See also "/home/richter/openscad_deps/src/CGAL-4.4/bin/CMakeFiles/CMakeError.log".

experienced with openscad-2015.03-1-620-g31211c6

t-paul, 2016-09-08:
Why not just use distribution dependencies? Ubuntu 16.04 has everything available.

donbright, 2016-09-09:
@t-paul i would gues the problem is they are trying to use the script from openscad-2015.03 on top of ubuntu 16, but ubuntu 16 wasn't released until april 2016. 

https://wiki.ubuntu.com/XenialXerus/ReleaseSchedule 

the "get dependencies" script didnt work on ubuntu 16 at that time (as, well, it's hard to test against operating systems that dont exist at the time)

----

i will try to see if there is an easy fix but.. it would be alot simpler to just build the current master from github using the current get-dependencies script (Which does work on ubuntu 16)

donbright, 2016-09-09:
OK here is the workaround to build openscad 2015 on ubuntu 16

```
       sudo echo
    wget http://files.openscad.org/openscad-2015.03-2.src.tar.gz
    tar xf openscad-2015.03-2.src.tar.gz
    cd openscad-2015.03-2/scripts
    mv uni-get-dependencies.sh uni-get-dependencies.sh.bak
    wget https://raw.githubusercontent.com/openscad/openscad/master/scripts/uni-get-dependencies.sh
    cd ..
    sudo apt update
    sudo bash ./scripts/uni-get-dependencies.sh
   
```

then you can proceed to qmake, make etc etc

donbright, 2016-09-09:
OK.

just tested uni-build-dependencies on ubuntu 16, made some fixes (basically you have to use a newer version of CGAL and a newer version of GMP) so if someone wants to try to build openscad on ubuntu 16 using the 'build dependency script' instead of getting the dependencies properly like above, then they need to do the same thing as above, but get the new "uni-build-dependencies.sh' script from master instead of the unit-get-dependencies script