openscad/openscad #1787: Double-click to center not working when compiled against Qt5

nairaner, 2016-09-12:
In Qt5 after frame is render framebuffer is switched to output frame. For finding new view port after double-click we use glReadPixels with Depth buffer which in qt5 is empty
[image]

superkartoffel, 2017-02-02:
I can confirm this issue. Even worse, I get a GLError 1282 from the glReadPixels call. It's probably because I'm on archlinux which uses wayland instead of xorg as default. I'm willing to admin that this might be a driver issue, so I'm not filing a new bug here.
Using xorg, I get z=1 and therefore still doesnt work. 

Any suggestion how to fix this?

AttentionalM, 2017-08-29:
Still an issue on Mac OS X (10.11) compiled using Qt version 5.7.0 from scratch on 28/08/2017

PR #3661 (2021-02-17): bring back doubleclick to center
Bring back double click to center, which occasionally did not work - problem was the Framebuffer was not current. 

closes #2466 (superseeded, simpler solution)
closes  #1787 (once and for all!)
closed 2021-02-17 by t-paul