openscad/openscad #1791: installing on KDE Neon with apckage manager

yeus, 2016-09-14:
my system:
```
lsb_release -a

No LSB modules are available.
Distributor ID: neon
Description:    KDE neon User Edition 5.7
Release:        16.04
Codename:       xenial
```

trying to install through package manager:

```
sudo apt install openscad
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 openscad : Depends: libqt5scintilla2-12v5 (>= 2.8.4) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

```
sudo apt install libqt5scintilla2-12v5
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libqt5scintilla2-12v5 : Depends: libqt5core5a (>= 5.4.0) but it is not going to be installed
                         Depends: libqt5gui5 (>= 5.0.2) but it is not going to be installed or
                                  libqt5gui5-gles (>= 5.0.2) but it is not going to be installed
                         Depends: libqt5printsupport5 (>= 5.0.2) but it is not going to be installed
                         Depends: libqt5widgets5 (>= 5.2.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

t-paul, 2016-09-14:
Which package version are we talking about and where will it be installed from?

`apt-cache policy openscad` should show more details

The missing dependencies might be a missing `universe` repo: http://packages.ubuntu.com/xenial/libqt5scintilla2-12v5

yeus, 2016-09-14:
```
apt-cache policy openscad
openscad:
  Installed: (none)
  Candidate: 2015.03-1+dfsg-3
  Version table:
     2015.03-1+dfsg-3 500
        500 http://debian.charite.de/ubuntu xenial/universe amd64 Packages
```

and also installing the package manually from the repository didn't help. universe repository is aready enabled.

I also filled this bug report here:
https://bugs.kde.org/show_bug.cgi?id=368792

t-paul, 2016-09-14:
Ok, as we don't have any control over those packages and repositories, I'll close this for now. This looks like some inconsistencies in the selected repository, e.g. packages not rebuilt against updated dependencies.

If there's anything coming up that needs to be done on OpenSCAD side, feel free to reopen.

closed 2016-09-14 by t-paul

donbright, 2016-09-16:
fyi for any future people looking at this, qscintilla qt5 is broken in kde neon circa september 2016 but openscad builds fine against qt4 on that platform. (just use qmake-qt4)