openscad/openscad #1792: building on KDE Neon

yeus, 2016-09-14:
my system:

```
lsb_release -a
No LSB modules are available.
Distributor ID: neon
Description:    KDE neon User Edition 5.7
Release:        16.04
Codename:       xenial
```

I did  "sudo ./uni-get-dependencies.sh"

it can not do this:

he following packages have unmet dependencies:
 libqt5scintilla2-dev : Depends: libqt5scintilla2-12v5 (< 2.9.1+dfsg+1~) but it is not going to be installed
                        Depends: libqt5scintilla2-12v5 (>= 2.9.1+dfsg) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

t-paul, 2016-09-14:
Please enable `universe`:

http://packages.ubuntu.com/xenial/libqt5scintilla2-dev

closed 2016-09-14 by t-paul

yeus, 2016-09-14:
"universe" is already enabled it worked by manually installing the package from the provided link above

t-paul, 2016-09-14:
Hmm, that's strange. We also just call the system tools, why would that result in a different behavior? Or do you mean you downloaded the *.deb file?

If that's the case maybe they are using a different package source than upstream Ubuntu but that's really a distribution issue then.

yeus, 2016-09-14:
yepp.  don't really  konw what to about it right now 
I also opened a bug report here:

https://bugs.kde.org/show_bug.cgi?id=368792

there is another bug realting to the same issue where they apparently recompiled the application and it worked after that but that doesn't work for me :), as I am just trying to recompile it:

https://bugs.kde.org/show_bug.cgi?id=368319

yeus, 2016-09-14:
after installing the package I get this here (qscintilla2 was on "NotOK" before):

`/usr/bin/ld: cannot find -lqt5scintilla2`

during compilation with `qmake openscad.pro ` and `make`

```
depname     minimum     found       OKness      
qt          4.4         unknown     NotOK       
qscintilla2 2.7         2.9.1       OK          
cgal        3.6         4.7         OK          
gmp         5.0         6.1.0       OK          
mpfr        3.0         3.1.4       OK          
boost       1.35        1.58        OK          
opencsg     1.3.2       1.4.        OK          
glew        1.5.4       1.7.0       OK          
eigen       3.0         3.2.92      OK          
glib2       2.0         2.48.1      OK          
fontconfig  2.10        2.11.       OK          
freetype2   2.4         2.6.1       OK          
harfbuzz    0.9.19      1.0.1       OK          
bison       2.4         3.0.4       OK          
flex        2.5.35      2.6.0       OK          
make        3           4.1         OK
```

running 

```
sudo apt-get -f install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages will be REMOVED:
  libqt5scintilla2-dev
0 upgraded, 0 newly installed, 1 to remove and 74 not upgraded.
1 not fully installed or removed.
After this operation, 495 kB disk space will be freed.
Do you want to continue? [Y/n]
```

also shows that there are some problems left

t-paul, 2016-09-14:
There might be multiple `qmake` versions as it's possible to install both 4.x and 5.x as default.

Try forcing the Qt5 `qmake`, Debian uses `QT_SELECT=5 qmake ...` for that.

donbright, 2016-09-15:
qt5 scintilla is completely broken on KDE Neon . i altered the build dependency scripts to use qt4.

GHPS, 2017-03-03:
Just an update: Updated versions of libqt5scintilla2-12v5 and libqt5scintilla2-l10n are available (2.9.3) [1]. They are compatible with Qt 5.6.1 and OpenSCAD  2015.03-1.

Background:  I had the problem described with OpenSCAD and Kubuntu 16.04 LTS. The upgrade to Qt 5.6.1 deleted libqt5scintilla2-12v5 (2.9.1) since it broke Qt5.6 - and removed OpenSCAD with it.

I solved the problem with the newer versions of libqt5scintilla2-12v5 and libqt5scintilla2-l10n. Since they haven't come upon us the official way, I had to download and install them directly.

And voila - OpenSCAD installs and works fine.

GHPS

1: https://www.ubuntuupdates.org/package/core/yakkety/universe/base/libqt5scintilla2-12v5
   https://www.ubuntuupdates.org/package/core/yakkety/universe/base/libqt5scintilla2-l10n

t-paul, 2017-03-21:
For reference.... same issue also with Mint 18.

yeus, 2017-03-24:
works ony my system, too !

thx GPHS, installing those two apckages first install
* https://www.ubuntuupdates.org/package/core/yakkety/universe/base/libqt5scintilla2-l10n
then 
* https://www.ubuntuupdates.org/package/core/yakkety/universe/base/libqt5scintilla2-12v5

using `sudo dpkg -i <package-name>`

then  
             `sudo apt install openscad`

core-engineering, 2017-03-28:
works on my Ubuntu 16.04 ... thanks GHPS and yeus

Regards,

Camille

ipsod, 2017-04-05:
I'm having the same problem on `Linux Mint 18 KDE`, but can't install https://www.ubuntuupdates.org/package/core/yakkety/universe/base/libqt5scintilla2-12v5 - I get this error:

    The following packages have unmet dependencies:
     libqt5scintilla2-12v5 : Depends: libqt5core5a (>= 5.4.0) but it is not going to be installed
                         Depends: libqt5gui5 (>= 5.0.2) but it is not going to be installed or
                                  libqt5gui5-gles (>= 5.0.2) but it is not going to be installed
                         Depends: libqt5printsupport5 (>= 5.0.2) but it is not going to be installed
                         Depends: libqt5widgets5 (>= 5.2.0) but it is not going to be installed
    E: Unable to correct problems, you have held broken packages.

csun92, 2017-04-28:
thanks GHPS and yeus, worked on my KDE Neon 5.9, i just downloaded the deb packages and installed them using dpkg.

For information.

battaglia01, 2018-06-14:
Those two packages are no longer available - is there some other way to install them?

GHPS, 2018-06-15:
Yes - this has become quite a blind alley. Not only have the package
sources disappeared but also - the tricky part - have the respective
files changed their names. 

libqt5scintilla2-l10n -> libqscintilla2-qt5-l10n
libqt5scintilla2-12v5 -> libqscintilla2-qt5-13 

I've checked under Kubuntu 18.04 LTS. After adding the
openscad package source (ppa:openscad/releases) the package
lists libqscintilla2-qt5-13 as a dependency. 

So the easiest way of installing openscad should be to grab
this version and the latest version of libqscintilla2-qt5.