openscad/openscad #1793: Coincident surfaces where there should be none in preview mode

deffi, 2016-09-14:
So I'm trying to create a simple square-shaped object with the following cross section:

    //  __            __
    // |__|__________|__| Yellow
    // |________________| Red

The lower (red) part is simply a cube, and the upper (yellow) part is created by subtracting a cutout from a similar cube:

    //     __________
    //    |          |
    //  __|__________|__
    // |__|__________|__| Upper
    //    |          |
    //    |__________|
    //     Cutout

Full source code:

    color("red") {
        cube([100, 100, 1]); // Lower
    }
    
    color("yellow") {
        difference() {
            translate([ 0,  0,  1]) cube([100, 100, 1]); // Upper
            translate([10, 10, -1]) cube([ 80,  80, 5]); // Cutout
        }
    }   

Now the preview looks like there were two coincident surfaces at the plane between the lower and the upper part:

[image]

You can easily verify that the upper part does not have a surface at this position by prefixing the `color("yellow")` with a `!`.

nophead, 2016-09-14:
The bottom of the upper yellow cube is coincident with the top of the red
one. They way OpenCSG works you can't have any coincident faces if they are
visible, even if they are not in the final result. This is because it
doesn't compute the geometry it just draws pixels and relies on the Z
buffer to sort it out.

Putting render() at the top level of objects before they are coloured fixes
them so they don't interfere with each other.

deffi, 2016-09-14:
Using `render()` as suggested by @nophead resolves the issue, but makes the preview very slow (not in this simple case, but it's an issue if the cutout consists of an extruded text instead of a cube).

nophead, 2016-09-14:
It makes the first preview slower but subsequent ones much faster as the result of render() is cached and only recomputed it it changes. It is the only way to make assemblies with hundreds of parts in OpenScad as without the renders the GUI becomes totally unresponsive and you get lots of interference between objects like above.

It isn't a bug in OpenScad that can be fixed. It is just the way OpenCSG works.

nophead, 2016-09-14:
The other way to work around it is to make the planes no exactly coincident by adding small offsets.
```
color("red") {
    cube([100, 100, 1]); // Lower
}

color("yellow") {
    difference() {
        translate([ 0,  0,  .999]) cube([100, 100, 1]); // Upper
        translate([10, 10, -1]) cube([ 80,  80, 5]); // Cutout
    }
} 
```

.999 or 1.001 works until you zoom out far enough to run out of Z buffer resolution.

kintel, 2016-09-16:
This is addressed in the FAQ: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview.3F

closed 2016-09-16 by kintel

deffi, 2016-09-16:
I believe that this is not the issue addressed in the FAQ: the FAQ describes the case where two surfaces coincide. This case here is different in that one of the surfaces (the yellow one) is removed by the `intersection`, but is still rendered by OpenCSG.

May I suggest that the FAQ should explicitly mention that the issue can occur _even if one of the surfaces is removed_ by a CSG operation, and give the workaround using `render()`.

nophead, 2016-09-16:
Nothing is ever removed in OpenCSG. It draws three full cubes in the right
order and correct modes to produce a preview that visually simulates the
CSG operations using the depth buffer. When you add preview() CGAL computes
the difference and then OpenCSG draws a cube and a square ring.

deffi, 2016-09-16:
Thanks for the explanation. I have amended the FAQ.