openscad/openscad #1796: Add render mode settings

l0b0, 2016-09-18:
[`$vpr`, `$vpt` and `$vpd`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#.24vpr.2C_.24vpt_and_.24vpd) are very useful, and it would be great to have options to hint about other rendering settings: 
[image]

For a very crude example, some files are more usefully rendered in either perspective or orthogonal projections. This could be set by either `$vpp = "orthogonal";` (the latter "p" meaning simply "projection"); `$vpp = true;` or `$vpo = true;` ("p" meaning "perspective", "o" meaning orthogonal); or `$vpd = infinity; $vps = [x, y]` for orthogonal view, while any other `$vpd` is perspective.

t-paul, 2016-09-18:
I'd vote for not expanding the special variable based way to do this. It's going to be very messy with lots of variables. Still, I agree the feature is useful, so maybe we should have a look at how to give more detailed control via a new and better defined API.

donbright, 2016-09-20:
create a variable called `$vcmdline` and then re-use all the syntax and code from the command line options

this will prevent the two 'ways of doing things' from getting out of synch