openscad/openscad #1797: Respect $vpr, $vpt, $vpd when rendering from command line

l0b0, 2016-09-18:
The `openscad` command seems to ignore `$vpr`, `$vpt` and `$vpd` when rendering PNGs. Example file:

```
$vpr = [90, 0, 0];
$vpt = [0, 0, 0.5];
$vpd = 3;
cube();
```

Example command:

```
openscad -o example.png example.scad
```

Result:

[image]

Compare to the default rendered view in OpenSCAD:

[image]

Duplicating the viewport settings in the command line (`--camera=0,0,0.5,90,0,0,3`) works, but I would expect the command to use the variables in the file if `--camera` was not specified.

t-paul, 2016-09-18:
The `$vpX` (and also `$t`) variables are currently a GUI only feature. I think that should stay that way, but I'll set the issue to "discussion" to collect more input...

donbright, 2016-10-06:
as a user i would prefer it to "just work" without me having to think about it, so i support the $vp (camera settings) being read from .scad and have them work even in cmdline mode. 

as a user i would also hope that the camera setting commands in gui are similar if not identical to those on the command line, to avoid confusion and keep things simple. At this point, i would consider if it is possible to deprecate the --camera cmdline arguments and make them undocumented, (keep only for backwards compatability) and just replace them with new command line options based on --vp, something like --vpr= and --vpd= so that they match the syntax and mental model of how the camera setting system works in the gui. (as long as vp can be extended to be feature identical with what --camera currently offers, like projection, colorscheme, etc). 

Either that or .... allow a $cmdline variable inside the source code scad file that will allow camera setting, like // $cmdline --camera=0,0,4,3,4,2,3,4 --projection=ortho --colorscheme=starnight 

thats my two cents

t-paul, 2016-10-06:
There's also the question what happens if $vpX is assigned in the script and also given on command line. Is the command line able to override the script settings?

doug-moen, 2016-10-06:
Currently, command line settings always override script settings. It would
be inconsistent and unexpected for this new feature to work any other way.

nophead, 2016-10-06:
If it obeyed the $vp variables in command line mode you can set or override
them with the -D option. No surprises there and no new command syntax. I
think it would also be OK for the $vp variables to override the existing
command line options as the only use case I can think of where both are set
at the moment is for them to be set the same anyway.

t-paul, 2016-10-06:
As `-D` is probably the main problem that prevents fixing the reassignment issues, it might be a good thing to slowly move away from this, so I'd vote for the suggestion @donbright made to have those specifically set (either by specific option like `--vpr` or maybe some generic one which really only allows to set variable values, not paste strings like `-D` does, e.g. `--var=a:4` or whatever the syntax could be). But that's probably a different topic.

So values set via command line would always take precedence in case something is set in the script too. I guess that's matching the expectation of most users, so that's good. The camera option currently allows 2 different sets of parameters, not sure if we need to find a way to retain that feature.

donbright, 2016-10-07:
lol i forgot that -D existed. using -D sounds better than what i wrote above, because -D is already there... but what are "the reassignment issues"?

t-paul, 2016-10-07:
The often discussed
```
a = 1;
echo(a);
a = 2;
echo(a);
```
Right now, there's no warning and no error, so people do get confused (and serves as bad example in talks ;-), but I guess that's not so important).

Fixing that is not possible as that breaks `-D` which just pastes the text value at the end of the script. Simply changing the behavior of variable assignment is also not possible as `-Dcube()` works too right now.

donbright, 2016-10-07:
the two different camera options IIRC was because the old original test suite was based on a different type of camera than the GUI, but we were rebuilding things so that the openscad GUI binary itself could be used in the test suite. Also i used the word '--camera' because it was relatively simple and easy to understand what it does just by looking at it in the --help screen. 

The first camera type is for the 'gimbal camera' using rotation, translation, and distance (7 numbers). This is what is in the GUI viewport thingy.

The other is for a "Vector Camera" which is related to the traditional computer graphics idea of having two points, a camera position-point, which indicates where the camera is located in space, and a "center point" also called "target point" to which the camera is looking at.

 vpr/vpt/vpd doesnt match up with the 'vector camera' idea so to fully replace --camera, there would need to be new vp varibles, but "vp" by itself implies the word "viewport", which is not the same thing as "camera", so i am not sure how to name it. i am one who favors spending a lot of time thinking about naming things when possible.

nophead, 2016-10-07:
Isn't it easy to look if a reassignment is in the same source file and give
an error or warning if it is. Command lines ones would then still work (as
they are in no source file) and also included library files with variable
set  to defaults that you want to override would work. I can't see a use
case for setting a variable twice in the same source file. I think that is
always accidental and causes bugs.

doug-moen, 2016-10-07:
I don't think this is difficult, it's just a matter of coding. My idea is
to append the -D settings to the parse tree, not the source file. That way,
the assignment nodes are marked with a different source location than
settings inside the source file. But this does require some refactoring.

One approach is to modify the parse tree after it has been constructed,
another is to hack the lexical analyzer to read the -D settings and insert
them into the token stream using a special string for the filename like "-D
flags".

t-paul, 2016-10-07:
@doug-moen what you describe is basically what I meant with "not possible". Yes of cause it's *possible* to do for maybe 99.98% of the cases, but it's quite a horrible solution to parse the `-D` values and then trying to merge back into the main parse tree. If we could rely on this having only variable assignment, that might be ok-ish, but the point is, there's no limit to what's in there. It can be includes, functions, modules, anything which also could be in the main file. In the general case, it's not even safe to assume both parts are separately parse-able.

It might be acceptable to just remember the last source code line before adding the `-D` stuff and use that to suppress the error/warning, the framework for that already exists in the GSoC branch. It's still not an especially nice solution.

Example scad file (which does not compile as is):
```
a = undef;
echo(a);
a =
```
Command line call:
```
$ openscad -o /tmp/test.ast -D'"hi!"' test.scad
ECHO: "hi!"
```

doug-moen, 2016-10-07:
@t-paul wrote:
> Example scad file (which does not compile as is):
```
a = undef;
echo(a);
a =
```
Command line call:
```
$ openscad -o /tmp/test.ast -D'"hi!"' test.scad
ECHO: "hi!"
```

Personally, I think it's a bug that this works, and I don't think it's important to continue supporting it in the future.

However, in my previous note, I suggested two implementations:
 1. Parse each -D flag separately, and append the resulting parse tree node to the parse tree for the script. This implementation is appropriate if we want to validate that each -D flag parses correctly.
 2. Modify the lexical analyzer to tokenize the concatenation of all the -D flags, after tokenizing the source file. This method is compatible with @t-paul's requirement that -D flags can contain arbitrary source text, with no validation.

t-paul, 2016-10-07:
Just knowingly breaking backward compatibility is a bad thing in my opinion, even if the feature maybe was not intended to work that way. People do use OpenSCAD in server environments, it's not just Thingiverse. Hence I'd suggest to deprecate `-D` and start moving to a better defined (and more limited) API.

nophead, 2016-10-07:
Those two sentences seem contradictory to me. First you say breaking an
obscure non intended feature is bad and then you advocate removing a big
feature that lots of people use.

I imagine quite a few users pass statements with -D but I can't imagine
many pass half a statement.

t-paul, 2016-10-07:
No, I said _deprecate_. Which means it will be there for a couple of releases, which at the current pace would mean like 3 or 5 years.

nophead, 2016-10-08:
My personal view would be that -D should be kept able to override variables and inject statements as it is useful to be able to instantiate an object from the command line but dropping completing half statement is no loss.

Since errors in the -D parameters are currently reported with line numbers past the end of the source file surely it is easy to say redefining a variable is an error unless it is a different source file, or the line number is past the end of the source file?

kintel, 2016-10-08:
With the recent improvements on how included files and error locations are handled (by @doug-moen I think), it would be pretty easy to consider all -D parameters as belonging to an included, unnamed, file at the end of the main .scad file, with correct error reporting. This would be one small step in the right direction.

While I'm generally concerned about breaking backwards compatibility, I don't think breaking corner-case usage of -D parameters is too bad, as these are mostly used by power users who should be able to adapt. Requiring each -D parameter to be a valid statement would be a good start. Subsequently we could consider requiring them to be a valid assignment from a literal. As @t-paul points out, we might be better off using a different argument for this, and slowly deprecate -D.

nophead, 2016-10-08:
But some of us want to continue to be able to instantiate an object with -D as it is very useful. Why restrict it to an assignment when it currently works with a statement?

kintel, 2016-10-08:
@nophead I wrote "_could_". That would be a future discussion; we're not in a habit of breaking common workflows overnight :)

kintel, 2016-10-08:
Btw., related to this discussion: I'd like to explore parsing ```include``` files as separate compilation units, and perform merging of namespaces after parsing, similar to how to manage ```use``` today. That could make dealing with the above a bit cleaner.

PR #3440 (2020-09-25): Honor view-port variables in command-line mode.
This patch-set moves updateCamera() into Camera as updateView() so it can be used in GUI and command-line mode.
The command-line default settings (viewall and autocenter) are automatically disabled when one of the view-port variables is set.

Closes #1797
closed 2020-09-25 by t-paul