openscad/openscad #1798: Can't generate PNG file without an X server

l0b0, 2016-09-18:
From an [automated build](https://travis-ci.org/l0b0/graphics/builds/160819837#L230-L235):

```
/usr/bin/openscad --projection=ortho --camera=0,0,0.5,90,0,0,3 -o chisel/equilateral_plug_example.png chisel/equilateral_plug_example.scad
Compiling design (CSG Products normalization)...
Normalized CSG tree has 2 elements
Compiling background (1 CSG Trees)...
Unable to open a connection to the X server
Can't create OpenGL OffscreenView. Code: -1.
```

I tried the same in a text mode console, and got the same result. This creates a zero-length file and returns with a *zero exit code.*

```
$ openscad --info
OpenSCAD Version: 2015.03-3
System information: Linux 4.7.3-2-ARCH #1 SMP PREEMPT Thu Sep 8 09:44:02 CEST 2016 x86_64 Arch Linux 4 CPUs 3.78 GB RAM
Compiler: GCC "6.1.1 20160802" 64bit
Boost version: 1_61
Eigen version: 3.2.9
CGAL version, kernels: 4.8.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 4.8.7
QScintilla version: 2.9.2
MingW build: No
GLib version: 2.48.2
Application Path: /usr/bin
Documents Path: /home/username/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/username/.local/share/OpenSCAD/libraries
User Config Path: /home/username/.config/OpenSCAD
Backup Path: /home/username/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/username/.local/share/OpenSCAD/libraries
  /usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:

GLEW version: 2.0.0
OpenGL Version: 3.0 Mesa 12.0.3
GL Renderer: Mesa DRI Intel(R) Sandybridge Mobile 
GL Vendor: Intel Open Source Technology Center
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: GLX
PNG generator: lodepng
GLX version: 1.4
OS info: Linux 4.7.3-2-ARCH #1 SMP PREEMPT Thu Sep 8 09:44:02 CEST 2016
Machine: x86_64
```

t-paul, 2016-09-18:
Yes, that's currently a limitation, it's possible to use `Xvfb` or something similar to cover this case (like the test framework does).
I'll close this as it's intended as bug report.

closed 2016-09-18 by t-paul

l0b0, 2016-09-18:
Surely the zero exit code and empty output file are not as intended?

t-paul, 2016-09-18:
I guess it's not intended. #1799 should fix the missing the exit code. The empty output file is currently not easy to prevent as the output streams are created before the actual export function runs. If that's really needed, we can probably create a separate issue for that.

donbright, 2016-09-19:
the answer is to get rid of my C++ Exception and just clean up properly.

spuder, 2021-12-20:
Workaround using Xvfb

```
cd stl
Xvfb :99 & export DISPLAY=:99
for i in *.stl; do
  T=__tmp__$i
  b=`basename $i`
  echo import\(\"$i\"\)\; >$T
  /usr/bin/openscad -o $b.png --imgsize=200,200 $T
  rm $T
done
  ```

eXpire163, 2023-06-25:
Thanks for the hint with xvfb. 

I have put it all in a PR to an already existing openscad github action. 
https://github.com/flat35hd99/openscad-actions
In case it's not merged yet, feel free to checkout the PR
https://github.com/flat35hd99/openscad-actions/pull/6

t-paul, 2023-06-25:
For completeness, there's also an EGL docker image (openscad/openscad:egl) that works without X. So far this needs special build flags and also dedicated builds of OpenCSG and GLEW. A better solution might happen at some point via #4582