openscad/openscad #1803: intersection() not working together with difference()

engblom, 2016-09-20:
When doing a preview and a full render of the following code, only the full render gives the expected result. In the preview, the whole intersection is absent, while pieces not belonging to the intersection are shown.

$fn=32;

intersection(){
    difference(){
        cube(size=50, center=true);
        cylinder(d=30,h=80, center=true);
        rotate([90,0,0]) cylinder(d=30,h=80, center=true);
        rotate([0,90,0]) cylinder(d=30,h=80, center=true);
    }
    cylinder(d=65,h=80,center=true);
    rotate([90,0,0]) cylinder(d=65,h=80,center=true);
    rotate([0,90,0]) cylinder(d=65,h=80,center=true);
}

t-paul, 2016-09-20:
Confirmed in nightly build on Linux but works with the release version 2015.03-1.

Release version: 2015.03-1 / OpenCSG 1.4.0
Nightly build: 2016.09.19.nightly (git 4a0f3b2) / OpenCSG 1.3.2

Adding `render(convexity = 3)` to the difference fixes the preview in the nightly build.

thehans, 2016-09-20:
Just did a little testing/playing around.  

I simplified the test case by removing a few lines.  The following still exhibits broken preview:

```
$fn=32;

intersection(){
    difference(){
        cube(size=50, center=true);
        cylinder(d=30,h=80, center=true);
    }
    cylinder(d=65,h=80,center=true);
    rotate([90,0,0]) cylinder(d=65,h=80,center=true);
}
```

Another note is that it works ok if the difference goes last in the intersection:

```
$fn=32;

intersection(){
    cylinder(d=65,h=80,center=true);
    rotate([90,0,0]) cylinder(d=65,h=80,center=true);
    difference(){
        cube(size=50, center=true);
        cylinder(d=30,h=80, center=true);
    }
}
```

Tested on 2016.09.19.nightly (git 4a0f3b2)

kintel, 2016-10-01:
Thanks for the report.
I've bisected it to this commit: https://github.com/openscad/openscad/commit/44a063275188b68b757fb9b0899168c081d217a0

kintel, 2016-10-07:
See https://github.com/openscad/openscad/tree/issue1803

kintel, 2016-10-08:
Fixed by #1821

closed 2016-10-08 by kintel