openscad/openscad #1806: Critical error on including files to SCAD object file

ManfredEP, 2016-09-27:
just a few minutes ago my OpenSCAD installation crashes with error message:
a critical error was caught. boost::filesystem::read_symlink
original in german the next:
request is not supported (or so)
and the path.
It comes up in use and include statements. Ther is no symlink in path but a NTFS mount point. 
These are screen shots:
[image]
[image]
The module ModBoden uses some of the other modules so there are double-includes in it.

kintel, 2016-09-27:
This looks like a duplicate of #1309

t-paul, 2016-09-27:
Yeah, same error message, just in German. Unfortunately we can't easily solve that as the issue seems to be in boost itself (linked in #1309).

Closing as duplicate.

closed 2016-09-27 by t-paul