openscad/openscad #1810: Restore code after crash

Ma-XX-oN, 2016-09-29:
I thought that this was already implemented, but I guess I was wrong.  I was creating a openscad doc, and I did some stupid infinite recursion thing, which it never recovered from.  So I closed the programme, and reopened it, to my surprise, there was no backup or any backup that was reported when I reloaded the file I had been working on.  The file was one that I saved some time ago.

Most editors that execute what was being edited now save the doc prior to trying to execute, if not to the main file, to a backup file.  If such a backup is found later while trying to load the main file, the editor would ask if the user if they would like to recover the last backup.

Can something similar be implemented with OpenScad please?

t-paul, 2016-09-29:
There should be a backup file, see https://github.com/openscad/openscad/wiki/Path-locations

The location is also mentioned in `Help`->`Library Info`.

closed 2016-09-29 by t-paul

Ma-XX-oN, 2016-09-29:
Ah, yes, I see that now.  Guess the text rolled off the output window.  However, on loading the file, why not tell the user that there is one available and give them the option to restore, instead of them having to dig it out manually?

t-paul, 2016-09-29:
Yes, that would be nice.