openscad/openscad #1811: Clarifying behavior

RonaldoCMP, 2016-09-30:
This is my first post in this forum. I hope it is the right place to do it. If I am wrong, please, let me know.

I see three possible behaviors of the evaluation of an expression by OpenSCAD: it does what is expected, it does something surprising but reasonable and it does something unreasonable. The last one is clearly a bug. The first one is the normal case. My interest now is on the second case: a reasonable surprising behavior. It is surprising when it comprises less common conditions that are not documented. Being undocumented make them not eligible to be a feature and that is the issue.

A first example is the product of a list by a float. It is mentioned in the manual just for vectors. But it works as expected with any list of floats, like matrices, n-dimensional matrices or even trees of floats. The same happens with the addition (subtraction) of lists. I cannot expect to add two lists with different structures (like a vector and a bi-dimensional matrix) but OpenSCAD is able to add lists of floats with the same structure:  `[ 1, [2, 3] ] + [ 5, [5, 5] ] = [ 6, [7, 8] ]` .

As those more general cases are missing in the documentation, I can not rely on it as feature. However, a lot of code I have been written (my Bezier/Spline library, for instance) benefits from that generality. If needed I can provide simple examples.

So, my question is: could we rely on the current behavior of those operators as a feature? Is there a commitment with this feature for the future versions? This should be clarified and the documentation be expanded accordingly. I hope the answer be yes :)

I have a couple of other issues like that. I will post about them later.

kintel, 2016-09-30:
This is an issue tracker, not a forum. Could you either rephrase this into a bug report, alternatively restart the discussion on the mailing list (http://lists.openscad.org/mailman/listinfo/discuss_lists.openscad.org) ?

t-paul, 2016-09-30:
What's an example for the case considered as bug?

kintel, 2016-09-30:
I didn't read the details :) If it's a pure question we can just close it.

RonaldoCMP, 2016-10-01:
I am not reporting anything I consider a bug. I want to clarify if the results I get with the product of a scalar float and a list are intentional, part of the language specification and will be preserved in future versions. That is important for me but it is undocumented. As far as my tests show, the product of a number and a list is a list with the same list structure were each atomic element of the list is multiplied by the number. Note that this is a generalization of the scalar product for vectors in linear algebra.
The same question applies to the sum of lists.
If this cannot be answered here, please close the issue.

closed 2016-10-02 by kintel