openscad/openscad #1813: OpenSCAD has switched to French for macOS Sierra

mprogers, 2016-10-01:
After upgrading to macOS Sierra, OpenSCAD's menus are all in French, even though my primary language is set to English and my region to the US. Avez-vous des idées? (Do you have any ideas?)

t-paul, 2016-10-01:
Can you please open a Terminal window, run `locale` and paste the output of that here?

mprogers, 2016-10-01:
> On Oct 1, 2016, at 11:17 AM, Torsten Paul <notifications@github.com> wrote:
> 
> Can you please open a Terminal window, run locale and paste the output of that here?
> 
> 
Yes, here you go:

$ locale
LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL=

t-paul, 2016-10-01:
That's strange. What happens when running OpenSCAD from a terminal? Is that still giving the translated texts? e.g. assuming it's in the user Applications folder:

`$HOME/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD`

As workaround it's possible to disable translations in the preferences completely which should give the default English text again.

mprogers, 2016-10-01:
> On Oct 1, 2016, at 11:36 AM, Torsten Paul <notifications@github.com> wrote:
> 
> That's strange. What happens when running OpenSCAD from a terminal? Is that still giving the translated texts?
> 
No, that opens up in English.

Michael

Pointedstick, 2016-10-02:
I have the same issue, and opening from Terminal loads it in English.

```
$ locale
LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL=
```

kintel, 2016-10-04:
Does anyone know how to install macOS Sierra on an OS X VirtualBox host? (i.e. not someone who can Google it, but someone who actually did it)

doug-moen, 2016-10-04:
https://developer.apple.com/library/content/technotes/tn2418/_index.html

"With iOS 9, the results returned by Locale.preferredLanguages() can differ
from previous releases. In iOS 8 and earlier, only certain language,
script, and region combinations were returned by this API. However, in iOS
9, more combinations of language, script, and region are permitted. This
change is also applied to macOS Sierra, watchOS 3, and tvOS 10.

For example, when a user has configured their iOS device with language set
to English and region set to India, Locale.preferredLanguages() will now
return [ "en-IN" ], instead of [ "en" ]. This allows for smarter language
fallbacks; for this user, if an app doesn’t support en-IN as a
localization, but does support en-GB, the fallback mechanism will select
en-GB instead of en."

kintel, 2016-10-07:
We use gettext and setlocale() to manage translations. I don't think any of those would call into the Locale library. ..unless Apple simply didn't document any changes to libc..

ipartola, 2016-10-20:
So... has there been a fix or a workaround for this?

t-paul, 2016-10-20:
Fix, no, it's still not clear what change causes this. Workarounds only the 2 mentioned above.

kintel, 2016-10-20:
@ipartola I would look into it if I could get access to a dev machine running Sierra. ..or if someone could help getting Sierra to install in VirtualBox on a Yosemite host.

donbright, 2016-10-21:
Here is a brief list of some other programs facing similar issues.

**Xamarin**

https://forums.xamarin.com/discussion/76678/xs-changed-language-after-upgrade-to-macos-sierra-how-to-fix

According to Marius Ungureanu on the linked site,,"this is a bug in OS X Sierra caused by regressions in OS X's internationalization stack. We have deployed a fix on our side for the alpha channel"

**Matlab**

(warning, harsh thread.)
https://www.mathworks.com/matlabcentral/answers/303757-why-does-matlab-fail-to-load-my-preferred-language-on-macos-sierra?requestedDomain=www.mathworks.com

"Apple released macOS Sierra on September 20, 2016 and changed the way macOS reports Language & Region preferences to MATLAB."

note that the Matlab people are linking to the same Technical Note 2418 page as @doug-moen did above. https://developer.apple.com/library/content/technotes/tn2418/_index.html

**Lilypond**

http://lilypond.1069038.n5.nabble.com/language-issues-with-Mac-OS-X-Sierra-10-12-td195168.html

Doesn't look like they have a complete fix yet. Note Lilypond is a GNU / open source project so probably the most similar to OpenSCAD in terms of language backend as compared to say closed source projects like MatLab. 

***iOS9 apps***

If you type into google "iOS9 localization bug" you will find a huge amount of hits, many of them link to that same Technical Note 2418 https://developer.apple.com/library/content/technotes/tn2418/_index.html

Example

http://stackoverflow.com/questions/32710913/ios-localization-is-broken-with-the-upgrade-ios-9-xcode-7/33524891

t-paul, 2016-10-21:
Thanks for digging, that's really some nasty discussions going on, especially in that Matlab thread.

I'm still not sure this is exactly the same issue we see. The Matlab thread also mentions:
```
The following locales are not impacted by this issue:

    * English, United States
    ...
```
But that's exactly the locale shown on command line in the cases listed above as `en_US.UTF-8`. @mprogers, @Pointedstick could you please check if the configuration is indeed one of the locales listed in the [Answer in the Matlab forum](https://de.mathworks.com/matlabcentral/answers/303757-why-does-matlab-fail-to-load-my-preferred-language-on-macos-sierra?#answer_235386) (e.g. showing "English, United States" as the top-most entry)?

I guess one way to find out is to add the locale values to our library info screen and see what that shows when the application is started from finder.

Pointedstick, 2016-10-21:
My top locale appears to be English - U.S. 
[image]

I continue to see the issue as long as I have localization checked in OpenSCAD's preferences.

kintel, 2016-10-21:
@Pointedstick What happens if you remove all languages except English from your preferred languages?

Pointedstick, 2016-10-21:
@kintel Aha! removing all but English makes OpenScad launch properly in English even with localization is turned on.

kintel, 2016-10-22:
@Pointedstick Thanks for testing. Yes, the problem only occurs when launching from the Finder.

bbum, 2016-10-23:
When launched from the Finder, there really isn't a standard shell like environment.   In past OS X releases, more of the user's shell configuration was made available to apps that were double-clicked in the Finder.   That has changed over time and it appears that the locale was the latest to be dropped.

You'll need a bit of OS X specific glue to grab the localizations from NSLocale.

kintel, 2016-10-23:
@bbum That would indeed explain a lot.

kintel, 2017-08-03:
This turned out to be a bug in gettext (https://savannah.gnu.org/bugs/?49560). Awaiting an update to gettext, this has been fixed by patching our gettext build.
Note: This means that until other builds of gettext are patched (e.g. homebrew, MacPorts), only the official OS X binaries of OpenSCAD will work as expected.

closed 2017-08-03 by kintel

MichaelAtOz, 2018-04-29:
@t-paul @kintel 
So who should claim the bounty for this?

kintel, 2018-04-29:
Can we refund it? It was fixed by someone in the `gettext` library..

MichaelAtOz, 2018-04-30:
There doesn't seem to be anything built-in to do a refund. I could email BS.

_Or_, for the effort ($9), @donbright @t-paul & @kintel seemed to share the co-ord for this;  **$3 each? Buy yourselves  [something special](https://www.shapeways.com/product/6VEMXWHGV/logo2-15mm?key=e260d6e6f5bd261cda619158b02b4064&optionId=65365611)** :1st_place_medal:  -  (the metalic plastic is < $3) - at cost ATM for those who see this.

The 18k Gold Plated polishes up nice (this is the 10mm long version):
[image]

Or you may want to save up for the developer moonlight ninja edition ( 15mm raw bronze): 
[image]
(please excuse the phone camera limitations)

MichaelAtOz, 2018-04-30:
Or, I just found this in the FAQs
```
If the issue was closed as a "won't fix", or is deemed not in line with the project's goals, 
the bounty is refunded.

If I posted a bounty using team funds, where does the refund go?
Refunds will go back into the team account.
```

Could try reopening, labelling Won't fix & closing.

t-paul, 2018-04-30:
Refund sounds good to me. As far as I can see the bounty was posted by @kintel anyway, so just claiming it would be same effect, I guess?

ErnestoBorio, 2018-07-16:
As you can see in the picture, my preferred language is english and locale is en-US.
Only after removing all preferred languages and leaving only english did it.
Before this, Russian was chosen, which was like 5th on the preferred language list.
Also, nothing like this has ever happened to me in 10 years of using MacOS, going back to 10.5 Leopard, so I guess it's not OS standard.

[image]

kintel, 2018-07-16:
@ErnestoBorio This was a bug in `gettext` (a 3rd party library). It's fixed in our latest development snapshots.

ErnestoBorio, 2018-07-16:
Something I didn't know is that when you add language inputs on MacOS, the OS adds those languages to the preferred language list without even telling you.
I think this may go unadvertedly for most users as usually when you are studying a new language you like to add its keyboard input maybe just for fun or studying but you hardly know the language, and you certainly do not want software UI to be in it.

commit cecf3b98 (2017-08-03): Fixes #1813: Pathed a bug in gettext