openscad/openscad #1814: nonmanifold 3 cube union

trayracing, 2016-10-02:
union of 3 cubes is nonmanifold. RepetierHost and FreeCAD both report this for this stl ouput. 
See attached, which slightly tweaked from the openscad recursion example. (BTW. levels=10 is nice!)
[stump.zip](https://github.com/openscad/openscad/files/504747/stump.zip)

Manifoldness of output is my top obstacle for Openscad usefulness.
Meshmixer and FreeCAD often can't repair the resulting stl. netfabb did better at repair, but now only emits 3mf.

I understand perfect real world manifold CSG generation is quite hard, but maybe this simple case will be easier to track down and the fix perhaps apply to more complex models.

using 2016.09.18 (git a5e4def) 

Thanks

nophead, 2016-10-02:
Yes cubes meeting at an edge are non-manifold. Overlapping them slightly fixes that.
```
        tree(rnd(0.6, 0.8, r) * length, 0.8 * thickness, count - 1, m * mt(0, length - 0.1) * mr(rnd(10, 45, r + 1)), 8 * r);
        tree(rnd(0.4, 1.0, r + 1) * length, rnd(0.7, 0.9, r + 4) * thickness, count - 1, m * mt(0, length - 0.1) * mr(-rnd(20, 35, r + 3)), 8 * r + 4);
```

If you describe a non-manifold object with openscad then that is what you get.

trayracing, 2016-10-02:
Thanks for the tweak nophead. Unfortunately, it doesn't scale well - if you crank "levels" to 10 in the file, it still fails.

nophead, 2016-10-02:
I think that is due to a well known bug that OpenScad has problems exporting objects with very close vertices.

doug-moen, 2016-10-02:
Your problem is actually a limitation in the STL file format. Your model
isn't "2-manifold", and STL can't represent these kinds of models. AMF has
the same limitation.

3MF is free of this limitation, so one solution is to convert the model to
3MF (you were able to do this with NetFabb). Or wait for 3MF support in
OpenSCAD, which is coming, see issue #1802. But you'll need a to use a tool
downstream of OpenSCAD that can import 3MF.

Some tools are able to import and use a certain class of non-manifold STL
files (your model belongs to this class), interpreting the mesh using the
same rules that are specified by the 3MF standard. So that's another
option, if you can find such tools.

A general description of the limitation in STL is that it isn't closed over
union and difference. It's possible to construct two models, both valid STL
files, such that the union or difference of the models is not valid STL.
(By contrast, 3MF is closed over union and intersection.) Another possible
solution to your problem is to decompose your model into two or more
disjoint subsets, where each subset is 2-manifold and therefore valid STL.
Some slicers are able to import multiple STL files and then union them
together during slicing, and I hypothesize that this might solve your
problem.

kintel, 2024-03-27:
Status as of 2024-03-24: When using the Manifold backend, this produces a manifold STL, which can also be re-imported into OpenSCAD (even reimported into an old OpenSCAD).