openscad/openscad #1815: Warn or produce error if multiple Root Modifiers in source

thehans, 2016-10-02:
Often times during development and debugging of a script I will use the root modifier "!" to single out a component.  Sometimes I add another "!" to a different component and forget to remove the last one.  In the case of multiple root modifiers, the behavior is undefined.

It would be nice if OpenSCAD would warn the user if it finds more than one of the "!" modifiers and points the user to the lines so they can be found quicker.

kintel, 2016-10-08:
Perfect task for new contributors :)

nophead, 2016-10-08:
It would be more useful if it rendered all of them if there is more than one. It would be obvious if one was left in by mistake but it also allows more control over what is displayed.

amarjeetkapoor1, 2016-10-11:
@kintel 
I gave it a shot as I want to understand more about OpenSCAD language.
Just added following at https://github.com/openscad/openscad/blob/master/src/localscope.cc#L49
 Thinking that this would output desired result but I didn't 
Here isRootModifier takes count of children having 
```
if(modinst->tag_root && isRootModifier>1){
    PRINTB("WARNING: Root Modifier Added At Line%d", modinst->location().firstLine());
}
```
Other option that I thought and It worked is add following in parser. It worked. If its right way then I would send PR.
 
```
module_instantiation:
          '!' module_instantiation
            {
                $$ = $2;
                if ($$) $$->tag_root = true;
                if(isRootModifier){
                    PRINTB("WARNING: Root Modifier Added At Line%d",LOC(@$).firstLine());
                }
                isRootModifier=true;
            }
```

kintel, 2016-10-11:
@amarjeetkapoor1 Look here: https://github.com/openscad/openscad/blob/master/src/node.cc#L94

That's the code responsible for finding the first root-tagged node.
I think searching for more root nodes would belong there.

@nophead's suggestion also sounds interesting, although that's a bit more work as we'd have to create a new temporary container for the tagged nodes.

We could start by showing a warning and perhaps extend it to show all root nodes later. One idea would be for ```find_root_tag()``` to return a list of nodes and display the warning if the list length is >1.

t-paul, 2016-10-11:
The warning is probably still needed even with multiple roots for cases where one of the root nodes is a sub-tree of another, e.g.
```
!difference() {
    !cube();
    sphere();
}
```
We could still show both, but I'm not sure that's a good idea.

MichaelAtOz, 2016-10-12:
I suggest the message shows the `!` ie
```
"WARNING: Root Modifier (!) Added At Line%d"
```

amarjeetkapoor1, 2017-03-04:
```
int f_root_tag(AbstractNode *n, int root_tag=0)
{
 for(auto v : n->children) {
   if (v->modinst->tag_root){
       PRINTB("WARNING: Root Modifier (!) Added At Line%d \n", v->modinst->location().firstLine());
       root_tag++;
   }
   root_tag=f_root_tag(v,root_tag);
 }
 return root_tag;
}
```

Will above code work it shows the warning for each node with ! ?

thehans, 2017-10-14:
Thanks @amarjeetkapoor1 for implementing this.

I think its safe to close this now.

nophead, 2017-10-14:
There is an issue if you put a single root modifier in a place where it
gets used multiple times. You get a warning that you have more than one.

thehans, 2017-10-14:
OK, I guess there is still room for improvement.  Reopening.

thehans, 2018-11-17:
@nophead here is a quick example of what (I think) you mean:
```
module test() {
    !cube();
}

test();
test();
```

```
WARNING: Root Modifier (!) Added At Line2 
WARNING: Root Modifier (!) Added At Line2 
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 728
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds
```

My question now is, what would be the ideal way to handle this example in terms of expected warnings?
There is only one root modifier but it gets called multiple times.  So it does seem reasonable to warn that its being used more than once.  I'm not sure what would be done to improve it.

thehans, 2020-01-07:
This was only re-opened for tracking the false multiple root warnings, but since that has its own issue: #2923 and a PR now on that, I'm closing this one.

closed 2020-01-07 by thehans