openscad/openscad #1818: need to move long term towards signing binaries for Windows

donbright, 2016-10-05:
Windows is slowly closing the garden and to survive long term, projects will have to start getting certified and signing their binaries. 

It is already incredibly 'sketchy' the way you have to run Windows programs that are not signed when downloading with the new IE ( Edge ) browser - you have to click through about 3 separate warnings just to run the installer, and they all tell you nasty messages ("Unknown Publisher") in nasty red colors about how this program may harm your computer, etc. It appears the rise of malware, botnets, hacks, and the profit motive of walled gardens is driving this shift.

A handful of open source projects are moving to do this. Right now things are pretty uncertain but within a few years it seems like this will be increasingly necessary to publish signed binaries when dealing with closed source platforms like Windows. 

This first link is where Notepad++ project discusses the problem,and winds up getting a cert from DigiCert, apparently donated to them. 

https://github.com/notepad-plus-plus/notepad-plus-plus/issues/49

Other links are random similar conversations

http://stackoverflow.com/questions/1177552/code-signing-certificate-for-open-source-projects

https://github.com/phonegap/phonegap-roadmap/issues/155#issuecomment-232148493

https://blog.aluxian.com/free-code-signing-certificate-for-open-source-software/

http://stackoverflow.com/questions/10202556/code-signing-for-an-individual-for-open-source-projects

doug-moen, 2016-10-05:
> notepad-plus-plus/notepad-plus-plus#49
> signing-certificate-for-open-source-projects
> phonegap/phonegap-roadmap#155 (comment)
> open-source-software/
> signing-for-an-individual-for-open-source-projects

Edge requires Windows 10, and Windows 10 has the ability to run Ubuntu
binaries natively. So we could add Windows/Ubuntu as a supported platform,
and in the long term, eventually drop Windows 7. Whether this is acceptable
depends on how well Ubuntu binaries run on Windows, something that I
haven't personally experienced.

kintel, 2016-10-06:
I'd be very surprised if Windows 10 i capable of running Ubuntu apps with any reasonable user experience. ..but I'm interesting in any test results!

donbright, 2016-10-07:
i just played around with ubuntu-on-windows / windows linux subsystem ( UOW LXXS ), and... first off, it doesn't solve the problem because you still have to click a bunch of stuff and reboot to get it to work (you change settings in Windows so it will run in "developer mode" and it warns you how this is not secure. then you set windows-linux-subsystem to on) The security model of UOW (Ubuntu On Windows) is not very good yet from what the blogosphere is saying. 

it's got that old frankenstein feeling that all of these kinds of systems tend to have. i read about it, and they purposely do not cater to graphics programs. you can install an X server on Windows10(TM) if you want, then we are in X11 on OSX land, which did not end well

https://support.apple.com/en-us/HT201341

Here is the Frankenstein:

```
don@DESKTOP-D3N9GH6:~$ cat /proc/version 
Linux version 3.4.0-Microsoft (Microsoft@Microsoft.com) (gcc version 4.7 (GCC) ) #1 SMP PREEMPT Wed Dec 31 14:42:53 PST 2014
don@DESKTOP-D3N9GH6:~$ cat /etc/issue
Ubuntu 14.04.5 LTS \n \l

don@DESKTOP-D3N9GH6:~$ uname -a
Linux DESKTOP-D3N9GH6 3.4.0+ #1 PREEMPT Thu Aug 1 17:06:05 CST 2013 x86_64 x86_64 x86_64 GNU/Linux
```

Key point----> kernel tag is  **_Linux version 3.4.0-Microsoft_**

We went from CEO of Microsoft saying linux was "Communist" to Microsoft shipping it's own linux kernel, in like 10 years. It's like Xenix rose from the grave https://en.wikipedia.org/wiki/Xenix 

So i tried 

      sudo apt-get install openscad

Uhm yeah. It works without gui. You can make an .stl file from a .scad file. 

Now what about running the GUI? We need an xserver, and the blogosphere suggests Xming by Colin  Harrison. 

That works too (!!!)

[image]

The funny thing is that the OpenCSG preview works properly, because it's using Gallium LLVMPIPE Mesa software renderer through Xming using the Ubuntu system. 

Windows 10 default OpenGL is like 1.1 and so native openscad binary for Windows 10 cannot do OpenCSG preview if you dont have any nice video drivers (like, on a QEMU KVM virtual machine i am testing on ). 

So yes, you can install "Ubuntu on Windows", then run apt-get install openscad, and then install Xming, and then set DISPLAY=:0, and then run openscad. And it ... basically works. 

But you are correct, you are running the exact same openscad ELF amd64 binary as for ubuntu. 

```

on@DESKTOP-D3N9GH6:~$ file /usr/bin/openscad 
/usr/bin/openscad: ELF 64-bit LSB  shared object, x86-64, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 2.6.24, BuildID[sha1]=49848e80d195b1707bec1520d633e9a1c4feeae8, stripped
don@DESKTOP-D3N9GH6:~$ sha256sum /usr/bin/openscad 
b24137800a33939d78e4c7fb8f118ac88f764cea4ca0e1ca49a7c4a098dddf77  /usr/bin/openscad

```

But .... it's not integrated with the Windows system. As far as the program is concerned, it's running under linux, using linux settings under $HOME/.config or whatever. Those map to  c:\yourname\AppData\Local\Lxss\home\yourname. The linux view can 'get to' the windows filesystem using /c/mnt but yeah. It's weird. I mean you are basically running this weird inverse version of Wine , ... "Line", inside of Windows 10. It's not an emulator, it's some kind of kernel call translator. 

Another example. . . click 'help/Openscad Manual' and ... if you dont have web browser installed under Ubuntu-on-Windows, nothing happens. 

It would be dubious to ship a windows program and tell Linux  people "run it under Wine, it's fine". Most programs are not, in fact, "fine" under wine. Many are OK, but many more are not. That is the situation with UOW LXXS i fear. Alot of things will be fine, but others will not.

MichaelAtOz, 2016-10-07:
How does it perform? e.g. example024

donbright, 2016-10-07:
it has no example 24. (current UOW is ubuntu 14.04 so it provides openscad 2014?) 

im running inside a VM so timing the opencsg preview is not going to be meaningful for comparison. i have no idea how well xming hardware opengl works or not. 
  
as far as CGAL rendering, example 21 finished in 45 seconds.

kintel, 2016-10-07:
@donbright The official Ubuntu openscad is ancient, but the official PPA has a more recent version, see http://www.openscad.org/downloads.html#linux

kintel, 2016-10-07:
@donbright Thanks for your thorough investigation on the Ubuntu-on-Windows frankenstein. Looks like a fun toy - not sure what else it's useful for :/

kintel, 2016-10-07:
Btw., if anyone is inspired to make progress on signed Windows binaries, it would be cool to see if it's possible to look into The equivalent issue for Mac (#880), especially in terms of getting a proper certificate issued. It would be interesting to see if it's possible to generate the certificates from the same key for both systems, or at least obtain the keys from the same provider.

t-paul, 2016-10-07:
I don't think it's possible to get macOS certificates from anywhere else than the Apple dev program and this CA is probably not available on Windows as trusted code signing root.
The np++ issue @donbright linked mentions that [Certum](https://www.certum.eu/certum/cert,offer_en_open_source_cs.xml) provides certificates for open source projects at $14 a year. I've not seen any free offering so far. letsencrypt does not seem to have moved on to code signing yet. StartCom/WoSign also offer only free SSL certificates, but have enough trouble with staying trusted root right now.

donbright, 2016-10-07:
So in other words, you have to have the $100/year apple developer account to sign OSX /  MacOS apps? If I am reading your post and this link correctly?

http://stackoverflow.com/questions/6257506/how-much-does-it-cost-to-get-a-code-signing-certificate-from-apple

t-paul, 2016-10-07:
Yes. For iOS, apps are automatically pulled, if the account is not renewed. I don't know how it behaves for the code signing on macOS / the macOS app store.

kintel, 2016-10-07:
My understanding is that you can sign Mac binaries with any certificate, but you need an Apple-issued certificate to bypass the annoying dialog box when people run the app => $100/year.

User friendly or freedom - choose one :(

I've been toying with the idea of offering pretty, signed binaries as a paid download for the mentioned two platforms, and regular choose-your-own-adventure or build-your-own binaries freely to whoever accepts bypassing the security measures.

DrCodeWizzard, 2016-11-20:
So it looks like the "correct" solution would be to have ability to do native windows builds. A week or two I've started trying to build OpenScad native on windows in visual studio 15 (community edition). It is definitely a bit more involved than building on Linux, so i gave up due to other commitments. But perhaps i  should retry and attempt to do it again...

kintel, 2016-11-20:
@DrCodeWizzard The challenge of a Windows native build is not so much OpenSCAD itself, but how to find or build binaries for the dependencies. One option could be to look into how other projects solve this.

doug-moen, 2016-11-20:
Why do we need a Windows native build? My understanding is that you create
the executable, any way you want, then as a separate step you sign it,
using either signtool.exe on Windows, or using one of several open source
alternatives on linux or mac.

I guess you found a problem that prevents this from working, but what is it?

DrCodeWizzard, 2016-11-21:
Not sure how the windows signing works. Assumed that it was similar to iOS
signing. Switching to driving on windows for work so it is a bit of a new
territory for me.

t-paul, 2016-11-21:
It's possible to sign iOS apps/libraries separately too, it's just much simpler using the integration in XCode.

But even if a native Windows build is not required for actual signing (and this specific issue here), it would be still nice to have a native build using the "normal" Windows tool chain. There were earlier attempts which might still give some hints (e.g. see #431).

donbright, 2022-12-02:
closing. whatever system / issues i was talking about are now doubtless obsolete.

closed 2022-12-02 by donbright