openscad/openscad #1824: Calling difference() on a imported STL model and a sphere, caused CGAL error.

andrewwutw, 2016-10-09:
Using version 2015.03-3 for OSX.

Use the file "Lid.stl"

[Lid.stl.zip](https://github.com/openscad/openscad/files/518513/Lid.stl.zip)

OpenSCAD code is following:

    difference()
    {
      import("Lid.stl", convexity=10);
      sphere(d = 1);
    }

If I use preview function(F5), I get correct output.

But if I use render function(F6), I got error message:

> 
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe_prev->is_border() || !internal::Plane_constructor::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-3/../libraries/install/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 251
Geometries in cache: 5
Geometry cache size in bytes: 625672
CGAL Polyhedrons in cache: 6
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 0 seconds
Rendering finished.

The imported STL file and sphere don't overlap, so it is very strange.

If I comment out the line "sphere(d = 1);", render function can output correctly.

If I use netfabb cloud service to fix the model : 

[Lid_repaired.stl.zip](https://github.com/openscad/openscad/files/518514/Lid_repaired.stl.zip)

    difference()
    {
      import("Lid_repaired.stl", convexity=10);
      sphere(d = 1);
    }

Render function has different error message :

>
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-3/../libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326
Geometries in cache: 9
Geometry cache size in bytes: 1833576
CGAL Polyhedrons in cache: 10
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 0 seconds
Rendering finished.

t-paul, 2016-10-09:
Blender still flags lots of zero-faces. Check the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_imported_STL_file_only_showing_up_with_F5_but_not_F6.3F) for suggestions.

nophead, 2016-10-10:
>If I comment out the line "sphere(d = 1);", render function can output
correctly.

That is because without the sphere there are CSG operations for CGAL to do,
so it isn't used, the STL is simply displayed. As soon as you do an
operation on it it needs to be 2 manifold.

andrewwutw, 2016-10-13:
Well, I tried Blender 3D Print ToolBox and MeshLab, but after fixing the model by the 2 programs, OpenSCAD still can't do boolean difference on it.
I think maybe it is a limitation of current OpenSCAD design, and this problem can't be fixed easily.

t-paul, 2016-10-13:
The limitation is that the underlying CSG library enforces 2-manifold geometry and can't handle zero-faces (as it's not possible to determine a face normal in that case). It should always be possible to fix this, but depending on the model, that might need an unreasonable amount of work in rare cases.

Normally MeshLab does the trick when following the steps shown in the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_imported_STL_file_only_showing_up_with_F5_but_not_F6.3F), but I had one case where that also did not work.

TLC123, 2016-10-15:
This might not be a bug . Its possibly a wonderful feature.

Yes interesting. I've been looking for something like this to put derivative blocking into stl files. Something that allow them to load, display, slice and print but grind to a halt if used as part in a boolean op. Still easy enough to clean up from source files for the persistent user. Thus blocking stl remixing-bots in particular by making them grind for nothing.

t-paul, 2016-10-15:
Non-manifold STL have a good risk to also break slicing while mesh based derivatives are still simple to generate. Deliberately causing more issues on us also seems not like a fair option.

Harvie, 2017-04-30:
So is there way to automaticaly fix such non-manifold broken STL? Is it possible for OpenSCAD to do it during import?

t-paul, 2017-04-30:
In general that's difficult, specifically for zero-faces it should be possible. See #1580.

Neon22, 2017-04-30:
However - in general - its an open problem in CG to fix(make watertight while preserving surfaces) an arbitrary set of polygons. There are so many examples of situtations which are undecideable without knowing information about the intention - which is absolutely not encoded in the model.

So - no - there isn;t a way to load an arbitrary STL into anything and get out a 'corrected' model. Always requires manual intervention. Blender has a usefuil key sequence for finding holes and fixing them. Meshlab adopts another approach with more flexible but harder to use tools. etc etc etc...

kintel, 2024-03-27:
Status as of 2024-03-24: This now works when using the Manifold backend.

Harvie, 2026-11-08:
still a bug tho

jordanbrown0, 2026-11-08:
But in general
* I'd say that failing to handle a defective file is not a bug.
  * Could we do some repair?  Maybe.  But that would be a feature request, not a bug.
* We aren't in a position to fix CGAL issues.
* CGAL is on its way out, maintained only for the possibility that there are cases that work in CGAL and fail in Manifold.  If it's working in Manifold, it's extremely unlikely that we will fix the CGAL path.

MichaelAtOz, 2026-11-08:
It's got a better error message.
(With Lid.stl & Lid_repaired.stl)
>Rendering Polygon Mesh using CGAL...
ERROR: The given mesh is not closed! Unable to convert to CGALNefGeometry.

I'd say that is fixed. It tells you that the mesh is bad.

Even OpenCSG (Preview) doesn't like it:

[image]

```
difference() {
  import("C:/Users/me/Documents/OpenSCAD/Tests/#1824-Lid_repaired.stl");
  translate([0,-40,-1])
    cube(80);
  }
```

Netfabb has self-intersections:

[image]

[image]

[image]

@kintel You removed **Stalled** on 9th Oct, I can't see why this is still Open?