openscad/openscad #1825: API for Python?

dingzeyuli, 2016-10-17:
Hi, I have been using OpenSCAD for a few years and it is a great software. Recently I switched to [SolidPython](https://github.com/SolidCode/SolidPython) which allows me to code only in python and generates the corresponding openscad files for rendering. 

I am curious if there is any ongoing progress on opening some official scripting interface? The current openscad language is not perfect for procedural generation of complicated models. Any comment?

Thanks!
-Ding

t-paul, 2016-10-17:
I don't think it would be possible to maintain an additional interface, there's simply not enough development resources available for that. If that will change in the future, I have no idea. It's possible and could add some nice options.

We are working on making the language more expressive, it's still focused on describing geometry and not going to be a procedural scripting language.

There's a number of projects using OpenSCAD as backend, quite a lot of those using python. I don't know if any of those are currently maintained.

dingzeyuli, 2016-10-18:
Thanks for the explanation! Really appreciate the development of openscad. 

I guess, for now, I will keep using third-party interface to openscad.

closed 2016-10-18 by dingzeyuli

donbright, 2016-10-18:
you know what is interesting is that when @shaina7837 Google Summer of Code text editor got integrated... the text editor Qt Widget got a little bit more 'modular'.... so if someone wanted to make a solid python fork... they could maybe just subclass the existing Scintilla text editor in C++... and that new SolidPython texteditor widget would be like... an editor that had SolidPython text inside of it. Then when you hit buttons like 'preview' or 'render'... it would automatically send your SolidPython editor-text through Python, which would generate a temporary .scad file, and then that .scad file could go through the openscad engine like normal and get rendered into the 3d view. I actually think you would really only need to change a little bit of code to make it do this as an experiment. 

The maintenance and polish is what would take the huge amount of time i think. For example getting back meaningful error highlighting on the python code, displaying those errors in the console widget, figuring out what to do when the python generates bad .scad code, making sure the SolidPython code stays compatible by creating regression tests, etc etc etc.

dingzeyuli, 2016-10-18:
Thanks for the info. In my opinion, since openscad is branded to be "[The Programmers Solid 3D CAD Modeller](https://github.com/openscad/openscad)", I expected a more scriptable API interface. I would argue the currently openscad a good modeling language, but it's far from a programmer's modeler... It's not easy to achieve simple for loops and if conditions.

I think the visualizing/interacting/highlighting is only part of the problem. For me, another issue which is as important is the easiness to script. By script I mean I am able to use a real programming language like python/go/c++ with API interface.

Again, the openscad project is great; there are existing third-party tools that generate openscad files. I believe a native API support may benefit the development better in the long run. Don't know if you agree, currently it's relatively difficult to maintain a complex library of parametric shapes.

t-paul, 2016-10-18:
Yes, a native API allows a number of new and interesting options, but it also massively increases the effort on the dev team which is [not exactly big](https://github.com/openscad/openscad/graphs/contributors) at this point. 

Actually I personally would like to have this too, but it seems impossible without someone really having the time and motivation to dig into this topic and do the work. Maybe (hopefully?) this will happen eventually. I'm not holding my breath though.
(There is some work going on to extract the parser part into a library, but obviously that's just a tiny part of the topic.)

As for the library of complex shapes, if that's something which is publicly available, there might be quite some people with suggestions on the forum who have created impressively huge projects already.

donbright, 2016-10-19:
sorry i am a bit slow sometimes... what would such an API look like?

dingzeyuli, 2016-10-19:
@donbright I am no expert in API design. I can share my limited idea on this. The use case that I envision is: when I am working on a design project that requires large amounts of parametric shapes, I don't have to pre-generate all of them. Instead, I can call an openscad_generate_my_geometry() function. Inside this function, I can call openscad APIs (on the fly) to perform CSG operations or other geometry processing on all the basic built-in shapes. 

In one of my computational fabrication project, [Acoustic Voxels](http://www.cs.columbia.edu/cg/lego/), I have to control the size of each of the internal voxels. In my implementation, I basically constructed a large database that has everything. But ideally I would want an on-the-fly data generation, which is only possible with openscad API support.

My use case may be very unique and specific. Hope this can motivate some thought among you guys.

donbright, 2016-10-19:
something like this??

```
generate_geometry()
  vox1 = openscad_create( "difference() cube() sphere(); " )
  vox2 = openscad_create( "translate([1,0]) difference() cube(2) cube(1);" )
  vox3 = openscad_union(vox1, vox2)
  return vox3 
```

doug-moen, 2016-10-20:
I've been thinking about this for a while, in the context of OpenSCAD2.

1. OpenSCAD is turned into a C++ library. The public API includes
   * A Value type (which includes geometric shapes in OpenSCAD2),
    with subtypes of Value for each native type.
   * Import of various file types, including *.scad (returning a Value).
   * Export of a geometric shape to various file types.
   * A preview API, taking a shape as an argument and displaying it in a window.

2. This C++ library is wrapped as a native Python module, using the CPython C API.
   * OpenSCAD Value types are wrapped as Python objects.
   * Native shape operations like cube, rotate, etc, are wrapped as Python functions, exported from an OpenSCAD module.
   * A *.scad script can be imported as a Python module.

FluxIX, 2016-10-20:
@doug-moen That was the method of providing a Python interface I was imagining.

dingzeyuli, 2016-10-20:
This looks like a promising approach. If all the core libraries are all implemented in C++, it should be quite straightforward to bind python modules. For example, the following library has been frequently used by my colleagues. https://github.com/pybind/pybind11

donbright, 2016-10-21:
one thing that seems like it would be very different, is the underlying 'data type' for the shapes  would become exposed to users of this API, where as currently these data types are often hidden inside of the evaluation of the csg tree. Not only the data type holding the vertices and faces, but the underlying number type used to store coordinates of those vertices.  "Shapes" currently go from file representations like ascii decimal STL, to OpenSCAD PolySet using binary IEEE 754 double in a c++ vector, then to CGAL Polyhedron3d, then to CGAL Nef Polyhedron3 (based on a Selective Nef Complex) using large precision Rationals (based on GNU GMP )  , and back and forth between these as needed. Sometimes you lose information doing this. It has changed over time if i remember correctly but was almost always hidden to the users. An API would expose that, which seems like a big change.

doug-moen, 2016-10-21:
@donbright "one thing that seems like it would be very different, is the underlying 'data type' for the shapes would become exposed to users of this API, where as currently these data types are often hidden inside of the evaluation of the csg tree. Not only the data type holding the vertices and faces, but the underlying number type used to store coordinates of those vertices."

There is no necessity for this, and it's not what I have in mind. C++ supports information hiding and abstract data types. In the C++ version of this API, the "Shape" class should be an abstract data type. Its data members should not be public, and you should only be able to operate on Shapes using the public functions provided by the library. The API designer gets to decide what operations to expose in the public API. It's not necessary to expose any operations that go beyond the operations provided by the OpenSCAD language and GUI. It's not necessary to expose the underlying vertex and face data: it might be desirable, but that's a question open to discussion.

In the version of this system that I'm prototyping, not all Shape objects have vertex information available. A cube does, and a polyhedron does, but a sphere does not, and a union of shapes does not. A sphere is just represented by a radius and an origin; a union is just represented by a list of children. Cylinder and Prism are different primitives: cylinders don't have vertices, prisms do. In my API, you test whether a shape (2D or 3D) has vertex information by calling is_polytope(). In my system, you don't need to compute all of the vertices of a shape in order to preview it. That's only required for STL export. I don't want to invoke CGAL to compute all the vertices in a union, just in order to preview the shape, because that's too expensive. So my shape values are not guaranteed to contain vertex information. At this point, I'm not planning to use anything other than 64 bit floats in the external representation of vertex data. So far, nothing in my planned public API requires the internal use of CGAL.