openscad/openscad #1826: Image export on Travis-CI

syky27, 2016-10-19:
Hi, I am sorry for creating a new issue but, I have nowhere else to ask.

I have problem with exporting images from openscad, while running tests on Travis-CI.

I suppose that this relates to missing X11, I noticed that you use travis, so I figured I would ask if you have any idea how could I achieve exporting images on Travis-CI.

Thank you very much for any help.

t-paul, 2016-10-19:
See https://github.com/openscad/openscad/blob/master/tests/virtualfb.sh

This runs Xvfb as X11 display. Note that latest travis images install the AMD drivers breaking OpenGL, so those need to be deinstalled.

syky27, 2016-10-19:
Hi, thanks I have tried it without any luck :(

donbright, 2016-10-19:
```
Xvfb :1 -screen 0 800x600x24 &
export DISPLAY=:1
openscad x.scad -o x.png

```

https://en.wikipedia.org/wiki/Xvfb#Usage_examples

syky27, 2016-10-19:
```
[dix] Could not init font path element /usr/share/fonts/X11/cyrillic, removing from list!
[dix] Could not init font path element /usr/share/fonts/X11/100dpi/:unscaled, removing from list!
[dix] Could not init font path element /usr/share/fonts/X11/75dpi/:unscaled, removing from list!
[dix] Could not init font path element /usr/share/fonts/X11/100dpi, removing from list!
[dix] Could not init font path element /usr/share/fonts/X11/75dpi, removing from list!
```

Didn't seemed to do the trick :(

syky27, 2016-10-19:
@t-paul how can I uninstall those drivers?

t-paul, 2016-10-20:
We use `sudo apt-get purge -qq fglrx` in [`.travis.yml`](https://github.com/openscad/openscad/blob/master/.travis.yml). Obviously that needs `sudo` enabled for the Travis worker which we need for other reasons too. I don't know if there's a way to remove the packages without the need for `sudo`.

syky27, 2016-10-20:
I used OSX on Travis and it started working...

closed 2016-11-06 by kintel