openscad/openscad #1836: Save As does a compile (preview) after saving

MichaelAtOz, 2016-10-29:
version 2016.10.05 (git 4b59170)  GSoC - so presumably side effect of customizer(?).
2015.03-2 doesn't do it.

kintel, 2016-10-29:
-> add to blocker in #1781

amarjeetkapoor1, 2016-11-14:
The reason for this behavior is that on Save As FileName is set and its the same point where the customizer picks name of JSON file to read and after reading it creates the combo-box for set of parameters. On creating Combo-box index of current selected item gets changed. So, It emits signal to recompile it.
Possible solution is that we disconnect Combo-box at https://github.com/openscad/openscad/blob/master/src/parameter/ParameterWidget.cc#L104 and reconnect after https://github.com/openscad/openscad/blob/master/src/parameter/ParameterWidget.cc#L105.

closed 2016-11-19 by amarjeetkapoor1