openscad/openscad #1837: Fails to build with flex 2.6.2: 'yypop_buffer_state' was not declared in this scope

akien-mga, 2016-10-29:
Since my distro (Mageia) upgraded flex to the current upstream version 2.6.2, openscad 2015.03-3 won't build:
```
g++ -c -O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fomit-frame-pointer -march=i586 -mtune=generic -fasynchronous-unwind-tables -fno-strict-aliasing -frounding-math -DEIGEN_DONT_ALIGN -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/libxml2 -I/usr/include/freetype2 -I/usr/include/libpng16 -O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fomit-frame-pointer -march=i586 -mtune=generic -fasynchronous-unwind-tables -fno-strict-aliasing -DPIC -fPIC -std=gnu++0x -D_REENTRANT -Wall -W -Wno-unused-local-typedefs -fPIC -DOPENSCAD_VERSION=2015.03-3 -DOPENSCAD_SHORTVERSION=2015.03 -DOPENSCAD_YEAR=2015.0 -DOPENSCAD_MONTH=03.0 -DOPENSCAD_DAY=.0 -DENABLE_CGAL -DENABLE_OPENCSG -DUSE_SCINTILLA_EDITOR -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_CORE_LIB -I. -Isrc -I/usr/lib/qt5/include -Isrc/libtess2/Include -I/usr/lib/qt5/include/QtOpenGL -I/usr/lib/qt5/include/QtPrintSupport -I/usr/lib/qt5/include/QtWidgets -I/usr/lib/qt5/include/QtGui -I/usr/lib/qt5/include/QtConcurrent -I/usr/lib/qt5/include/QtCore -Iobjects -isystem /usr/include/eigen3 -I/usr/lib/qt5/mkspecs/linux-g++ -o objects/lexer_lex.o lexer_lex.cpp
src/lexer.l: In function 'int lexerlex()':
src/lexer.l:145:21: error: 'yypop_buffer_state' was not declared in this scope
  yypop_buffer_state();
                     ^
src/lexer.l: In function 'void includefile()':
src/lexer.l:285:58: error: 'yypush_buffer_state' was not declared in this scope
   yypush_buffer_state(yy_create_buffer(yyin, YY_BUF_SIZE));
                                                          ^
Makefile:7127: recipe for target 'objects/lexer_lex.o' failed
```

Note that I'm also using `sed -i -e 's!/usr/bin/flex!/usr/bin/flex --prefix=lexer!' Makefile` as a workaround for #1546.

t-paul, 2016-10-29:
Same question as in the other issue: Is there an easy to use live image (or something similar) to test this on?. I see images for Magaia 5 which seems to be from June last year, so that probably does not help?

akien-mga, 2016-10-29:
As of now no, we're having issues spinning live images recently. But if flex is the issue, it's likely not Mageia specific and could probably be reproduced on another distro with flex 2.6.1 - from a quick search I can't find one yet (it was released a few days ago, even Arch doesn't ship it yet), so I'll try downgrading on Mageia and see if that fixes the issue.

t-paul, 2016-10-29:
No, Debian/testing here, 2.6.1 is fine:
```
$ flex --version
flex 2.6.1
```
And https://build.opensuse.org/package/show/home:t-paul/OpenSCAD also shows no issues, OpenSUSE Tumbleweed has flex 2.6.1 too and compiles without error. So maybe it actually is an issue in flex, that version is 5 days old?

akien-mga, 2016-10-29:
Yeah, I meant could be reproduced with 2.6.2, was a typo.

t-paul, 2016-10-29:
Unless we are having something stupid in our lexer, that would still make it a flex bug, no? 

I'll try if I can build flex 2.6.2 from source.

akien-mga, 2016-10-29:
Well I'm currently trying to build with flex 2.6.1 to confirm that. If it does build, then yeah, it would be either a flex bug, or an API change that they willingly made and that you'd have to take into account.

akien-mga, 2016-10-29:
Alright, openscad 2015.03-3 builds fine on Mageia Cauldron with flex 2.6.1, so it's definitely an issue with version 2.6.2. I'll report it upstream and link this report.

akien-mga, 2016-10-29:
I'm not familiar with flex, but https://github.com/westes/flex/issues/113 sounds like a good candidate.

t-paul, 2016-10-29:
Thanks for the quick check. That flex issue could be related, so lets see what the answer there is and if we need to add some flex version detection (considering this is just a minor version change, it's probably not intended though).

t-paul, 2018-11-01:
Closing as there's no more feedback and no progress. Official builds are now at flex versions later than 2.6.2 without showing issues.

closed 2018-11-01 by t-paul