openscad/openscad #1838: QT build error on macOS

dmopalmer, 2016-10-29:
When running macosx-build-dependencies.sh with the -f flag, it fails to build qt with the problem described in [this discussion](http://stackoverflow.com/questions/39492617/xcode-8-error-project-error-xcode-not-set-up-properly-even-though-the-licens).

The symptom error message is `Xcode not set up properly. You may need to confirm the license agreement by running /usr/bin/xcodebuild.`

The fix is, apparently, in ../libraries/src/qt-everywhere-opensource-src-5.7/qtbase/mkspecs/features/mac/default_pre.prf
```
    # Make sure Xcode is set up properly
+    isEmpty($$list($$system("/usr/bin/xcrun -find xcodebuild 2>/dev/null"))): \
-    isEmpty($$list($$system("/usr/bin/xcrun -find xcrun 2>/dev/null"))): \
        error("Xcode not set up properly. You may need to confirm the license agreement by running /usr/bin/xcodebuild.")
```
(manually copied/pasted/modified: this is not an automatic diff suitable for a patchfile)

I haven't actually implemented and/or tested the fix because I don't know the openscad build system, where to put patches, etc.

kintel, 2016-10-29:
@dmopalmer is there an upstream bug report for this that we could track?

dmopalmer, 2016-10-29:
On the qt site:
https://bugreports.qt.io/browse/QTBUG-55649
https://bugreports.qt.io/browse/QTBUG-56004
note that it is fixed in 5.7.1

kintel, 2016-10-29:
thx - we could patch our build system in the interim.
If you want to have a crack at it, this is how we did it last time:
https://github.com/openscad/openscad/commit/e15c36f53fa2ecdbcef189c5e4a93d08cdb33fc3#diff-1b921c0a239f34c540d19ba7197c4e7b

dmopalmer, 2016-10-29:
I am so far from being able to successfully compile openscad that I can't expect to reach a good endpoint, therefore I would not be able to enter a validated pull request.

(Next problem, my MacPorts-installed qt is conflicting with the one installed by anaconda.)

kintel, 2016-10-29:
I don't expect there to be other problems than qt, so it shouldn't be that hard to get it to work. I use that script all the time, but I'm still trying to get an Xcode 8 system up and running..

dmopalmer, 2016-10-30:
I just submitted a pull request.  Two files had to be patched, vs the singe file of QTBUG-56004

As I predicted, being able to make Qt got me just one step closer to being able to build OpenSCAD.  The build of QScintilla (the next dependency after Qt) fails.

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -std=c++11 -stdlib=libc++ -O2 -std=gnu++11 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.8 -fvisibility=hidden -fvisibility-inlines-hidden -w -fPIC -DQSCINTILLA_MAKE_DLL -DSCINTILLA_QT -DSCI_LEXER -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MACEXTRAS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I. -I. -I../include -I../lexlib -I../src -I../../../install/lib/QtPrintSupport.framework/Headers -I../../../install/lib/QtWidgets.framework/Headers -I../../../install/lib/QtMacExtras.framework/Headers -I../../../install/lib/QtGui.framework/Headers -I../../../install/lib/QtCore.framework/Headers -I. -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/AGL.framework/Headers -I../../../install/mkspecs/macx-clang -F/Users/palmer/repo/notmine/3dprinting/libraries/install/lib -o qsciapis.o qsciapis.cpp
In file included from In file included from qsciscintilla.cpp:23:
./Qsci/qsciscintilla.h:2138:25: error: implicit instantiation of undefined template 'QPointer<QsciLexer>'
qsciapis.cpp:23:
In file included from ./Qsci/qsciapis.h:33:
In file included from ./Qsci/qsciabstractapis.h:33:
./Qsci/qsciscintilla.h:2138:25: error: implicit instantiation of undefined template 'QPointer<QsciLexer>'
    QPointer<QsciLexer> lex;    QPointer<QsciLexer> lex;
```

kintel, 2016-10-30:
@dmopalmer Thanks! Chasing QScintilla is next :)
That's also reported somewhere else: https://github.com/paralect/robomongo/issues/1220

kintel, 2016-10-30:
@dmopalmer See #1847 for a fix for QScintilla

kintel, 2016-10-30:
Next issue is glib:
```
configure: error:
*** GLib requires a 64 bit type. You might want to consider
*** using the GNU C compiler.
```

This turns out to be related to a new security feature in OS X 10.11, see e.g. http://apple.stackexchange.com/questions/215030/el-capitan-make-check-dyld-library-path

I don't have a fix for that yet. It looks like all shell scripts are considered protected binaries and thus are not allowed to be passed an DYLD_LIBRARY_PATH, meaning it cannot find the libraries we just built.

I can think of two workarounds:

1. Build our own version of bash to be able to run any script we want
2. Create all libraries using absolute install_name paths, and rewrite these to @rpath after we've built OpenSCAD binaries.

kintel, 2016-10-31:
Looks like GTK is working around this by compiling their own system utils for build purposes, most notably bash.

kintel, 2016-10-31:
More info: https://forums.developer.apple.com/message/31148

donbright, 2016-10-31:
whats the downside of using the install_name thing?
i am curious because i am slowly working on getting openscad to build on linuxbrew (a linux version of homebrew) and i noticed their project made a conscious decision to avoid using  LD_LIBRARY_PATH which is apparently somewhat the linux equivalent of  DYLD_LIBRARY_PATH

kintel, 2016-10-31:
I have to look closer at it, but the primary downside would be that I think we'd need one install_name for when we're building locally, and another install_name for when we're distributing. ..so we'd need to (recursively) go through all libs and dependencies and rewrite the install_name with an @rpath or equivalent after linking the final binary.

Right now, we set the correct @rpath at the time of building each dependency. That's what breaks the build script.

t-paul, 2016-11-01:
Oh, linuxbrew? Never heard of that before, that sounds interesting. I'll have to read up on that.

Could that maybe help with creating AppImage binaries too? With all that competing non-distribution stuff being pushed lately, AppImage seems like a good choice as it does not depend on special servers, base-images and such stuff.

donbright, 2016-11-01:
i highly recommend this podcast on Brew https://changelog.com/podcast/223

imho appimage/flatpak/snap is very different from brew/pkgsrc, however they both have at their core the question of when / where / how to set paths for linking. (maybe "portable linux binary" would be a good separate discussion issue,,,)

t-paul, 2016-11-01:
Yeah, right, it's a bit off-topic in this issue, I'll create a new issue for discussion...

kintel, 2016-11-01:
yeah, Apple is threading a dangerous path. Sounds like a lot of people are looking for a way out..

kintel, 2016-11-05:
@dmopalmer I've merged a few issues related to this. If you check out #1855, you should be able to build on El Capitan with Xcode 8.

Btw., just out of curiosity: Are there any particular reasons why you want to build OpenSCAD yourself rather than using the official or snapshot binaries?

dmopalmer, 2016-11-05:
I checked out the el-capitan branch and am building now.  I am actually on macOS Sierra (10.12.1 and Xcode 8.1).

I am trying to build it myself because the official binary is a year and a half old, and because I didn't scroll down far enough to see that there were snapshot binaries.  I expected building it to be simple (hope over experience, as usual).

kintel, 2016-11-05:
ok, perhaps we should make snapshot binaries more visible.

If you have further ideas on how to make builds easier, I'm interested in ideas. The goal has been to make Homebrew the "easy way", but Homebrew currently has some issues with Qt on Sierra.

dmopalmer, 2016-11-06:
That gave me a clean build.

I use macports instead of Brew because of the cavilier way in which Brew stomps all over system directories (at least in the past).

kintel, 2016-11-06:
Good :)
FYI: Brew has stopped messing with system directories.

One question about MacPorts: What's missing from MacPorts today which would let you build OpenSCAD using a clean MacPorts install (i.e. just download the OpenSCAD sources and build)?

PR #1855 (2016-11-06): Build tweaks to not depend on DYLD_LIBRARY_PATH during building
Should fix #1838
closed 2016-11-06 by kintel