openscad/openscad #1840: Customizer: Don't scan for parameters in included files

kintel, 2016-10-29:
Fix handling of includes, those should not be scanned for parameters (see [forum post](http://forum.openscad.org/GSoC-OpenSCAD-project-td18102i20.html#a18761))

amarjeetkapoor1, 2016-11-14:
Any idea on how to solve this issue. Only solution that is coming to my mind is that we keep the name of file like the location is stored for each node (might be usefull for long run) or simply store a boolean to tell whether its from from included file or not to make customizer behave like that of Thingiverse. Otherwise, The behavior is somewhat correct (50%) as it's meant to behave like if part of main file.

MichaelPFrey, 2018-02-25:
Currently, parameters from includes get either no or the wrong description:

[image]

tricky situation

amarjeetkapoor1, 2018-02-25:
#2044 Could solve this BUT there is a problem attached.
Which would be solved easily by increasing stack size that wouldn't be a viable solution but not good one and another one would require some extra work.

PR #2322 (2018-03-05): Add the filename into the AST Tree
originally _Don't scan for parameters in included files without changing YYLTYPE [WIP]_
#2044 rebased to current master and changed, so that we do not need to change YYLTYPE.

closes #1840

Work in Progress.
closed 2018-03-05 by kintel