openscad/openscad #1841: Customizer: Stop parsing at the first non variable assignment

kintel, 2016-10-29:
Stop parsing at the first non variable assignment (not counting use<> and include<>) (see [forum post](http://forum.openscad.org/GSoC-OpenSCAD-project-td18102i20.html#a18761))

(moved from #1781)

amarjeetkapoor1, 2016-10-30:
@kintel, Should we add this as it would only limit our customizer. Although it will make it compatible to Thingiverse's customizer. Might be after this release they might adopt this customizer but if we still want to make this I think we could do it by using visitors(if I understood concept of visitors correctly) that you were working on.

kintel, 2016-10-30:
I think the question is: How do we stop top-level "internal" variables from being visible in the customizer?

We could use other rules, for example "don't show parameters without a description or parameter comment".

amarjeetkapoor1, 2016-10-30:
Couldn't understand what you said like what you meant by top-level internal variable and How above rule will help in this.

What I understand is we need to start visiting the top level node in sequence and stop when we get node expect the assignment node and thing like visitor would help in traversing top level nodes or Maybe there would be is a simpler solution which I am unable to see.

kintel, 2016-10-30:
The purpose of stopping parsing at the first statement that is not variable assignment, is to hide some variables from the customizer.

This is a hack done by Thingiverse Customizer, which we don't need to follow.
..but we do need a way of hiding some parameters from the customizer GUI. One way of doing that could be to hide all variables that don't include a description or comment.
There might be other ways too, but I haven't thought about it.

amarjeetkapoor1, 2016-10-30:
If we don't need to replicate that and we only need way of hiding some parameters we can do that by putting them under group /*[Hidden]*/ then that parameters will not be visible.

```
/*[Hidden ]*/
a=1;
```

amarjeetkapoor1, 2016-10-30:
Above method is supported and used by thingiverse only and already part of functionality of our customizer.

t-paul, 2016-10-30:
Yep, I know, I was referring to the comment from @kintel to maybe skip variables that have no comments. That would make maybe 50% of the existing models unusable for our implementation as there there seem to be a lot of models just using descriptive variables names and no extra comments (at least from the small selection I've had a look at so far).

amarjeetkapoor1, 2016-10-30:
okay. I thought you were referring to my comment. 
By the way, According to me we shouldn't hide parameters after first non-assignment as that is not a hack done by thingiverse but they just saved some extra efforts required in starting and provided it as feature which they now can't take way (just my speculation). I think we shouldn't limit our customizer with it. Might be thingiverse people might adopt our customizer as its in build feature.

kintel, 2016-10-30:
@amarjeetkapoor1 Thanks for the clarification. I agree - closing

kintel, 2016-10-30:
After quick discussion with @t-paul, I think we should implement this after all, to be compatible with Thingiverse.

..mostly because Thingiverse users are likely to be the largest consumers of this function in the start. In the future we could make this an optional "Thingiverse compatibility mode", or turn it off, depending on how we progress with a better syntax.

amarjeetkapoor1, 2016-10-30:
Okay, Read  discussion. If you both believe, then we should add this and then we also have to work on issue related to include #1840 otherwise I was about to discuss that also as It was correct behaviour according to me, as include means that all code in the file that is included should behave like if that code is written in main file. So, parameters of that file should be shown in customizer.

kintel, 2016-10-30:
In terms of include, let's discuss that in #1840

MichaelAtOz, 2016-10-30:
Thinigiverse doco says "Parameters should be placed at the top of your script before the first module declaration" so presumably that is what it uses as the delimiter.
I just PMed tbuser to see if he is willing to share the specifics.

tbuser, 2016-10-31:
@MichaelAtOz That is basically correct, Thingiverse only looks for parameters before the first module defined.  To be more specific, it actually parses all lines above the first { character ie from Customizer code:

```
# first get everything before the first {
# this assumes all the variables we want are at the top of the script before any real code
header = /[^{]*/.match(@code)[0]
```

And to hide variables, like @amarjeetkapoor1 said, you can put those params inside the [Hidden] tab.

That being said, I'd really like to make sure Customizer is compatible with this new system and am willing to make changes to it to make it compatible.  However, since we have over 5,000 customizer scripts on Thingiverse, we need to be careful not to break existing scripts hosted there.

amarjeetkapoor1, 2016-10-31:
@tbuser, Is getting  first { character only thing after which thingiverse customizer stop parsing lines. If thats the case then we can also do the same i.e. search for { in SCAD file and then find the line number at which it first occurred  and then we can add a check if the line number stored in assignment node is greater than that we found for first { .Stop parsing there only.
It would save processing need to find first non assignment node.

kintel, 2016-10-31:
@amarjeetkapoor1 I think that makes sense for now. Just be aware of potentially quoted `{` characters in comments and strings.

For a more correct extraction, let's wait until we have an AST visitor API.

amarjeetkapoor1, 2016-10-31:
Okay , I will start and take care of the pointers mentioned by you. Just want a confirmation from the @tbuser about what he said i.e. there is no other parameter apart from getting first `{` to stop.

tbuser, 2016-10-31:
That's correct.  And yeah Customizer is pretty lazy, doesn't analyze the syntax, just does string parsing.  :)

amarjeetkapoor1, 2016-10-31:
Thanks @tbuser. Now I could start.

amarjeetkapoor1, 2016-11-02:
I have done this part just their is one difference that in thingiverse customizer :
```
a=12; {} //parameter visible
```
```
{} a=13; //no parameter visible
```
but with our customizer is could be either parameters will be visible in both cases or in non as I am just comparing the line number not the column till now.
 
Do we need to take care of above cases?

and when I tried that and checked out the column number for assignments  for this (https://github.com/openscad/openscad/blob/gsoc2016-refactored/examples/Parametric/candleStand.scad ) I got this
 `26 27 29 29 27 26 34 1 1 1 1 1` this sequence where each number tells the starting column number of each assignment but it should be 1 for all as all assignments are starting for first column only.

amarjeetkapoor1, 2016-11-02:
@tbuser Just want to know how does customizer handle groups. could you point me to it or just let that if the group is defined like  

```
/* [ hello

world ] */

```

Will it be considered group or not.

tbuser, 2016-11-02:
@amarjeetkapoor1 I do not think that would result in a tab currently.  Looking at the code it needs to be all on one line I believe.

closed 2016-11-07 by amarjeetkapoor1