openscad/openscad #1843: Add line number to Warnings

MichaelAtOz, 2016-10-30:
Any chance? 
```
WARNING: Ignoring unknown variable 'DysonV'. 
ECHO: "recessV8", i = 1, logo = false, dysonV = 8, 2.5
WARNING: Ignoring unknown variable 'DysonV'. 
ECHO: "recessV8", i = 1, logo = false, dysonV = 8, 2.5
```

kintel, 2016-10-30:
We're getting closer do being able to do this - we recently added location tracking of all expressions. We just don't have that information available where we output that warning, so we still need a bit of refactoring.

t-paul, 2016-10-30:
I guess this should be "add file name and line number" so it works for multi file designs too. I've added the line number to `assert()` as it is available already, we still need some way to determine the file name.

kintel, 2016-10-30:
Let's open a separate issue for filename as that's a slightly different task.

Ma-XX-oN, 2016-11-23:
Don't forget the Errors as well.

MichaelPFrey, 2018-10-31:
which Warnings and/or Errors would also benefit from the line number (at later a relative path)?
Best is a short snippet of code that triggers the warning/error.

nophead, 2018-10-31:
I think all errors and warnings should show line number and file. Otherwise, with a large library, it can be like trying to find a needle in haystack.

MichaelPFrey, 2018-10-31:
The problem with "all" is, that not all errors and warnings belong (meaningfully) to a specific line.

One example for an ERROR not belonging to user code:

https://github.com/openscad/openscad/blob/1bcfae4388c6b4674422e03fba88510260ceffe8/src/PlatformUtils.cc#L134-L150

One example where the ERROR should not be able to be triggered by user code: (meaning I can not write a regression test for it)

https://github.com/openscad/openscad/blob/1bcfae4388c6b4674422e03fba88510260ceffe8/src/context.cc#L128

One example where a call stack would be more useful then a line number:

https://github.com/openscad/openscad/blob/1bcfae4388c6b4674422e03fba88510260ceffe8/src/exceptions.h#L24-L28

MichaelPFrey, 2018-11-01:
here we have a few:
```
polygon(points=[[0,0],[100,0],[130,50],[30]]);

polyhedron(
  points=[ [10,10,0],[10,-10,0],[-10,-10,0],[-10,10,0],
           [0]  ],
  faces=[ [0,1,4],[1,2,4],[2,3,4],[3,0,4],
              [1,0,3],[2,1,3] ]
 );
```

MichaelPFrey, 2018-11-01:
next:
```
import("example012.aaa");
import("example012.off");
hello();
circle(r=1,d=2);
```
(note: the files tobe imported to not exist)

nophead, 2018-11-01:
There seems to be some confusion with path separators on Windows. 
```
WARNING: Ignoring unknown module 'hello'. 
WARNING: Ignoring radius variable 'r' as diameter 'd' is defined too. 
Compiling design (CSG Products generation)...
WARNING: PolySet has degenerate polygons 
ERROR: Unsupported file format while trying to import file '"example012.aaa"' 
WARNING: Can't open import file 'D:/CHRIS/Downloads\example012.off'. 
```

MichaelPFrey, 2018-11-01:
Linux for comparison:
```
WARNING: Ignoring unknown module 'hello'. 
WARNING: Ignoring radius variable 'r' as diameter 'd' is defined too. 
Compiling design (CSG Products generation)...
ERROR: Unsupported file format while trying to import file '"../example012.aaa"' 
WARNING: Can't open import file '/home/michael/example012.off'. 
```
Note that this issue is about adding line numbers (and later file name) to warning and errors.
I am currently collecting test cases for that.

The file path issue within windows is separate.

Thinking about it: that message outputs an absolute path - that will not work with the test framework.

Well: in the long run I want to uncomplete all paths we output "anyway" so that the test framework - maybe/likely that will fix the path separator inconsistency  - but that is a long therm goal.

MichaelPFrey, 2018-11-02:
recursion ERROR see #2547 I assume that this kind of error will need more discussion then the other kinds.

MichaelPFrey, 2018-11-02:
Bad Range could also be improved with a line number:
```
a=100;

for (i = [0:a*a]){
    echo(i);
}
```

MichaelPFrey, 2018-11-06:
`echo([ for (i=0;i<=10000000;i=i+1) i ]);`

MichaelPFrey, 2019-02-01:
I want to make my self an overview:
   * cgal has various warnings without location
   * csg has various warnings without location
   * clipper-utils has a warning without location
  * `WARNING: Attempt to modify constant` does not have a location, but I do not know how to trigger that warning
  * ControlModule::getChild( has warnings  without location (we already have evalcontext there, so that should not be hard)
  * ControlModule::instantiate( has warnings  without location (easy to do, see #2751)
I will proceed during the weekend.

nophead, 2019-02-01:
Looks like the only way to get the constant warning is to define PI a
second time in the C++ code.

ringerc, 2019-10-21:
This appears to have been implemented.

```
WARNING: max() parameter could not be converted, in file pegboard.scad, line 103
```

However, #381 added stack traces for the `assert()` function. I'd really like to be able to enable that for warnings too, otherwise it's a nightmare to find out what the context is in terms of parameters passed etc. `echo()` debugging is so 1990s ;)

t-paul, 2019-10-21:
Right, closing this one and moving the overview list to #2620.

closed 2019-10-21 by t-paul