openscad/openscad #1845: Customizer GUI Improvements

amarjeetkapoor1, 2016-10-30:
Moved from #1781 
- [ ] Make whole parameter display a bit more compact (borders, name only bold not bigger font size...)
- [ ] Match Window border width with other windows (see [forum post](http://forum.openscad.org/GSoC-OpenSCAD-project-td18102i20.html#a18743))
- [ ] Fix font sizes in Windows (see [forum post](http://forum.openscad.org/GSoC-OpenSCAD-project-td18102i20.html#a18743))
- [ ] Check vertical spacing (see [forum post](http://forum.openscad.org/GSoC-OpenSCAD-project-td18102i20.html#a18743))
- [ ] Fix slider paging (see [forum post](http://forum.openscad.org/GSoC-OpenSCAD-project-td18102i20.html#a18743))
- [ ] Replace labels "+" and "-" with three separate labels ( someone found them misleading):
    1. New - for a new Preset.
    2. Update - to update the current Preset with the current parameter values.
    3. Delete - to delete the current Preset (after a confirmation).
- [ ] Check mouse scrolling behavior (parameter pane vs. range controls) (see [forum post](http://forum.openscad.org/GSoC-OpenSCAD-project-td18102i20.html#a18761))

amarjeetkapoor1, 2017-02-08:
Should buttons be used for New, Update and Delete or icons?

kintel, 2017-02-08:
Starting with buttons makes sense to me, to get good functionality in place. Then we can look into making it pretty.

PR #2176 (2017-11-05): Customizer: Refractoring and beautify
**Cosmetic**
- Cosmetic changes should close #1845 _(except the scrolling issue)_
- layout more compact
- new option "inline details" options for designers that use short variables names _(note: line wrap is intentionally disabled)_
- removed the headline, make the button full width _(-> helps with short headlines, "feels" more consistent)_
- sliderValue
  * now above the slider, so that the value does not affect the length of the slider _(when dealing with "big" numbers it gets "glitch" as the growing text forces the slider to scale down, changing the val [... cut]
closed 2017-11-05 by kintel