openscad/openscad #1848: openscad: xcb_io.c:165: dequeue_pending_request: Assertion `!xcb_xlib_unknown_req_in_deq' failed.

solenskiner, 2016-10-31:
When clicking new in the first window of openscad, it craches speewing this in the terminal. Opening older projects seems to work though.

[xcb] Unknown request in queue while dequeuing
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
openscad: xcb_io.c:165: dequeue_pending_request: Assertion `!xcb_xlib_unknown_req_in_deq' failed.
Aborted (core dumped)

Using

> openscad -v
OpenSCAD version 2015.03-3

kintel, 2016-10-31:
Which OS this this and how did you install OpenSCAD?

solenskiner, 2016-10-31:
Archlinux. Installed it with the package manager today.

Here is a list of relevant libraries and versions, in case the world has
changed underneath your feet; archlinux is quite bleeding edge.

> for lib in $(ldd $(which openscad) | cut -d">" -f 2 | cut -d"(" -f1 | tr
-d "\t "); do pacman -Qo $lib; done
error: failed to find 'linux-vdso.so.1' in PATH: No such file or directory
/usr/lib/libCGAL.so.12 is owned by cgal 4.9-1
/usr/lib/libmpfr.so.4 is owned by mpfr 3.1.5-1
/usr/lib/libgmp.so.10 is owned by gmp 6.1.1-1
/usr/lib/libopencsg.so.1 is owned by opencsg 1.4.2-1
/usr/lib/libGLEW.so.2.0 is owned by glew 2.0.0-1
/usr/lib/libboost_program_options.so.1.62.0 is owned by boost-libs 1.62.0-2
/usr/lib/libboost_filesystem.so.1.62.0 is owned by boost-libs 1.62.0-2
/usr/lib/libboost_system.so.1.62.0 is owned by boost-libs 1.62.0-2
/usr/lib/libboost_regex.so.1.62.0 is owned by boost-libs 1.62.0-2
/usr/lib/libglib-2.0.so.0 is owned by glib2 2.50.1-1
/usr/lib/libharfbuzz.so.0 is owned by harfbuzz 1.3.3-1
/usr/lib/libfontconfig.so.1 is owned by fontconfig-infinality-ultimate
2.11.95-4
/usr/lib/libfreetype.so.6 is owned by freetype2-infinality-ultimate 2.6.3-4
/usr/lib/libqscintilla2.so.12 is owned by qscintilla-qt4 2.9.3-1
/usr/lib/libQtOpenGL.so.4 is owned by qt4 4.8.7-10
/usr/lib/libQtGui.so.4 is owned by qt4 4.8.7-10
/usr/lib/libQtCore.so.4 is owned by qt4 4.8.7-10
/usr/lib/libGL.so.1 is owned by nvidia-libgl 370.28-1
/usr/lib/libGLU.so.1 is owned by glu 9.0.0-4
/usr/lib/libX11.so.6 is owned by libx11 1.6.4-1
/usr/lib/libpthread.so.0 is owned by glibc 2.24-2
/usr/lib/libstdc++.so.6 is owned by gcc-libs-multilib 6.2.1-1
/usr/lib/libm.so.6 is owned by glibc 2.24-2
/usr/lib/libgcc_s.so.1 is owned by gcc-libs-multilib 6.2.1-1
/usr/lib/libc.so.6 is owned by glibc 2.24-2
/usr/lib/ld-linux-x86-64.so.2 is owned by glibc 2.24-2
/usr/lib/librt.so.1 is owned by glibc 2.24-2
/usr/lib/libicudata.so.57 is owned by icu 57.1-1
/usr/lib/libicui18n.so.57 is owned by icu 57.1-1
/usr/lib/libicuuc.so.57 is owned by icu 57.1-1
/usr/lib/libpcre.so.1 is owned by pcre 8.39-1
/usr/lib/libgraphite2.so.3 is owned by graphite 1:1.3.8-1
/usr/lib/libexpat.so.1 is owned by expat 2.2.0-2
/usr/lib/libz.so.1 is owned by zlib 1.2.8-4
/usr/lib/libbz2.so.1.0 is owned by bzip2 1.0.6-5
/usr/lib/libpng16.so.16 is owned by libpng 1.6.26-2
/usr/lib/libdl.so.2 is owned by glibc 2.24-2
/usr/lib/libXrender.so.1 is owned by libxrender 0.9.10-1
/usr/lib/libgobject-2.0.so.0 is owned by glib2 2.50.1-1
/usr/lib/libSM.so.6 is owned by libsm 1.2.2-2
/usr/lib/libICE.so.6 is owned by libice 1.0.9-1
/usr/lib/libXext.so.6 is owned by libxext 1.3.3-1
/usr/lib/libGLX.so.0 is owned by libglvnd 0.1.1-1
/usr/lib/libGLdispatch.so.0 is owned by libglvnd 0.1.1-1
/usr/lib/libxcb.so.1 is owned by libxcb 1.12-1
/usr/lib/libffi.so.6 is owned by libffi 3.2.1-2
/usr/lib/libuuid.so.1 is owned by libutil-linux 2.28.2-1
/usr/lib/libXau.so.6 is owned by libxau 1.0.8-2
/usr/lib/libXdmcp.so.6 is owned by libxdmcp 1.1.2-1

kintel, 2016-10-31:
@keenerd Does this ring any bells?

keenerd, 2016-11-01:
Works for me...

solenskiner I see you are using custom fontconfig/freetype stuff.  Maybe there was a soname bump and a missing rebuild?

donbright, 2016-11-02:
cannot reproduce inside VM of latest Arch updated with pacman -Syu

```
[don@heart ~]$  for lib in $(ldd $(which openscad) | cut -d">" -f 2 | cut -d"(" -f1 | 
tr -d "\t "); do pacman -Qo $lib; done

error: failed to find 'linux-vdso.so.1' in PATH: No such file or directory
/usr/lib/libCGAL.so.12 is owned by cgal 4.9-1
/usr/lib/libmpfr.so.4 is owned by mpfr 3.1.5-1
/usr/lib/libgmp.so.10 is owned by gmp 6.1.1-1
/usr/lib/libopencsg.so.1 is owned by opencsg 1.4.2-1
/usr/lib/libGLEW.so.2.0 is owned by glew 2.0.0-1
/usr/lib/libboost_program_options.so.1.62.0 is owned by boost-libs 1.62.0-2
/usr/lib/libboost_filesystem.so.1.62.0 is owned by boost-libs 1.62.0-2
/usr/lib/libboost_system.so.1.62.0 is owned by boost-libs 1.62.0-2
/usr/lib/libboost_regex.so.1.62.0 is owned by boost-libs 1.62.0-2
/usr/lib/libglib-2.0.so.0 is owned by glib2 2.50.1-1
/usr/lib/libharfbuzz.so.0 is owned by harfbuzz 1.3.3-1
/usr/lib/libfontconfig.so.1 is owned by fontconfig 2.12.1-3
/usr/lib/libfreetype.so.6 is owned by freetype2 2.7-2
/usr/lib/libqscintilla2.so.12 is owned by qscintilla-qt4 2.9.3-1
/usr/lib/libQtOpenGL.so.4 is owned by qt4 4.8.7-10
/usr/lib/libQtGui.so.4 is owned by qt4 4.8.7-10
/usr/lib/libQtCore.so.4 is owned by qt4 4.8.7-10
/usr/lib/libGL.so.1 is owned by mesa-libgl 13.0.0-1
/usr/lib/libGLU.so.1 is owned by glu 9.0.0-4
/usr/lib/libX11.so.6 is owned by libx11 1.6.4-1
/usr/lib/libpthread.so.0 is owned by glibc 2.24-2
/usr/lib/libstdc++.so.6 is owned by gcc-libs 6.2.1-1
/usr/lib/libm.so.6 is owned by glibc 2.24-2
/usr/lib/libgcc_s.so.1 is owned by gcc-libs 6.2.1-1
/usr/lib/libc.so.6 is owned by glibc 2.24-2
/usr/lib/ld-linux-x86-64.so.2 is owned by glibc 2.24-2
/usr/lib/librt.so.1 is owned by glibc 2.24-2
/usr/lib/libicudata.so.57 is owned by icu 57.1-1
/usr/lib/libicui18n.so.57 is owned by icu 57.1-1
/usr/lib/libicuuc.so.57 is owned by icu 57.1-1
/usr/lib/libpcre.so.1 is owned by pcre 8.39-1
/usr/lib/libgraphite2.so.3 is owned by graphite 1:1.3.8-1
/usr/lib/libexpat.so.1 is owned by expat 2.2.0-2
/usr/lib/libz.so.1 is owned by zlib 1.2.8-4
/usr/lib/libbz2.so.1.0 is owned by bzip2 1.0.6-5
/usr/lib/libpng16.so.16 is owned by libpng 1.6.26-2
/usr/lib/libdl.so.2 is owned by glibc 2.24-2
/usr/lib/libXrender.so.1 is owned by libxrender 0.9.10-1
/usr/lib/libgobject-2.0.so.0 is owned by glib2 2.50.1-1
/usr/lib/libSM.so.6 is owned by libsm 1.2.2-2
/usr/lib/libICE.so.6 is owned by libice 1.0.9-1
/usr/lib/libXext.so.6 is owned by libxext 1.3.3-1
/usr/lib/libxcb-dri3.so.0 is owned by libxcb 1.12-1
/usr/lib/libxcb-present.so.0 is owned by libxcb 1.12-1
/usr/lib/libxcb-sync.so.1 is owned by libxcb 1.12-1
/usr/lib/libxshmfence.so.1 is owned by libxshmfence 1.2-1
/usr/lib/libglapi.so.0 is owned by mesa 13.0.0-1
/usr/lib/libXdamage.so.1 is owned by libxdamage 1.1.4-2
/usr/lib/libXfixes.so.3 is owned by libxfixes 5.0.3-1
/usr/lib/libX11-xcb.so.1 is owned by libx11 1.6.4-1
/usr/lib/libxcb.so.1 is owned by libxcb 1.12-1
/usr/lib/libxcb-glx.so.0 is owned by libxcb 1.12-1
/usr/lib/libxcb-dri2.so.0 is owned by libxcb 1.12-1
/usr/lib/libXxf86vm.so.1 is owned by libxxf86vm 1.1.4-1
/usr/lib/libdrm.so.2 is owned by libdrm 2.4.71-1
/usr/lib/libffi.so.6 is owned by libffi 3.2.1-2
/usr/lib/libuuid.so.1 is owned by libutil-linux 2.28.2-1
/usr/lib/libXau.so.6 is owned by libxau 1.0.8-2
/usr/lib/libXdmcp.so.6 is owned by libxdmcp 1.1.2-1
```

```
OpenSCAD Version: 2015.03-3
System information: Linux 4.8.6-1-ARCH #1 SMP PREEMPT Mon Oct 31 18:51:30 CET 2016 x86_64 Arch Linux 1 CPU 1.96 GB RAM
Compiler: GCC "6.2.1 20160830" 64bit
Boost version: 1_62
Eigen version: 3.2.10
CGAL version, kernels: 4.9, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 4.8.7
QScintilla version: 2.9.3
MingW build: No
GLib version: 2.50.0
Application Path: /usr/bin
Documents Path: /home/don/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/don/.local/share/OpenSCAD/libraries
User Config Path: /home/don/.config/OpenSCAD
Backup Path: /home/don/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/don/.local/share/OpenSCAD/libraries
  /usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:

GLEW version: 2.0.0
OpenGL Version: 3.0 Mesa 13.0.0
GL Renderer: Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits)
GL Vendor: VMware, Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: GLX
PNG generator: lodepng
GLX version: 1.4
OS info: Linux 4.8.6-1-ARCH #1 SMP PREEMPT Mon Oct 31 18:51:30 CET 2016
Machine: x86_64
```

donbright, 2016-11-02:
i also cannot reproduce this after installing the alternate versions of fontconfig and freetype2 called "infinality" per this site

https://wiki.archlinux.org/index.php/Infinality

donbright, 2016-11-02:
here is a guess. from web search it may be due to graphics driver. i notice the list above shows @solenskiner is using nvidia library.

```
       /usr/lib/libGL.so.1 is owned by nvidia-libgl 370.28-1
```

i would wonder what happens if you try to force software rendering 

```
       $ LIBGL_ALWAYS_SOFTWARE=1 openscad
```

or if you disable nvidia thread stuff

```
      $  __GL_THREADED_OPTIMIZATIONS=0 openscad
```

(per https://github.com/chjj/compton/issues/344) 

or if a different GL driver is used?

kintel, 2016-11-02:
The fontconfig idea does sound plausible. The first thing we do after setting up a QGuiApplication is to spawn a thread to set up the font cache: https://github.com/openscad/openscad/blob/master/src/openscad.cc#L623

Could this be because we don't explicitly set the `Qt::AA_X11InitThreads` Application Attribute?
http://doc.qt.io/qt-4.8/qt.html#ApplicationAttribute-enum

If so, why has this never crashed earlier?

@keenerd Perhaps I've asked this earlier, but is there any reasons why Arch is still using Qt4 for OpenSCAD?

keenerd, 2016-11-02:
It still uses Qt4 because Qt5 is still buggy.  (I tried switching it over the other month when the opencsg guys fixed a big issue.)  Touching the mouse wheel to zoom causes the preview to immediately zoom all the way in.

kintel, 2016-11-02:
@keenerd The mouse wheel zoom issue doesn't sound related to opencsg at all. Was this discussed elsewhere?

solenskiner, 2016-11-02:
>i would wonder what happens if you try to force software rendering

       $ LIBGL_ALWAYS_SOFTWARE=1 openscad
>or if you disable nvidia thread stuff

      $  __GL_THREADED_OPTIMIZATIONS=0 openscad

Indeed, both of those works.

solenskiner, 2016-11-02:
```
LIBGL_ALWAYS_SOFTWARE=1 openscad
Segmentation fault (core dumped)
```
It works, but segfaults on exit

    __GL_THREADED_OPTIMIZATIONS=0 openscad

doesn't segfault.

t-paul, 2018-10-12:
Closing as there are no further updates and the issue seems to be incompatible optimizations in NVIDIA libs we have no control over.

closed 2018-10-12 by t-paul